Pulse-Eight/libcec #54: USB-CEC not working on my Kodi Helix

MischaBoender, 2015-01-07:
I'm running Kodi(buntu) Helix on my x86-64 NAS, but my CEC adapter is not working. I've upgraded to firmware 4 and downgraded to firmware 3, but I keep getting the same issues.

TV is a 2014 model Philips 55PUS8809.

"cec-client" output:
http://pastebin.com/mg9hx2G0

First "cec-client -l" output:
WARNING: [              68]	TRANSMIT_FAILED_TIMEOUT_LINE
WARNING: [              78]	the adapter did not respond with a correct firmware version (try 1, size = 1)
WARNING: [             109]	COMMAND_REJECTED: GET_DEVICE_TYPE
WARNING: [             120]	COMMAND_REJECTED: GET_LOGICAL_ADDRESS_MASK
WARNING: [             130]	COMMAND_REJECTED: GET_OSD_NAME
Found devices: 1

device:              1
com port:            /dev/ttyACM1
vendor id:           2548
product id:          1002
firmware version:    3
firmware build date: Thu Nov 15 11:09:45 2012 +0000
type:                Pulse-Eight USB-CEC Adapter

Second "cec-client -l" output:
Found devices: 1

device:              1
com port:            /dev/ttyACM1
vendor id:           2548
product id:          1002
firmware version:    3
firmware build date: Thu Nov 15 11:09:45 2012 +0000
type:                Pulse-Eight USB-CEC Adapter

MischaBoender, 2015-01-07:
I've also seen errors that the device was not reporting the correct firmware version, and a "cec-client -l" at that time reported firmware version 1, but no "build date" or "type".

opdenkamp, 2015-02-16:
there's an issue on the CEC line which is preventing the adapter from working correctly. these issues are often caused by devices that don't support CEC, but didn't configure the CEC pin as input, or a device that's connected to HDMI but not powered.

if you have devices connected that are not powered: power them or disconnect them.

please send an email to support@pulse-eight.com if this doesn't solve the issue.

closed 2015-02-16 by opdenkamp