Pulse-Eight/libcec #55: cec don't work after libevent upgrade

jlazkano, 2015-01-12:
Hello,

I have a Pulse Eight device in Debian Jessie PC. I just upgrade my system and now it doesn't work.

This is the packages that I upgrade:

Commandline: apt-get upgrade
Upgrade: libevent-2.0-5:amd64 (2.0.21-stable-1.1, 2.0.21-stable-2), ifupdown:amd64 (0.7.51, 0.7.52), python-reportbug:amd64 (6.6.1, 6.6.3), reportbug:amd64 (6.6.1, 6.6.3), libcurl3-gnutls:amd64 (7.38.0-3, 7.38.0-4)

And this is the log:

$ echo "on 0" | cec-client -s
opening a connection to the CEC adapter...
DEBUG:   [              30]	unregistering all CEC clients
DEBUG:   [              30]	Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [              32]	connection opened, clearing any previous input and waiting for active transmissions to end before starting
DEBUG:   [              32]	communication thread started
DEBUG:   [              43]	turning controlled mode on
NOTICE:  [              64]	connection opened
DEBUG:   [              64]	<< Broadcast (F) -> TV (0): POLL
DEBUG:   [              64]	processor thread started
TRAFFIC: [              64]	<< f0
DEBUG:   [              64]	setting the line timeout to 3
DEBUG:   [             105]	>> POLL sent
DEBUG:   [             105]	TV (0): device status changed into 'present'
DEBUG:   [             105]	<< requesting vendor ID of 'TV' (0)
TRAFFIC: [             105]	<< f0:8c
TRAFFIC: [             291]	>> 0f:00:8c:00
DEBUG:   [             291]	>> TV (0) -> Broadcast (F): feature abort ( 0)
DEBUG:   [             291]	marking opcode 'give device vendor id' as unsupported feature for device 'TV'
DEBUG:   [             292]	expected response received (87: device vendor id)
NOTICE:  [             292]	registering new CEC client - v2.2.0
DEBUG:   [             292]	detecting logical address for type 'recording device'
DEBUG:   [             292]	trying logical address 'Recorder 1'
DEBUG:   [             292]	<< Recorder 1 (1) -> Recorder 1 (1): POLL
TRAFFIC: [             292]	<< 11
DEBUG:   [             322]	CEC transmission - received response - TRANSMIT_FAILED_ACK
TRAFFIC: [             322]	<< 11
DEBUG:   [             359]	CEC transmission - received response - TRANSMIT_FAILED_ACK
DEBUG:   [             359]	>> POLL not sent
DEBUG:   [             359]	using logical address 'Recorder 1'
DEBUG:   [             359]	Recorder 1 (1): device status changed into 'handled by libCEC'
DEBUG:   [             359]	Recorder 1 (1): power status changed from 'unknown' to 'on'
DEBUG:   [             359]	Recorder 1 (1): vendor = Pulse Eight (001582)
DEBUG:   [             359]	Recorder 1 (1): CEC version 1.4
DEBUG:   [             359]	AllocateLogicalAddresses - device '0', type 'recording device', LA '1'
DEBUG:   [             359]	setting ackmask to  2
DEBUG:   [             364]	Recorder 1 (1): osd name set to 'CECTester'
DEBUG:   [             365]	Recorder 1 (1): menu language set to 'eng'
DEBUG:   [             365]	GetPhysicalAddress - trying to get the physical address via ADL
DEBUG:   [             365]	GetPhysicalAddress - ADL returned physical address 0000
DEBUG:   [             365]	GetPhysicalAddress - trying to get the physical address via nvidia driver
DEBUG:   [             365]	GetPhysicalAddress - nvidia driver returned physical address 0000
DEBUG:   [             365]	GetPhysicalAddress - trying to get the physical address from the OS
DEBUG:   [             365]	GetPhysicalAddress - OS returned physical address 0000
DEBUG:   [             365]	SetDevicePhysicalAddress - not setting invalid physical address 0000
NOTICE:  [             365]	setting HDMI port to 1 on device TV (0)
DEBUG:   [             366]	Recorder 1 (1): physical address changed from ffff to 1000
DEBUG:   [             366]	<< Recorder 1 (1) -> broadcast (F): physical adddress 1000
TRAFFIC: [             366]	<< 1f:84:10:00:01
DEBUG:   [             499]	using persisted autonomous mode setting: 'enabled'
DEBUG:   [             504]	using persisted CEC version setting: '1.4'
DEBUG:   [             509]	using persisted logical address setting: 'Recorder 1'
DEBUG:   [             515]	using persisted device type setting: 'recording device'
DEBUG:   [             520]	using persisted logical address mask setting: '206'
DEBUG:   [             535]	using persisted device name setting: 'CECTester'
DEBUG:   [             540]	using persisted physical address setting: '1000'
NOTICE:  [             540]	CEC client registered: libCEC version = 2.2.0, client version = 2.2.0, firmware version = 4, firmware build date: Thu Dec  6 11:15:20 2012 +0000, logical address(es) = Recorder 1 (1) , base device: TV (0), HDMI port number: 1, physical address: 1.0.0.0, host: x86_64-pc-linux-gnu, features: 'P8 USB' 'P8 USB detect' 'EXYNOS', compiled on: Fri Oct 31 15:05:35 UTC 2014 by pbuilder@www.marillat.net on Linux 3.16.6 (x86_64)
DEBUG:   [             540]	<< Recorder 1 (1) -> TV (0): OSD name 'CECTester'
TRAFFIC: [             541]	<< 10:47:43:45:43:54:65:73:74:65:72
DEBUG:   [             814]	<< requesting power status of 'TV' (0)
TRAFFIC: [             814]	<< 10:8f
DEBUG:   [             850]	CEC transmission - received response - TRANSMIT_FAILED_ACK
TRAFFIC: [             850]	<< 10:8f
TRAFFIC: [            1016]	>> 01:90:01
DEBUG:   [            1016]	>> TV (0) -> Recorder 1 (1): report power status (90)
DEBUG:   [            1016]	TV (0): power status changed from 'unknown' to 'standby'
DEBUG:   [            1016]	expected response received (90: report power status)
DEBUG:   [            1016]	<< requesting vendor ID of 'TV' (0)
DEBUG:   [            1017]	'give device vendor id' is marked as unsupported feature for device 'TV'
NOTICE:  [            1017]	<< powering on 'TV' (0)
TRAFFIC: [            1017]	<< 10:04
DEBUG:   [            1073]	TV (0): power status changed from 'standby' to 'in transition from standby to on'
DEBUG:   [            1073]	unregistering all CEC clients
NOTICE:  [            1073]	unregistering client: libCEC version = 2.2.0, client version = 2.2.0, firmware version = 4, firmware build date: Thu Dec  6 11:15:20 2012 +0000, logical address(es) = Recorder 1 (1) , base device: TV (0), HDMI port number: 1, physical address: 1.0.0.0, host: x86_64-pc-linux-gnu, features: 'P8 USB' 'P8 USB detect' 'EXYNOS', compiled on: Fri Oct 31 15:05:35 UTC 2014 by pbuilder@www.marillat.net on Linux 3.16.6 (x86_64)
DEBUG:   [            1073]	Recorder 1 (1): power status changed from 'on' to 'unknown'
DEBUG:   [            1073]	Recorder 1 (1): vendor = Unknown (000000)
DEBUG:   [            1073]	Recorder 1 (1): CEC version unknown
DEBUG:   [            1073]	Recorder 1 (1): osd name set to 'Recorder 1'
DEBUG:   [            1073]	Recorder 1 (1): device status changed into 'unknown'
DEBUG:   [            1073]	setting ackmask to  0
DEBUG:   [            1078]	turning controlled mode off
DEBUG:   [            1083]	unregistering all CEC clients
DEBUG:   [            2071]	communication thread ended

Before the upgrade, this command use to turn on the TV correctly.

Regards.

opdenkamp, 2015-02-16:
libCEC does not do anything with libevent, and the command is sent successfully.
try using this command instead, as that will also make libCEC wait for a response and switch sources: `echo as | cec-client -s`

if your PC is not connected to HDMI 1 of the TV, then you have to pass the `-p` setting to cec-client too. see `cec-client -h` for details

closed 2015-02-16 by opdenkamp

jlazkano, 2015-02-16:
Thanks and sorry for the issue, this is not related with libevent.

I reset my TV configuration and know it is working.

Regards.