Pulse-Eight/libcec #56: libCEC not working on LG TV (LD330)

OMID-313, 2015-02-07:
Hi all,

I've just installed libcec on my Raspberry Pi B+ with (raspbian on it), following the instructions from here (https://nyxi.eu/blog/2013/04/15/raspbian-libcec/).

I want to be able to turn on/off the tv via the `echo "on 0" | cec-client -s` and `echo "standby 0" | cec-client -s` commands.
But they don't work.

Here's what errors/logs I get on `echo "standby 0" | cec-client -s` :

    DEBUG:   [             123]	unregistering all CEC clients
    DEBUG:   [             127]	Broadcast (F): osd name set to 'Broadcast'
    DEBUG:   [             131]	InitHostCEC - vchiq_initialise succeeded
    DEBUG:   [             134]	InitHostCEC - vchi_initialise succeeded
    DEBUG:   [             137]	InitHostCEC - vchi_connect succeeded
    DEBUG:   [             141]	logical address changed to Free use (e)
    DEBUG:   [             145]	Open - vc_cec initialised
    NOTICE:  [             148]	connection opened
    DEBUG:   [             152]	<< Broadcast (F) -> TV (0): POLL
    DEBUG:   [             155]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
    DEBUG:   [             158]	processor thread started
    TRAFFIC: [             162]	<< e0
    DEBUG:   [             226]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
    TRAFFIC: [             229]	<< e0
    DEBUG:   [             322]	>> POLL not sent
    DEBUG:   [             325]	TV (0): device status changed into 'not present'
    NOTICE:  [             329]	registering new CEC client - v2.2.0
    DEBUG:   [             332]	detecting logical address for type 'recording device'
    DEBUG:   [             335]	trying logical address 'Recorder 1'
    DEBUG:   [             339]	<< Recorder 1 (1) -> Recorder 1 (1): POLL
    TRAFFIC: [             342]	<< 11
    TRAFFIC: [             436]	<< 11
    DEBUG:   [             529]	>> POLL not sent
    DEBUG:   [             532]	using logical address 'Recorder 1'
    DEBUG:   [             536]	Recorder 1 (1): device status changed into 'handled by libCEC'
    DEBUG:   [             540]	Recorder 1 (1): power status changed from 'unknown' to 'on'
    DEBUG:   [             544]	Recorder 1 (1): vendor = Pulse Eight (001582)
    DEBUG:   [             548]	Recorder 1 (1): CEC version 1.4
    DEBUG:   [             551]	AllocateLogicalAddresses - device '0', type 'recording device', LA '1'
    DEBUG:   [             556]	logical address changed to Recorder 1 (1)
    DEBUG:   [             560]	Recorder 1 (1): osd name set to 'CECTester'
    DEBUG:   [             564]	Recorder 1 (1): menu language set to 'eng'
    DEBUG:   [             568]	GetPhysicalAddress - physical address = 2000
    DEBUG:   [             573]	AutodetectPhysicalAddress - autodetected physical address '2000'
    DEBUG:   [             577]	Recorder 1 (1): physical address changed from ffff to 2000
    DEBUG:   [             581]	<< Recorder 1 (1) -> broadcast (F): physical adddress 2000
    TRAFFIC: [             586]	<< 1f:84:20:00:01
    NOTICE:  [             741]	CEC client registered: libCEC version = 2.2.0, client version = 2.2.0, firmware version = 1, logical address(es) = Recorder 1 (1) , physical address: 2.0.0.0, host: armv6l-unknown-linux-gnueabihf, features: 'P8 USB' 'P8 USB detect' 'RPi' 'EXYNOS', git revision: 9f000e4, compiled on: Thu Feb  5 19:25:20 UTC 2015 by root@raspberrypi on Linux 3.12.28+ (armv6l)
    DEBUG:   [             753]	<< Recorder 1 (1) -> TV (0): OSD name 'CECTester'
    DEBUG:   [             758]	<< Recorder 1 (1) -> TV (0): POLL
    TRAFFIC: [             763]	<< 10
    TRAFFIC: [             858]	<< 10
    DEBUG:   [             953]	>> POLL not sent
    DEBUG:   [             957]	not sending command 'set osd name': destination device 'TV' marked as not present
    DEBUG:   [             962]	<< requesting power status of 'TV' (0)
    DEBUG:   [             967]	<< Recorder 1 (1) -> TV (0): POLL
    TRAFFIC: [             971]	<< 10
    TRAFFIC: [            1066]	<< 10
    DEBUG:   [            1161]	>> POLL not sent
    DEBUG:   [            1165]	not sending command 'give device power status': destination device 'TV' marked as not present
    DEBUG:   [            1171]	<< Recorder 1 (1) -> TV (0): POLL
    TRAFFIC: [            1176]	<< 10
    TRAFFIC: [            1271]	<< 10
    DEBUG:   [            1366]	>> POLL not sent
    NOTICE:  [            1370]	<< putting 'TV' (0) in standby mode
    DEBUG:   [            1374]	<< Recorder 1 (1) -> TV (0): POLL
    TRAFFIC: [            1379]	<< 10
    TRAFFIC: [            1474]	<< 10
    DEBUG:   [            1569]	>> POLL not sent
    DEBUG:   [            1573]	not sending command 'standby': destination device 'TV' marked as not present
    DEBUG:   [            1578]	unregistering all CEC clients
[... 146 lines cut ...]
    DEBUG:   [            8362]	Recorder 1 (1): vendor = Unknown (000000)
    DEBUG:   [            8366]	Recorder 1 (1): CEC version unknown
    DEBUG:   [            8370]	Recorder 1 (1): osd name set to 'Recorder 1'
    DEBUG:   [            8374]	Recorder 1 (1): device status changed into 'unknown'
    DEBUG:   [            8378]	unregistering all CEC clients
    DEBUG:   [            9192]	UnregisterLogicalAddress - releasing previous logical address
    DEBUG:   [            9197]	logical address changed to Broadcast (f)

-

And the following is what I get on `cec-client -l` :

    Found Devices: 1
    
    device: 1
    com port: RPI
    vendor id: 2708
    product id: 1001
    firmware version: 1
    type: Raspberry Pi

It's also worth mentioning that I enable **Simplink** on my LG tv. However, its status is always *Check Connection*.

Any ideas !?
What is causing the problem?

OMID-313, 2015-02-07:
Could it be the HDMI cable @opdenkamp ? Why?
Why shouldn't the cable work, even if it has the correct internal wiring for pin 13 (CEC)?

opdenkamp, 2015-02-16:
your hdmi cable is probably faulty indeed. if your TV doesn't show up in "scan", then the cable is not working.
LG cannot be powered off via CEC because LG didn't include support for this.

closed 2015-02-16 by opdenkamp