Pulse-Eight/libcec #58: LibCecSharp not working on Windows 8.1

FlorentGz, 2015-02-16:
Hi,

I recently bought a Pulse Eight USB - CEC Adapter.
I’m a developper and would like to use the LibCECSharp dll in order to use it in my .NET C# app running on a Windows 8.1 Zotac. However I can’t use the DLL.

I used dependency walker which lead me to the fact that some dll were missings such as 

API-MS-WIN-CORE-KERNEL32-PRIVATE-L1-1-1.DLL
API-MS-WIN-CORE-PRIVATEPROFILE-L1-1-1.DLL
API-MS-WIN-SERVICE-PRIVATE-L1-1-1.DLL
API-MS-WIN-CORE-SHUTDOWN-L1-1-1.DLL
EXT-MS-WIN-NTUSER-UICONTEXT-EXT-L1-1-0.DLL
IESHIMS.DLL

I read that those were missing because the app is running on Window 8.1 devices. Do you have a fix so that I can use the LibCECSharp.dll on Windows 8.1 ?

Thanks in advance,

Florent

opdenkamp, 2015-02-23:
are you using some embedded version of windows, or are you running on something not x86/x86_64 compatible? we only build for standard x86/x86_64 windows versions.

FlorentGz, 2015-02-23:
I'm not running an embedded version of windows.
I decompiled, made some updates and then recompiled the dll and now it seems to work.

opdenkamp, 2015-02-23:
the source code is available here in this repository, no need to decompile. what did you change to make it work? would be good to get that in the main source. thanks

FlorentGz, 2015-02-23:
I needed to change the framework version from 2 to 4.5.
And update the warning level from visual studio in the LibCecSharp project.

opdenkamp, 2015-02-23:
okay, but it's using v2 on purpose, so it can be loaded by older applications too (like windows media center). you were just missing the framework in that case, which is described in the readme file in the root folder: https://github.com/Pulse-Eight/libcec/blob/master/README#L51

closed 2015-02-23 by opdenkamp