Pulse-Eight/libcec #60: Reference library and API/ABI for interfacing with uinput (and Python)

Hedda, 2015-02-27:
LibCEC already provide C, C++ and .NET CLR interfaces to Pulse-Eight adapter, but lack simple interfaces from uinput (or Python)

One possible solution for this is to add libcec-daemon (Linux daemon) as a reference library  for connecting libcec to uinput, this would enable all uinput compatible devices to interface with libcec.  

https://github.com/bramp/libcec-daemon

libcec-daemon is licensed under GPL right now but might be changed to BSD license later.

To extend even further on this you could add Python-uinput on top of that as is, or maybe even modify it for native libcec interface to add a Python interface for libcec

https://github.com/tuomasjjrasanen/python-uinput
http://tjjr.fi/sw/python-uinput/

Snyder1977, 2015-03-14:
A Python interface would be great! 1+

opdenkamp, 2015-03-20:
python interface is on it's way, through swig which exports our C api: https://github.com/Pulse-Eight/libcec/tree/cmake
it's not 100% finished yet. you can find an example of how the API will work here: https://github.com/Pulse-Eight/libcec/blob/cmake/support/test-libcec.py
uinput shouldn't be too hard to add. will take a look at that later.

Snyder1977, 2015-03-21:
Great! Many thanks. At the moment I have installed this python-CEC Bindung: 

https://github.com/trainman419/python-cec

It works fine for me but when it comes to setting the active source. I' am a sort of stuck. Setting the active source is not implemented in trainman's binding. My Panasonic TV is always choosing the wrong input after it has been turned on by a python script via hdmi cec. I could call the system to operate a shell command, but I am quite sure that this would be a cross platform pain. I am looking fortwarf to hearing from you....

Snyder1977, 2015-04-04:
Are there any news regarding the python interface for libcec?

opdenkamp, 2015-04-04:
See the master branch

Snyder1977, 2015-04-04:
ok, great! This is exactly what I was looking for. I found the example in src/pyCecClient. Many thanks. Are there any build instructions for Rasbian? I

opdenkamp, 2015-04-08:
just follow the instructions for linux, or wait for the maintainers to update to libCEC 3

Snyder1977, 2015-04-08:
ok, many thanks. I´ll probably wait and hope that the KODI Team will integrate libCEC3 very soon.

Hedda, 2015-04-14:
https://github.com/xbmc/xbmc/pull/6891

opdenkamp, 2015-04-14:
why not?

Snyder1977, 2015-04-14:
It would be a pain, if they didn´t integrate it! The new pyhon API of libCEC would be very usefull for many KODI plugins. It should definitley be integrated in Kodi v.15.....

opdenkamp, 2015-04-14:
I'm a team XBMC member myself, so I can press the nice green button on the PR ;-)
It'll be merged in once I've updated kodi-platform.

As for Python: Kodi won't use that (yet) and it won't be re-exported to add-ons (yet)

Hedda, 2015-04-14:
OK cool! I just thought that Kodi was already in feature freeze for v15.

opdenkamp, 2015-04-14:
merge window closes on the 20th iirc, and it's not a feature (for Kodi) but a combination of fixes. the new python stuff isn't used by Kodi. but because of the changes to the build system (cmake, platform lib), it's a bit more work than a usual libCEC bump this time

Snyder1977, 2015-04-14:
yeah, great to know that you can merge the PRs.

If it is too much work to integrate the full python API into KODI, it would be very much appreciated if there were at least some more bulit-in-functions for handling the cec stuff in KODI.

opdenkamp, 2015-07-06:
Changing Kodi's implementation to use the new Python API rather than the current C++ one is going to be a lot of work. That's going to be something for later.

closed 2025-02-27 by github-actions[bot]