Pulse-Eight/libcec #63: LibCEC hangs when a device is powered on.

UEckhardt, 2015-03-14:
Currently i am developing a plugin for vdr (http://www.uli-eckhardt.de/vdr/cec.en.shtml) and encountered a problem when the TV is powered on and off. I could also reproduce this problem via cec-client.

 The TV is powered off (really no power) and I start cec-client:
vdr:/tmp # cec-client 
No device type given. Using 'recording device'                                                              
CEC Parser created - libCEC version 2.2.0                                                                              
no serial port given. trying autodetect:                                                                                  
 path:     /sys/devices/pci0000:00/0000:00:14.0/usb1/1-3                                                                               
 com port: /dev/ttyACM0                                                                                                                     
                                                                                                                                                      
opening a connection to the CEC adapter...
DEBUG:   [              27]     unregistering all CEC clients
DEBUG:   [              27]     Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [              28]     connection opened, clearing any previous input and waiting for active transmissions to end before starting
DEBUG:   [              29]     communication thread started
DEBUG:   [              40]     turning controlled mode on
NOTICE:  [              63]     connection opened
DEBUG:   [              64]     processor thread started
DEBUG:   [              64]     << Broadcast (F) -> TV (0): POLL
TRAFFIC: [              64]     << f0
DEBUG:   [              64]     setting the line timeout to 3
DEBUG:   [             101]     CEC transmission - received response - TRANSMIT_FAILED_ACK
TRAFFIC: [             101]     << f0
DEBUG:   [             132]     CEC transmission - received response - TRANSMIT_FAILED_ACK
DEBUG:   [             133]     >> POLL not sent
DEBUG:   [             133]     TV (0): device status changed into 'not present'
NOTICE:  [             133]     registering new CEC client - v2.2.0
DEBUG:   [             133]     detecting logical address for type 'recording device'
DEBUG:   [             133]     trying logical address 'Recorder 1'
DEBUG:   [             133]     << Recorder 1 (1) -> Recorder 1 (1): POLL
TRAFFIC: [             133]     << 11
DEBUG:   [             171]     CEC transmission - received response - TRANSMIT_FAILED_ACK
TRAFFIC: [             171]     << 11
DEBUG:   [             204]     CEC transmission - received response - TRANSMIT_FAILED_ACK
DEBUG:   [             204]     >> POLL not sent
DEBUG:   [             205]     using logical address 'Recorder 1'
DEBUG:   [             205]     Recorder 1 (1): device status changed into 'handled by libCEC'
DEBUG:   [             205]     Recorder 1 (1): power status changed from 'unknown' to 'on'
DEBUG:   [             205]     Recorder 1 (1): vendor = Pulse Eight (001582)
DEBUG:   [             205]     Recorder 1 (1): CEC version 1.4
DEBUG:   [             205]     AllocateLogicalAddresses - device '0', type 'recording device', LA '1'
DEBUG:   [             205]     setting ackmask to  2
DEBUG:   [             210]     Recorder 1 (1): osd name set to 'CECTester'
DEBUG:   [             210]     Recorder 1 (1): menu language set to 'eng'
DEBUG:   [             210]     GetPhysicalAddress - trying to get the physical address via ADL
DEBUG:   [             210]     GetPhysicalAddress - ADL returned physical address 0000
DEBUG:   [             210]     GetPhysicalAddress - trying to get the physical address via nvidia driver
DEBUG:   [             210]     GetPhysicalAddress - nvidia driver returned physical address 0000
DEBUG:   [             210]     GetPhysicalAddress - trying to get the physical address from the OS
DEBUG:   [             211]     GetPhysicalAddress - OS returned physical address 0000
DEBUG:   [             211]     SetDevicePhysicalAddress - not setting invalid physical address 0000
NOTICE:  [             211]     setting HDMI port to 1 on device TV (0)
DEBUG:   [             211]     << Recorder 1 (1) -> TV (0): POLL
TRAFFIC: [             211]     << 10
DEBUG:   [             242]     CEC transmission - received response - TRANSMIT_FAILED_ACK
TRAFFIC: [             242]     << 10
DEBUG:   [             275]     CEC transmission - received response - TRANSMIT_FAILED_ACK
DEBUG:   [             275]     >> POLL not sent
DEBUG:   [             275]     Recorder 1 (1): physical address changed from ffff to 1000
DEBUG:   [             275]     << Recorder 1 (1) -> broadcast (F): physical adddress 1000
TRAFFIC: [             275]     << 1f:84:10:00:01
DEBUG:   [             413]     using persisted autonomous mode setting: 'enabled'
DEBUG:   [             418]     using persisted CEC version setting: '1.4'
DEBUG:   [             424]     using persisted logical address setting: 'Recorder 1'
DEBUG:   [             429]     using persisted device type setting: 'recording device'
DEBUG:   [             434]     using persisted logical address mask setting: '206'
DEBUG:   [             451]     using persisted device name setting: 'CECTester'
DEBUG:   [             456]     using persisted physical address setting: '1000'
NOTICE:  [             457]     CEC client registered: libCEC version = 2.2.0, client version = 2.2.0, firmware version = 4, firmware build date: Thu Dec  6 11:15:20 2012 +0000, logical address(es) = Recorder 1 (1) , base device: TV (0), HDMI port number: 1, physical address: 1.0.0.0, host: x86_64-suse-linux-gnu, features: 'P8 USB' 'P8 USB detect' 'EXYNOS'
DEBUG:   [             457]     << Recorder 1 (1) -> TV (0): OSD name 'CECTester'
[... 103 lines cut ...]
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007f6ee410f408 in pthread_cond_timedwait@@GLIBC_2.3.2 () from /lib64/libpthread.so.0
Missing separate debuginfos, use: zypper install glibc-debuginfo-2.19-16.9.1.x86_64 libcec2-debuginfo-2.2.0-5.1.x86_64 libgcc_s1-debuginfo-4.8.3+r212056-2.2.4.x86_64 liblockdev1-debuginfo-1.0.3_git201003141408-27.2.1.x86_64 libstdc++6-gcc49-debuginfo-4.9.0+r211729-2.1.7.x86_64 libudev1-debuginfo-210-25.12.1.x86_64
(gdb) bt
#0  0x00007f6ee410f408 in pthread_cond_timedwait@@GLIBC_2.3.2 () from /lib64/libpthread.so.0
#1  0x00007f6ee32dc78b in CEC::CCECProcessor::RegisterClient(CEC::CCECClient*) () from /usr/lib64/libcec.so.2
#2  0x00007f6ee32ed7e9 in CEC::CCECClient::ChangeDeviceType(CEC::cec_device_type, CEC::cec_device_type) () from /usr/lib64/libcec.so.2
#3  0x00007f6ee330410c in CEC::CVLCommandHandler::InitHandler() () from /usr/lib64/libcec.so.2
#4  0x00007f6ee32f420d in CEC::CCECBusDevice::ReplaceHandler(bool) () from /usr/lib64/libcec.so.2
#5  0x00007f6ee32f5746 in CEC::CCECBusDevice::RequestVendorId(CEC::cec_logical_address, bool) () from /usr/lib64/libcec.so.2
#6  0x00007f6ee32f1774 in CEC::CCECBusDevice::GetVendorId(CEC::cec_logical_address, bool) () from /usr/lib64/libcec.so.2
#7  0x00007f6ee32e7678 in CEC::CCECClient::GetDeviceVendorId(CEC::cec_logical_address) () from /usr/lib64/libcec.so.2
#8  0x0000000000409c57 in ProcessCommandSCAN (parser=parser@entry=0xb1fb60, command="scan", UNUSED_arguments="") at main.cpp:910
#9  0x000000000040b3af in ProcessConsoleCommand (parser=0xb1fb60, input="") at main.cpp:982
#10 0x0000000000402e46 in main (argc=<optimized out>, argv=<optimized out>) at main.cpp:1304

Sometimes I get a result, but without the TV. In this case I get the additonal log output:
DEBUG:   [           37170]     expected response received (87: device vendor id)
DEBUG:   [           37170]     replacing the command handler for device 'TV' (0)
DEBUG:   [           37170]     Recorder 1 (1): vendor = Panasonic (008045)
DEBUG:   [           37170]     replacing the command handler for device 'Recorder 1' (1)
NOTICE:  [           37170]     changing device type 'recording device' into 'playback device'
NOTICE:  [           37170]     unregistering client: libCEC version = 2.2.0, client version = 2.2.0, firmware version = 4, firmware build date: Thu Dec  6 11:15:20 2012 +0000, logical address(es) = Recorder 1 (1) , base device: TV (0), HDMI port number: 1, physical address: 1.0.0.0, host: x86_64-suse-linux-gnu, features: 'P8 USB' 'P8 USB detect' 'EXYNOS'

The cec-client crashes in this case when quitting.

UEckhardt, 2015-03-15:
Also valgrind showed a lot of errors after wuitting cec-client when the TV was swiched on, after cec-client was started. I have cut the log, since it seems not be possible to attach a log file
<pre>
==16087== Memcheck, a memory error detector
==16087== Copyright (C) 2002-2013, and GNU GPL'd, by Julian Seward et al.
==16087== Using Valgrind-3.10.0 and LibVEX; rerun with -h for copyright info
==16087== Command: /usr/bin/cec-client
==16087== Parent PID: 2038
==16087== 
==16087== Thread 6:
==16087== Invalid read of size 8
==16087==    at 0x62848C7: CEC::CCECProcessor::RegisterClient(CEC::CCECClient*) (CECProcessor.cpp:824)
==16087==    by 0x6292C31: CEC::CCECClient::ChangeDeviceType(CEC::cec_device_type, CEC::cec_device_type) (CECClient.cpp:457)
==16087==    by 0x62AD6DC: CEC::CVLCommandHandler::InitHandler() (VLCommandHandler.cpp:92)
==16087==    by 0x6299675: CEC::CCECBusDevice::ReplaceHandler(bool) (CECBusDevice.cpp:240)
==16087==    by 0x628250F: CEC::CCECProcessor::ReplaceHandlers() (CECProcessor.cpp:238)
==16087==    by 0x6282726: CEC::CCECProcessor::Process() (CECProcessor.cpp:271)
==16087==    by 0x6286C02: PLATFORM::CThread::ThreadHandler(void*) (threads.h:66)
==16087==    by 0x50410A3: start_thread (in /lib64/libpthread-2.19.so)
==16087==    by 0x5B6206C: clone (in /lib64/libc-2.19.so)
==16087==  Address 0x5e292b0 is 0 bytes inside a block of size 408 free'd
==16087==    at 0x4C2A84C: operator delete(void*) (in /usr/lib64/valgrind/vgpreload_memcheck-amd64-linux.so)
==16087==    by 0x62A4869: CEC::CCECTV::~CCECTV() (CECTV.h:42)
==16087==    by 0x62A125F: CEC::CCECDeviceMap::Clear() (CECDeviceMap.cpp:129)
==16087==    by 0x62A0FFC: CEC::CCECDeviceMap::~CCECDeviceMap() (CECDeviceMap.cpp:84)
==16087==    by 0x62A1061: CEC::CCECDeviceMap::~CCECDeviceMap() (CECDeviceMap.cpp:85)
==16087==    by 0x6281D19: CEC::CCECProcessor::~CCECProcessor() (CECProcessor.cpp:108)
==16087==    by 0x6281E29: CEC::CCECProcessor::~CCECProcessor() (CECProcessor.cpp:109)
==16087==    by 0x628C72C: CEC::CLibCEC::~CLibCEC() (LibCEC.cpp:66)
==16087==    by 0x628C7DF: CEC::CLibCEC::~CLibCEC() (LibCEC.cpp:67)
==16087==    by 0x628E0C5: CECDestroy (LibCEC.cpp:497)
==16087==    by 0x4029F4: UnloadLibCec(CEC::ICECAdapter*) (cecloader.h:164)
==16087==    by 0x407A41: main (main.cpp:1323)
==16087== 
==16087== Invalid read of size 4
==16087==    at 0x629919C: CEC::CCECBusDevice::ReplaceHandler(bool) (CECBusDevice.cpp:175)
==16087==    by 0x62848DE: CEC::CCECProcessor::RegisterClient(CEC::CCECClient*) (CECProcessor.cpp:824)
==16087==    by 0x6292C31: CEC::CCECClient::ChangeDeviceType(CEC::cec_device_type, CEC::cec_device_type) (CECClient.cpp:457)
==16087==    by 0x62AD6DC: CEC::CVLCommandHandler::InitHandler() (VLCommandHandler.cpp:92)
==16087==    by 0x6299675: CEC::CCECBusDevice::ReplaceHandler(bool) (CECBusDevice.cpp:240)
==16087==    by 0x628250F: CEC::CCECProcessor::ReplaceHandlers() (CECProcessor.cpp:238)
==16087==    by 0x6282726: CEC::CCECProcessor::Process() (CECProcessor.cpp:271)
==16087==    by 0x6286C02: PLATFORM::CThread::ThreadHandler(void*) (threads.h:66)
==16087==    by 0x50410A3: start_thread (in /lib64/libpthread-2.19.so)
==16087==    by 0x5B6206C: clone (in /lib64/libc-2.19.so)
==16087==  Address 0x5e292cc is 28 bytes inside a block of size 408 free'd
==16087==    at 0x4C2A84C: operator delete(void*) (in /usr/lib64/valgrind/vgpreload_memcheck-amd64-linux.so)
==16087==    by 0x62A4869: CEC::CCECTV::~CCECTV() (CECTV.h:42)
==16087==    by 0x62A125F: CEC::CCECDeviceMap::Clear() (CECDeviceMap.cpp:129)
==16087==    by 0x62A0FFC: CEC::CCECDeviceMap::~CCECDeviceMap() (CECDeviceMap.cpp:84)
==16087==    by 0x62A1061: CEC::CCECDeviceMap::~CCECDeviceMap() (CECDeviceMap.cpp:85)
==16087==    by 0x6281D19: CEC::CCECProcessor::~CCECProcessor() (CECProcessor.cpp:108)
==16087==    by 0x6281E29: CEC::CCECProcessor::~CCECProcessor() (CECProcessor.cpp:109)
==16087==    by 0x628C72C: CEC::CLibCEC::~CLibCEC() (LibCEC.cpp:66)
==16087==    by 0x628C7DF: CEC::CLibCEC::~CLibCEC() (LibCEC.cpp:67)
==16087==    by 0x628E0C5: CECDestroy (LibCEC.cpp:497)
==16087==    by 0x4029F4: UnloadLibCec(CEC::ICECAdapter*) (cecloader.h:164)
==16087==    by 0x407A41: main (main.cpp:1323)
==16087== 
==16087== Invalid read of size 4
==16087==    at 0x5043244: pthread_mutex_lock (in /lib64/libpthread-2.19.so)
==16087==    by 0x62864CB: PLATFORM::CMutex::Lock() (mutex.h:90)
==16087==    by 0x62865E9: PLATFORM::CLockObject::CLockObject(PLATFORM::CMutex&, bool) (mutex.h:135)
==16087==    by 0x62991D0: CEC::CCECBusDevice::ReplaceHandler(bool) (CECBusDevice.cpp:180)
==16087==    by 0x62848DE: CEC::CCECProcessor::RegisterClient(CEC::CCECClient*) (CECProcessor.cpp:824)
==16087==    by 0x6292C31: CEC::CCECClient::ChangeDeviceType(CEC::cec_device_type, CEC::cec_device_type) (CECClient.cpp:457)
==16087==    by 0x62AD6DC: CEC::CVLCommandHandler::InitHandler() (VLCommandHandler.cpp:92)
==16087==    by 0x6299675: CEC::CCECBusDevice::ReplaceHandler(bool) (CECBusDevice.cpp:240)
==16087==    by 0x628250F: CEC::CCECProcessor::ReplaceHandlers() (CECProcessor.cpp:238)
==16087==    by 0x6282726: CEC::CCECProcessor::Process() (CECProcessor.cpp:271)
==16087==    by 0x6286C02: PLATFORM::CThread::ThreadHandler(void*) (threads.h:66)
[... 44 lines cut ...]
==16087==    by 0x62865E9: PLATFORM::CLockObject::CLockObject(PLATFORM::CMutex&, bool) (mutex.h:135)
==16087==    by 0x62991D0: CEC::CCECBusDevice::ReplaceHandler(bool) (CECBusDevice.cpp:180)
[... cut ...]
==16087==    by 0x62A1061: CEC::CCECDeviceMap::~CCECDeviceMap() (CECDeviceMap.cpp:85)
==16087==    by 0x6281D19: CEC::CCECProcessor::~CCECProcessor() (CECProcessor.cpp:108)
==16087==    by 0x6281E29: CEC::CCECProcessor::~CCECProcessor() (CECProcessor.cpp:109)
==16087==    by 0x628C72C: CEC::CLibCEC::~CLibCEC() (LibCEC.cpp:66)
==16087==    by 0x628C7DF: CEC::CLibCEC::~CLibCEC() (LibCEC.cpp:67)
==16087==    by 0x628E0C5: CECDestroy (LibCEC.cpp:497)
==16087==    by 0x4029F4: UnloadLibCec(CEC::ICECAdapter*) (cecloader.h:164)
==16087==    by 0x407A41: main (main.cpp:1323)
</pre>

opdenkamp, 2015-03-15:
interesting, haven't run into this yet. since you know how to use a debugger, could you run cec-client in gdb, reproduce the hang in the scan command after the tv started to respond, then do ctrl+c and show me the output of `thread apply all bt`.

it's a bit odd that your tv doesn't respond to polls at all if it's not powered up. I guess that's some deep sleep feature that's enabled in it's settings. because of that, you can't power up the TV via CEC.

UEckhardt, 2015-03-16:
The TV is powered off via a remote controlled power strip. The problem occurs when I gave the remote controlled power strip the command to switch on power for the TV.  So to reproduce the problem you have to remove and insert the power plug. Hoplefully this explanations is somewhat clearer, I am not a native english speaker.

I will run gdb in the evening, when I am at home and let you know the results.

UEckhardt, 2015-03-16:
I tried to reproduce the hang in gdb, but got crashes instead. Also one wired thing is, that after a scan the information about the CEC adapter itselve is not displayed anymore.

The scan with the TV disconnected from power is as expected:
<pre>
CEC bus information
===================
device #1: Recorder 1
address:       1.0.0.0
active source: no
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng
</pre>
Now i connect the TV to the power:
<pre>
CEC bus information
===================
device #0: TV
address:       0.0.0.0
active source: no
vendor:        Panasonic
osd string:    TV
CEC version:   unknown
power status:  unknown
language:      ???

currently active source: unknown (-1)
</pre>
Now device 1 is not displayed anymore. After enter the quit command the cec-client crashes:
<pre>
rogram received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffef7fe700 (LWP 14335)]
0x000000000061b910 in ?? ()
Missing separate debuginfos, use: zypper install libX11-6-debuginfo-1.6.2-5.1.2.x86_64 libXau6-debuginfo-1.0.8-5.1.2.x86_64 libXext6-debuginfo-1.3.3-2.1.2.x86_64 libXrandr2-debuginfo-1.4.2-4.1.2.x86_64 libXrender1-debuginfo-0.9.8-4.1.2.x86_64 libgcc_s1-debuginfo-4.8.3+r212056-2.2.4.x86_64 libstdc++6-gcc49-debuginfo-4.9.0+r211729-2.1.7.x86_64 libudev1-debuginfo-210-25.12.1.x86_64 libxcb1-debuginfo-1.11-2.1.2.x86_64
(gdb) thread apply all bt

Thread 6 (Thread 0x7fffef7fe700 (LWP 14335)):
#0  0x000000000061b910 in ?? ()
#1  0x00007ffff6b9194d in CEC::CCECProcessor::RegisterClient (this=0x616930, client=client@entry=0x61ac60)
    at CECProcessor.cpp:823
#2  0x00007ffff6ba345d in CEC::CCECClient::ChangeDeviceType (this=0x61ac60, from=<optimized out>, to=<optimized out>)
    at CECClient.cpp:457
#3  0x00007ffff6bba694 in CEC::CVLCommandHandler::InitHandler (this=0x7fffe8000c60) at implementations/VLCommandHandler.cpp:92
#4  0x00007ffff6baa1cd in CEC::CCECBusDevice::ReplaceHandler (this=0x616ce0, bActivateSource=<optimized out>)
    at devices/CECBusDevice.cpp:240
#5  0x00007ffff6b8e117 in CEC::CCECProcessor::ReplaceHandlers (this=this@entry=0x616930) at CECProcessor.cpp:237
#6  0x00007ffff6b9348e in CEC::CCECProcessor::Process (this=0x616930) at CECProcessor.cpp:270
#7  0x00007ffff6b950d3 in PLATFORM::CThread::ThreadHandler (_thread=0x616930) at platform/threads/threads.h:66
#8  0x00007ffff79c30a4 in start_thread () from /lib64/libpthread.so.0
#9  0x00007ffff6ed206d in clone () from /lib64/libc.so.6

Thread 1 (Thread 0x7ffff7fca740 (LWP 14327)):
#0  0x00007ffff79c705f in pthread_cond_wait@@GLIBC_2.3.2 () from /lib64/libpthread.so.0
#1  0x00007ffff6b94a87 in Wait (mutex=..., this=0x616940) at platform/threads/../posix/os-threads.h:111
#2  Wait (iTimeoutMs=<optimized out>, mutex=..., this=0x616940) at platform/threads/../posix/os-threads.h:117
#3  Wait (iTimeout=0, param=0x61693a, callback=0x7ffff6b94d40 <PLATFORM::CCondition<bool>::_PredicateCallbackDefault(void*)>, 
    mutex=..., this=0x616940) at platform/threads/mutex.h:274
#4  Wait (iTimeout=0, predicate=@0x61693a: false, mutex=..., this=0x616940) at platform/threads/mutex.h:282
#5  StopThread (iWaitMs=0, this=0x616930) at platform/threads/threads.h:125
#6  ~CThread (this=0x616930, __in_chrg=<optimized out>) at platform/threads/threads.h:49
#7  CEC::CCECProcessor::~CCECProcessor (this=0x616930, __in_chrg=<optimized out>) at CECProcessor.cpp:103
#8  0x00007ffff6b94cc9 in CEC::CCECProcessor::~CCECProcessor (this=0x616930, __in_chrg=<optimized out>)
    at CECProcessor.cpp:109
#9  0x00007ffff6b9958f in CEC::CLibCEC::~CLibCEC (this=0x6168c0, __in_chrg=<optimized out>) at LibCEC.cpp:66
#10 0x00007ffff6b99679 in CEC::CLibCEC::~CLibCEC (this=0x6168c0, __in_chrg=<optimized out>) at LibCEC.cpp:67
#11 0x0000000000403889 in UnloadLibCec (device=0x6168c0) at ../../include/cecloader.h:164
#12 0x0000000000402bc3 in main (argc=<optimized out>, argv=<optimized out>) at main.cpp:1323
</pre>

opdenkamp, 2015-03-17:
@UEckhardt I think that the commits linked in here now will fix the issue, but I need to test this still. If you want to try, see the cmake branch. As the name of the branch suggests, we switched from autotools/visual studio to cmake in there.

UEckhardt, 2015-03-17:
I tried to compile the cmake branch on a OpenSUSE 13.2 (x86_64) but got error messages. I am not sure if i have compiled it correctly or if there are some packages missing:
<b>vdr:/video/src/libcec/build # cmake ../libcec</b>
<pre>
CMake Error at src/cec-client/CMakeLists.txt:21 (find_package):
  By not providing "Findplatform.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "platform",
  but CMake did not find one.

  Could not find a package configuration file provided by "platform" with any
  of the following names:

    platformConfig.cmake
    platform-config.cmake

  Add the installation prefix of "platform" to CMAKE_PREFIX_PATH or set
  "platform_DIR" to a directory containing one of the above files.  If
  "platform" provides a separate development package or SDK, be sure it has
  been installed.
</pre>
I couldn't find the mentioned missing cmake files for OpenSUSE on a first quick search.

bimsarck, 2015-03-18:
same compile error.(crosscompile for raspberrypi)

@UEckardt, wende die beiden patche auf den master an! Musst dazu nur die Pfade anpassen im Patch(src/libcec --> src/lib). Hab auch nen Panasonic, aber nach dem patchen, gleiches Verhalten. Nach einigen Schaltvorgängen zwischen TV und RPi funktioniert die TV FB nicht mehr. Die funktioniert erst wieder wenn ich vorher im VieraLink Menu die RPi(Player2[HDMI2]) Quelle auswähle. Im OSMC Forum gibt es nen Thread dazu --> https://discourse.osmc.tv/t/fixed-osmc-alpha-1-4-rpi-cec-issue/814
Das ist keine Lösung. Laut OSMC-Entwickler wollte opdenkamp das fixen. :-)

bimsarck

opdenkamp, 2015-03-20:
the error is caused by a missing platform lib. you can find it here: https://github.com/Pulse-Eight/platform
readme files haven't been updated yet to include all the new stuff (cmake, swig, python)

UEckhardt, 2015-03-22:
I get a step forward, but i still can compile. I get now the following errors:
<pre>
[ 88%] Swig source
[ 91%] [ 94%] Building CXX object src/cec-client/CMakeFiles/cec-client.dir/cec-client.cpp.o
Building C object src/cecc-client/CMakeFiles/cecc-client.dir/cecc-client.c.o                            
Linking C executable cecc-client
/video/src/libcec/libcec/src/libcec/libcec.i:2: Warning 124: Specifying the language name in %typemap is deprecated - use #ifdef SWIG<LANG> instead.
[ 94%] Built target cecc-client
/video/src/libcec/libcec/src/libcec/libcec.i:10: Warning 450: Deprecated typemap feature ($source/$target).
/video/src/libcec/libcec/src/libcec/libcec.i:10: Warning 450: The use of $source and $target in a typemap declaration is deprecated.
For typemaps related to argument input (in,ignore,default,arginit,check), replace
$source by $input and $target by $1.   For typemaps related to return values (out,
argout,ret,except), replace $source by $1 and $target by $result.  See the file
Doc/Manual/Typemaps.html for complete details.
Linking CXX executable cec-client
[ 97%] [ 97%] Building CXX object src/libcec/CMakeFiles/_cec.dir/LibCECC.cpp.o
Swig source
[ 97%] Built target cec-client
/video/src/libcec/libcec/src/libcec/libcec.i:2: Warning 124: Specifying the language name in %typemap is deprecated - use #ifdef SWIG<LANG> instead.
/video/src/libcec/libcec/src/libcec/libcec.i:10: Warning 450: Deprecated typemap feature ($source/$target).
/video/src/libcec/libcec/src/libcec/libcec.i:10: Warning 450: The use of $source and $target in a typemap declaration is deprecated.
For typemaps related to argument input (in,ignore,default,arginit,check), replace
$source by $input and $target by $1.   For typemaps related to return values (out,
argout,ret,except), replace $source by $1 and $target by $result.  See the file
Doc/Manual/Typemaps.html for complete details.
[100%] Building C object src/libcec/CMakeFiles/_cec.dir/libcecPYTHON_wrap.c.o
Linking CXX shared module _cec.so
/usr/lib64/gcc/x86_64-suse-linux/4.8/../../../../x86_64-suse-linux/bin/ld: /usr/lib64/libplatform.a(fstrcmp.c.o): relocation R_X86_64_PC32 against undefined symbol `strlen@@GLIBC_2.2.5' can not be used when making a shared object; recompile with -fPIC
/usr/lib64/gcc/x86_64-suse-linux/4.8/../../../../x86_64-suse-linux/bin/ld: final link failed: Bad value
collect2: Fehler: ld gab 1 als Ende-Status zurück
src/libcec/CMakeFiles/_cec.dir/build.make:120: recipe for target 'src/libcec/_cec.so.2.3.0' failed
make[2]: *** [src/libcec/_cec.so.2.3.0] Error 1
CMakeFiles/Makefile2:177: recipe for target 'src/libcec/CMakeFiles/_cec.dir/all' failed
make[1]: *** [src/libcec/CMakeFiles/_cec.dir/all] Error 2
Makefile:117: recipe for target 'all' failed
make: *** [all] Error 2
</pre>

opdenkamp, 2015-03-22:
it's trying to link a static platform lib without `-fPIC` from the looks of it. try recompiling the platform lib, and use `cmake /path/to/platform/sources -DBUILD_SHARED_LIBS=1` instead

UEckhardt, 2015-03-24:
With the latest version (commit 7870cf7660) I got the same hang but on directly on the first attempt. I had my TV disconnected from power, performed a scan command, which works as expected, then connected the TV to the power and again performed a scan command which hangs.

The gdb stacktrace:
```
(gdb) thread apply all bt

Thread 7 (Thread 0x7f03a517f700 (LWP 3216)):
#0  0x00007f03a71be408 in pthread_cond_timedwait@@GLIBC_2.3.2 () from /lib64/libpthread.so.0
#1  0x000000000040ba09 in PLATFORM::CConditionImpl::Wait(pthread_mutex_t&, unsigned int) ()
#2  0x000000000040ca52 in PLATFORM::CCondition<bool>::Wait(PLATFORM::CMutex&, bool (*)(void*), void*, unsigned int) ()
#3  0x000000000040c8e3 in PLATFORM::CCondition<bool>::Wait(PLATFORM::CMutex&, bool&, unsigned int) ()
#4  0x00007f03a619a21e in PLATFORM::SyncedBuffer<CEC::CCECAdapterMessageQueueEntry*>::Pop(CEC::CCECAdapterMessageQueueEntry*&, int) () from /usr/lib64/libcec.so.2.0
#5  0x00007f03a61992a7 in CEC::CCECAdapterMessageQueue::Process() () from /usr/lib64/libcec.so.2.0
#6  0x000000000040c236 in PLATFORM::CThread::ThreadHandler(void*) ()
#7  0x00007f03a71ba0a4 in start_thread () from /lib64/libpthread.so.0
#8  0x00007f03a64c506d in clone () from /lib64/libc.so.6

Thread 6 (Thread 0x7f03a497e700 (LWP 3217)):
#0  0x00007f03a71be408 in pthread_cond_timedwait@@GLIBC_2.3.2 () from /lib64/libpthread.so.0
#1  0x000000000040ba09 in PLATFORM::CConditionImpl::Wait(pthread_mutex_t&, unsigned int) ()
#2  0x000000000040ca52 in PLATFORM::CCondition<bool>::Wait(PLATFORM::CMutex&, bool (*)(void*), void*, unsigned int) ()
#3  0x000000000040c8e3 in PLATFORM::CCondition<bool>::Wait(PLATFORM::CMutex&, bool&, unsigned int) ()
#4  0x000000000040c55d in PLATFORM::CThread::Sleep(unsigned int) ()
#5  0x00007f03a61951fa in CEC::CUSBCECAdapterCommunication::Process() () from /usr/lib64/libcec.so.2.0
#6  0x000000000040c236 in PLATFORM::CThread::ThreadHandler(void*) ()
#7  0x00007f03a71ba0a4 in start_thread () from /lib64/libpthread.so.0
#8  0x00007f03a64c506d in clone () from /lib64/libc.so.6

Thread 5 (Thread 0x7f039ffff700 (LWP 3218)):
#0  0x00007f03a71be408 in pthread_cond_timedwait@@GLIBC_2.3.2 () from /lib64/libpthread.so.0
#1  0x000000000040ba09 in PLATFORM::CConditionImpl::Wait(pthread_mutex_t&, unsigned int) ()
#2  0x000000000040ca52 in PLATFORM::CCondition<bool>::Wait(PLATFORM::CMutex&, bool (*)(void*), void*, unsigned int) ()
#3  0x000000000040c8e3 in PLATFORM::CCondition<bool>::Wait(PLATFORM::CMutex&, bool&, unsigned int) ()
#4  0x00007f03a6196e93 in CEC::CAdapterEepromWriteThread::Process() () from /usr/lib64/libcec.so.2.0
#5  0x000000000040c236 in PLATFORM::CThread::ThreadHandler(void*) ()
---Type <return> to continue, or q <return> to quit--- 
#6  0x00007f03a71ba0a4 in start_thread () from /lib64/libpthread.so.0                                    
#7  0x00007f03a64c506d in clone () from /lib64/libc.so.6                                                   
                                                                                                            
Thread 4 (Thread 0x7f039f7fe700 (LWP 3219)):                                                                 
#0  0x00007f03a71be408 in pthread_cond_timedwait@@GLIBC_2.3.2 () from /lib64/libpthread.so.0                   
#1  0x000000000040ba09 in PLATFORM::CConditionImpl::Wait(pthread_mutex_t&, unsigned int) ()                        
#2  0x000000000040ca52 in PLATFORM::CCondition<bool>::Wait(PLATFORM::CMutex&, bool (*)(void*), void*, unsigned int) ()
#3  0x000000000040c8e3 in PLATFORM::CCondition<bool>::Wait(PLATFORM::CMutex&, bool&, unsigned int) ()                 
#4  0x000000000040c55d in PLATFORM::CThread::Sleep(unsigned int) ()                                                   
#5  0x00007f03a6196ce0 in CEC::CAdapterPingThread::Process() () from /usr/lib64/libcec.so.2.0                           
#6  0x000000000040c236 in PLATFORM::CThread::ThreadHandler(void*) ()                                                      
#7  0x00007f03a71ba0a4 in start_thread () from /lib64/libpthread.so.0                                                        
#8  0x00007f03a64c506d in clone () from /lib64/libc.so.6                                                                          
                                                                                                                                  
Thread 3 (Thread 0x7f039effd700 (LWP 3220)):                                                                                      
#0  0x00007f03a71be408 in pthread_cond_timedwait@@GLIBC_2.3.2 () from /lib64/libpthread.so.0                                         
#1  0x000000000040ba09 in PLATFORM::CConditionImpl::Wait(pthread_mutex_t&, unsigned int) ()                                          
#2  0x000000000040c9a8 in PLATFORM::CCondition<bool volatile>::Wait(PLATFORM::CMutex&, bool (*)(void*), void*, unsigned int) ()          
#3  0x000000000040c835 in PLATFORM::CCondition<bool volatile>::Wait(PLATFORM::CMutex&, bool volatile&, unsigned int) ()                   
#4  0x000000000040bf64 in PLATFORM::CEvent::Wait(unsigned int) ()                                                                            
#5  0x000000000040bff9 in PLATFORM::CEvent::Sleep(unsigned int) ()                                                                              
#6  0x00007f03a6183896 in CEC::CCECProcessor::RegisterClient(std::shared_ptr<CEC::CCECClient>) ()                                               
[... cut ...]
DEBUG:   [           30751]     Recorder 1 (1): CEC version unknown
DEBUG:   [           30751]     Recorder 1 (1): osd name set to 'Recorder 1'
DEBUG:   [           30751]     expected response received (82: active source)
DEBUG:   [           30751]     Recorder 1 (1): device status changed into 'unknown'
DEBUG:   [           30751]     << requesting CEC version of 'TV' (0)
DEBUG:   [           30751]     setting ackmask to  0
DEBUG:   [           30755]     turning controlled mode off
DEBUG:   [           30761]     turning controlled mode on
```

opdenkamp, 2015-04-08:
should be solved in libCEC 3.0.0

UEckhardt, 2015-04-11:
Thanks, works fine now with 3.0.0

closed 2015-04-11 by UEckhardt