Pulse-Eight/libcec #66: libcec and philips tv

thor2002ro, 2015-03-21:
theres an issue with the cec on the  philips tv... it doesn't repeat remote buttons when you keep them pressed... holding buttons to scroll in the tv menus work fine..
this happened on the raspberry pi/pi2 and now I got a usb pulse-eight adapter and the same happens... 
is there anything that can be done? 
if you guys need any log please tell me how to get it and I'll gladly dump it

opdenkamp, 2015-04-03:
this is an issue of the TV's CEC implementation. it only sends key up + down repeating, and not very fast. nothing we can do about this from our side, sorry

closed 2015-04-03 by opdenkamp

Kroellie, 2015-05-14:
Hi, I've notices with the cec-client, that my Philips remote does in fact send repeated commands, I can see this scroll by when holding a button down. Though Kodi does not accept this input. What can I do to further troubleshoot this?

Debug log cec-client:
DEBUG:   [           31510]	>> TV (0) -> Recorder 1 (1): user control pressed (44)
TRAFFIC: [           31789]	>> 01:44:02
DEBUG:   [           31789]	>> TV (0) -> Recorder 1 (1): user control pressed (44)
TRAFFIC: [           32067]	>> 01:44:02
DEBUG:   [           32067]	>> TV (0) -> Recorder 1 (1): user control pressed (44)
TRAFFIC: [           32346]	>> 01:44:02
DEBUG:   [           32346]	>> TV (0) -> Recorder 1 (1): user control pressed (44)
TRAFFIC: [           32625]	>> 01:44:02
DEBUG:   [           32625]	>> TV (0) -> Recorder 1 (1): user control pressed (44)
TRAFFIC: [           32897]	>> 01:44:02
DEBUG:   [           32897]	>> TV (0) -> Recorder 1 (1): user control pressed (44)

Thanks.