Pulse-Eight/libcec #67: Compatibility with "HDMI CEC framework" support for the Linux kernel?

Hedda, 2015-03-24:
HDMI CEC Framework project have been revived for the Linux Kernel, read:

http://lists.freedesktop.org/archives/dri-devel/2015-March/079678.html

http://www.phoronix.com/scan.php?page=news_item&px=HDMI-CEC-Framework-Linux

http://www.phoronix.com/scan.php?page=news_item&px=MTIwMTk

http://lists.freedesktop.org/archives/dri-devel/2012-October/028716.html

Question is if libCEC will be supported through this framework too?

This new CEC API uses a CEC bus with CEC clients.

Guess this means that LibCEC could be a CEC client usable via this API?

PS: Looks like this "HDMI CEC framework" is sponsored by Samsung.

At least the main develop is employee "Kamil Debski"<k.debski@samsung.com>

http://lists.freedesktop.org/archives/dri-devel/2015-March/thread.html#79678
http://lists.freedesktop.org/archives/dri-devel/2015-March/079679.html
http://lists.freedesktop.org/archives/dri-devel/2015-March/079680.html
http://lists.freedesktop.org/archives/dri-devel/2015-March/079681.html
http://lists.freedesktop.org/archives/dri-devel/2015-March/079682.html
http://lists.freedesktop.org/archives/dri-devel/2015-March/079683.html
http://lists.freedesktop.org/archives/dri-devel/2015-March/079684.html
http://lists.freedesktop.org/archives/dri-devel/2015-March/079685.html
http://lists.freedesktop.org/archives/dri-devel/2015-March/079686.html
http://lists.freedesktop.org/archives/dri-devel/2015-March/079687.html

opdenkamp, 2015-03-24:
I've seen the discussion on the kernel thread and I've signed up to the mailing list so I can respond, as soon as someone posts something there. Couldn't find a proper way to comment to it when not signed up last week end, but might have missed it :)

I've taken a look at the implementation, and half of it is good: the promiscuous mode that got added, which will indeed allow interfacing with libCEC if done right. Patches to support it are welcome once it enters the kernel.

The other half, trying to process key presses directly in the kernel, not so much:
* it simply does not work with half the TVs on the planet because of how those implemented CEC. Simple example: some common brand requires you to respond with their vendor id or refuse to interoperate.
* doing some way to initialise things in the kernel and then doing something similar in userspace will confuse certain brands of TVs. Another simple example: using a different OSD name or logical address will make two devices show up in the list of CEC devices on another common brand, and only one of them will work, confusing users
* In my opinion, it just does not belong in the kernel, as there's no good reason to do it there when you can do it better in userspace. Like through libCEC which then sends back keys through udev, as suggested by https://github.com/Pulse-Eight/libcec/issues/60 . Kernel support for CEC is just bloat that won't work anyway, as pointed out in the above points.

So, in short, "promiscuous mode" as they call it, is good, if it only does the following:
* allocate one or more logical addresses, and ack CEC traffic sent to those logical addresses
* receive CEC traffic and forward it to userspace
* transmit CEC traffic initiated by userspace

This is the same as the implementations for the Raspberry Pi, TDA995x and Exynos (the current implementation) do.

All the rest should not be done in the kernel.

And if someone who's subscribed to the linux-media mailing list (@anssih or @fernetmenta perhaps?) could post a response to http://lists.freedesktop.org/archives/dri-devel/2015-March/079696.html with this, or tell me how to respond there when I wasn't signed up before the message was posted, then that would be great.

Hedda, 2015-03-25:
@opdenkamp Maybe just mail the main two developers directly with your feedback?

"Kamil Debski" <k.debski@samsung.com>

"Hans Verkuil" <hverkuil@xs4all.nl>

Like it that multiple applications can send CEC commands to the same API/client.

Hedda, 2015-07-09:
FYI, "Hans Verkuil" hans.verkuil@cisco.com is now on his seventh revision of this patch series

https://patchwork.linuxtv.org/patch/30412/

Hedda, 2016-04-11:
FYI, Hans Verkuil is still working on this and is now on his 13th version of this  patch series:

https://patchwork.linuxtv.org/project/linux-media/list/?submitter=676&archive=both&state=*

The patch is now also devided into seventeen parts.

gdachs, 2016-04-11:
I got aware of  a new cec driver in an unreleased amlogic kernel for android. I will write an adapter for libcec for any implementation I will get hands on first. As my development environment is OpenELEC/LibreELEC and my current hardware is Wetek Play/Core and Odroid C2, I have to wait till it will show up on them.

Hedda, 2016-04-12:
You could try contacting the developers, Hans Verkuil (a.k.a. hverkuil), either mail directly to <hans.verkuil@cisco.com> or via linuxtv.org mailing-list.

I understand that as reference hardware he and Kamil Debski <kamil@wypas.org> is at present using Samsung Exynos range of SoCs for testing this CEC framework.

Guess that means ODROID U & X series from Hardkernel, such as the ODROID-XU4 http://www.hardkernel.com/main/products/prdt_info.php?g_code=G143452239825

Not sure if OpenELEC is already supported on ODROID-XU4 or not(?).

gdachs, 2016-04-12:
If the Exynos driver is the reference, then no activity is necessary as there is already an adapter for Exynos in the libcec. 

OpenELEC doesn't support Odroid U & X. As far as I know the OpenELEC team doesn't own one of this devices.

Hedda, 2016-04-12:
You might want to checkout this v4l2-subdev patch, maybe LibCEC could be implemented through it?

https://patchwork.linuxtv.org/patch/33526/

While Exynos seems to be the reference they still look to be working on other hardware too, see patches: 

https://patchwork.linuxtv.org/project/linux-media/list/?submitter=676&archive=both&state=*

There are seperate CEC adapters for Cobalt, Exynos, Vivid, ADV7511, ADV7604, and ADV7842 drivers.

gdachs, 2016-04-12:
That looks strange, receiving cec messages by ioctl calls? Seems not to have something in common with the old exynos driver. 
Anyway, I don't expect that this will cross my way anytime soon.

Hedda, 2016-10-07:
@opdenkamp  FYI: Linux 4.8 release finally include this HDMI-CEC framework in the mainline kernel.

http://www.phoronix.com/scan.php?page=news_item&px=HDMI-CEC-Linux-4.8

Details on the framework in this PDF:

http://events.linuxfoundation.org/sites/events/files/slides/cec-wide_1.pdf

Full development changelog here:

https://lwn.net/Articles/685756/

Developers contributions here:

http://lkml.iu.edu/hypermail/linux/kernel/1607.3/00427.html

Hedda, 2016-11-16:
Nice video from the Embedded Linux Conference with Hans Verkuil on this HDMI CEC implementation:

https://www.youtube.com/watch?v=Q6S2FabX2WA

glance-, 2017-01-06:
For 4.10, the framework and drivers are moved out of staging.

Is there a plan on how libcec should use/co-exist with this? Or is the plan just to ignore it?

opdenkamp, 2017-01-06:
The plan is to use it's raw mode for libCEC, but that'll be something for the next major release of libCEC

Ekironji, 2017-06-29:
Hi, are there any news about integrating with the Linux CEC Framework? 
Just to know if this will be a very long development or there may be good news in a short time.

opdenkamp, 2017-07-13:
It's going to take some time because unfortunately. Too many other things that need to be done ;)

nkichukov, 2017-11-20:
Hello fellows,

Is there any progress on this new feature request?

This is the only feature that does not work in Kodi on Gentoo Linux on raspberry pi 3B that makes no use of any proprietary broadcom drivers/firmware.

Kodi development seems to have no interest into implementing direct CEC calls in their software to the new kernel framework, thus use of libcec will remain required.

I can do code testing if needed.

Cheers,
-Nik

Kwiboo, 2018-01-22:
@Borck there is an open PR/RFC at #380 that includes that commit, the cec-framework adapter is used to enable CEC support in LibreELEC on Rockchip devices.

nkichukov, 2018-01-23:
Excellent, I have tested it and it does implement the required functionality successfully! See:
https://github.com/Pulse-Eight/libcec/pull/380#issuecomment-359942517 for more details.

Hedda, 2019-01-07:
@opdenkamp any more thoughts on the Linux kernel's HDMI CEC framework and an libcec adapter?

opdenkamp, 2020-04-27:
merged now https://github.com/Pulse-Eight/libcec/pull/380
thanks!

Hedda, 2020-04-27:
@opdenkamp Very cool! Does that mean that we will see a libcec 5.0.x release very soon?

opdenkamp, 2020-04-27:
yes, but there will also be a 4.x release that includes it

closed 2020-04-27 by opdenkamp