Pulse-Eight/libcec #71: Source is not activated when receiver switch to ARC after switch on

PavelKuzub, 2015-03-29:
Hello Pulse Eight team,
Hello Lars,

I had this problem for a long time with my equipment and I have finally narrowed down scenario how to reproduce the problem.

Hardware setup:
HTPC > Pulse Eight > Receiver: Sony STR-DN1030 > TV: Samsung ES7500

AV receiver has a setting "Ctrl for HDMI" that when set to ON enables both these features at the same time: 1) Control AV from TV (volume, power, etc) 2) AV receiver can switch to Audio Return Channel (ARC) and stream the sounds from TV.
Receiver can automatically switch between HTPC source or TV source (via ARC).
If "Ctrl for HDMI" is set to OFF - TV cannot control volume of AV receiver, AV receiver will never switch to TV as a source from it's active source.

Description of problem:
Upon turning on TV and AV receiver - TV remote is not working for CEC device until CEC source is selected manually either on TV or on AV receiver.

Steps on how to reproduce the problem:
TV is switched from OFF to ON.
AV receiver is switched from OFF to ON.
XBMC / Kodi / libcec-daemon detects power on and sends signal to activate defined HDMI source on AV receiver and TV.
TV is yet not receiving signal as AV receiver is booting up.
AV receiver finish loading and enables signal from HTPC to TV.
TV starts displaying Video signal from HTPC.
(!!!) AV receiver change the source from BD/DVD (HTPC source) to TV (via ARC) and Kodi / libcec does not detect this event (!!!).
Kodi source is listed in TV sources, but it is NOT selected. HDMI1 source remains selected.
TV Remote key presses are NOT delivered to Kodi / libcec.

Manual workaround:
I have to change source on TV from HDMI1 to Kodi manually OR change the source on AV receiver from TV to BD/DVD manually - only then TV remote key presses start to work. 
Once this is done - if I select HDMI1 on TV - it will anyway go to Kodi as source. 
Also if I select TV on AV receiver - TV remote key presses continue being registered by Kodi / libcec.
Have to apply a manual workaround on every power on is very annoying. For my wife especially.

If I disable "Ctrl for HDMI" on AV receiver - then there is no problem - after TV and AV receiver are turned ON - source is activated and TV remote keys are sent to Kodi / libcec just fine all the time.
I cannot control the volume of AV receiver from TV remote, though, so this is not a solution for me.

So the problem is initial registration / activation of source that is distracted by ARC switching to TV, when auto switch to ARC is enabled.
libcec does not detect event when AV receiver automatically switch the source to TV and continue believes that that required sources are active, but the CEC chain is not active.

Why AV receiver is switching to TV as a sound source is not clear to me. 
My key suspects: 1) When libcec sends CEC message to TV on what HDMI channel it should switch - it triggers AV receiver to believe that TV became the source and switch to TV as being the most recent source; 2) TV connects to AV receiver to enable volume control and this may trigger AV receiver to think that TV should be the source of audio.

Unless there is actually a way to detect this event by libcec OR to poll that chain HTPC > TV is not active anymore - the solution may be just another force activation after certain time.

Looking forward your attention and solution to this scenario that got my wife pissed off badly already :)
Thank you! Happy Wife - Happy Life!

Pavel

jolan, 2015-04-01:
Hi,

Was just discussing this on the kodi forums with someone.  We both have Chromeboxes attached to Onkyo receivers attached to Panasonic TVs and have the same issue:

http://forum.kodi.tv/showthread.php?tid=223005

Will upgrade to libcec master and see if the issue is still present and if so get debug info.

mk01, 2015-06-07:
@PavelKuzub 

this problem is with xbmc's peripheralcecdevice implementation, not libcec (specifically SetInitialConfiguration).
I will try to catch @opdenkamp 's attention on this again.

opdenkamp, 2015-06-08:
again? must have missed the first time :P

opdenkamp, 2015-07-03:
closing this ticket, as it's not an issue in libCEC but in Kodi, and this tracker is only for libCEC issues.
please create a ticket on Kodi's Trac if there is none yet. I'll check Kodi's integration with libCEC after I finished the 3.0.1 build

yasij, 2017-01-06:
Did this ever get a Kodi trac bug? It's still happening for me in Kodi 17.0rc2 with libcec 4.0.1.

opdenkamp, 2017-01-07:
completely forgot about this one tbh, don't think so.