Pulse-Eight/libcec #72: libcec install directions using cmake fail - rpi

tvideovi, 2015-04-03:
HI. I tried to install libcec on a rpi 2 today using the exact directions which seem new as they use cmake rather than the previous .bootstrap and ,config protocol.

I installed cmake and ran cmake .. in the build directory as instructed but received error response:

By not providing "Findplatform.cmake" in CMAKE_MODULE_PATH this project has asked CMake to find a package configuration file provided by "platform" , but CMake did not find one.

Could not find a package configuration file provided by "platform" with any of the following names:
platformConfig.cmake
platform-config.cmake

Please advise and thanks.

@videovi

opdenkamp, 2015-04-08:
you have to install the platform lib first: https://github.com/Pulse-Eight/platform

closed 2015-04-08 by opdenkamp

tvideovi, 2015-04-08:
Thanks for the response but I still have a failure.

I cloned libcec and platform in that order but the message is the same.

By not providing "Findplatform.cmake" in CMAKE_MODULE_PATH this 
project has asked CMake to find a package configuration file provided by
 "platform" , but CMake did not find one.

Could not find a package configuration file provided by "platform" with any of the following names:

platformConfig.cmake

platform-config.cmake

I note there is a platformConfig.cmake file in libcec/cmake.

When I use the instructions and cmake .. in the /home/libcec/build directory I get the messages above.

I recall having to set some flags on the compile using the previous method to note that the build was for the RPi--am I missing a step somewhere in the cmake process?

I am not familiar with cmake so follow your direction but perhaps I am missing something obvious.

Thanks for you assistance,

  

Date: Wed, 8 Apr 2015 02:28:58 -0700

opdenkamp, 2015-04-08:
you have to provide the path to where you installed the platform lib to cmake

tvideovi, 2015-04-08:
Thank you.

This makes sense but I could not see anything in the documentation that advised where.

How do I do this? -- do I add a flag to the cmake .. instruction or somewhere else?

Your support is much appreciated.

Date: Wed, 8 Apr 2015 09:03:03 -0700

andreikun, 2015-04-17:
Please give us more details about where we should provide the path to platform lib installation. I searched a lot in the packages and wasn't able to find any solution yet. Please provide us some guidance about how we should include platform in order to make cmake work.

Thanks.

tvideovi, 2015-04-17:
Yes, please assist those wishing to build libcec for the Rpi as I understand the intent of a platform designation so that the correct platform .h files are included but in the past this was set with flags that designated the target (e.g. cubox, rpi) which then established the proper platform parameters for the build.

It seems the solution is to provide guidance on how to set the platform flags or parameters to assure that the correct platform functions are included in the build be it targeted to the cubox or rpi for example.

If my comments are confusing please ask for clarification as I feel if the new cmake instructions can be more detailed regarding the build steps especially as the new cmake protocol is different than the past ./bootstrap ./config and make methods.

Therefore the proper location and build flags (or whatever indicators define the target device) would be very much appreciated.

For example, in my case I build for both the cubox and the rpi and have no idea the differences and full process to build for either target.

Thank you in advance and also my complements on developing libcec and the supporting programs like cec-client. 

Date: Fri, 17 Apr 2015 04:31:22 -0700