Pulse-Eight/libcec #74: Using LibCecSharp.dll in C#-Code

mattbue, 2015-04-07:
I'm trying to use the LibCecSharp.dll in one of my own C#-applications.
Therefore i added the DLL into the references and i created some code based on the CecSharpTester which is located in the src folder.

But all time i'm trying to create my project i get an error message:

Zusätzliche Informationen: Die Datei oder Assembly "LibCecSharp.dll" oder eine Abhängigkeit davon wurde nicht gefunden.  ist keine zulässige Win32-Anwendung. (Ausnahme von HRESULT: 0x800700C1)

Translated this means something like the LibCecSharp.dll or one of its dependencies can not be found...

Does someone know the reason of this error message?

Thanks!

opdenkamp, 2015-04-08:
you have to put libcec.dll in the same directory as LibCecSharp.dll

closed 2015-04-08 by opdenkamp

Slion, 2015-09-24:
I got both DLL and I keep getting this error message too. Now my application is using .NET 4.6.
Could that be the issue?

opdenkamp, 2015-09-25:
Right, thanks for this info. I'll add that to the readme file. We're targeting 2.0 because we don't require any of the newer stuff, and using 2.0 will allow you to use the dll in older applications (like WMC)

BenoitArribart, 2017-11-12:
Hi, I got into the exact same problem.
However, useLegacyV2RuntimeActivationPolicy="true" helped me run the generared exe properly on one computer, whereas on the other I keep getting the "LibCecSharp.dll not found" error, wether the dll is present or not together with other dlls, or not.
How do you reference the dll only, without having it part of your project ? It is only references, as it seems, as opposed to libcec.dll, which is an item of the project, buit it doesn't make any difference.
It would be much easier to have a simple C# wrapper we could recompile on purpose, without having to bother about old 2.0 framework legacy.
I would like to run some tests, and implement simple cec on/off triggered by my application, but none of the provided samples simply work (on two computers, win10 64), and after several hours, I could only manager to get a working test program able to run on one of the 2 computers...
As a comparison, the usb-uirt C# wrapper worked out of the shelves...
I don't want my application to target framework 2.0 by the way... but even doing so, it doesn't seem to work neither....