Pulse-Eight/libcec #76: Source not deactivated when switching to non CEC aware hdmi source

yoyoma2, 2015-04-07:
Setup:

o Toshiba CEC aware TV
    o ubuntu CEC aware PC (pulse-eight adapter + bramp's libcec-daemon)
    o hdmi non CEC aware DVR

Steps:

1) Use the PC on the TV
2) Switch the TV to the DVR
= The PC does not get deactivated by libcec

Workaround:

Before switching to the DVR, switch the TV to a non hdmi input (component or tuner) which deactivates the PC.  Then switch to the DVR.

Possible Explanation:

The logs show that the TV broadcasts a "routing change" when switching to another hdmi input but broadcasts an "active source" when switching to a non-hdmi input.

The libcec Changelog states:
"* don't mark a device as inactive when receiving a routing change message,  only when another device was made acive."

Log file:

INFO - Opened /dev/uinput
INFO - Created uinput device
INFO - Physical Address is set to 0.0.0.0
INFO - Base device is set to 0
INFO - HDMI port is set to 1
INFO - Openning /sys/devices/pci0000:00/0000:00:1d.0/usb6/6-1
DEBUG - Main::onCecLogMessage(21 [D]unregistering all CEC clients)
DEBUG - Main::onCecLogMessage(21 [D]Broadcast (F): osd name set to 'Broadcast')
DEBUG - Main::onCecLogMessage(23 [D]connection opened, clearing any previous input and waiting for active transmissions to end before starting)
DEBUG - Main::onCecLogMessage(23 [D]communication thread started)
DEBUG - Main::onCecLogMessage(90 [D]turning controlled mode on)
DEBUG - Main::onCecLogMessage(111 [N]connection opened)
DEBUG - Main::onCecLogMessage(111 [D]processor thread started)
DEBUG - Main::onCecLogMessage(111 [D]<< Broadcast (F) -> TV (0): POLL)
DEBUG - Main::onCecLogMessage(111 [T]<< f0)
DEBUG - Main::onCecLogMessage(111 [D]setting the line timeout to 3)
DEBUG - Main::onCecLogMessage(147 [D]>> POLL sent)
DEBUG - Main::onCecLogMessage(147 [D]TV (0): device status changed into 'present')
DEBUG - Main::onCecLogMessage(147 [D]<< requesting vendor ID of 'TV' (0))
DEBUG - Main::onCecLogMessage(147 [T]<< f0:8c)
DEBUG - Main::onCecLogMessage(344 [T]>> 0f:87:00:00:39)
DEBUG - Main::onCecLogMessage(344 [D]>> TV (0) -> Broadcast (F): device vendor id (87))
DEBUG - Main::onCecCommand(Command 0->15[AA] 135)
DEBUG - Main::onCecLogMessage(344 [D]TV (0): vendor = Toshiba (000039))
DEBUG - Main::onCecLogMessage(344 [D]expected response received (87: device vendor id))
DEBUG - Main::onCecLogMessage(344 [D]replacing the command handler for device 'TV' (0))
DEBUG - Main::onCecLogMessage(344 [N]registering new CEC client - v2.2.0)
DEBUG - Main::onCecLogMessage(344 [D]detecting logical address for type 'recording device')
DEBUG - Main::onCecLogMessage(344 [D]trying logical address 'Recorder 1')
DEBUG - Main::onCecLogMessage(344 [D]<< Recorder 1 (1) -> Recorder 1 (1): POLL)
DEBUG - Main::onCecLogMessage(344 [T]<< 11)
DEBUG - Main::onCecLogMessage(381 [D]CEC transmission - received response - TRANSMIT_FAILED_ACK)
DEBUG - Main::onCecLogMessage(381 [T]<< 11)
DEBUG - Main::onCecLogMessage(412 [D]CEC transmission - received response - TRANSMIT_FAILED_ACK)
DEBUG - Main::onCecLogMessage(412 [D]>> POLL not sent)
DEBUG - Main::onCecLogMessage(412 [D]using logical address 'Recorder 1')
DEBUG - Main::onCecLogMessage(412 [D]Recorder 1 (1): device status changed into 'handled by libCEC')
DEBUG - Main::onCecLogMessage(412 [D]Recorder 1 (1): power status changed from 'unknown' to 'on')
DEBUG - Main::onCecLogMessage(412 [D]Recorder 1 (1): vendor = Pulse Eight (001582))
DEBUG - Main::onCecLogMessage(412 [D]Recorder 1 (1): CEC version 1.4)
DEBUG - Main::onCecLogMessage(412 [D]AllocateLogicalAddresses - device '0', type 'recording device', LA '1')
DEBUG - Main::onCecLogMessage(412 [D]setting ackmask to  2)
DEBUG - Main::onCecLogMessage(417 [D]Recorder 1 (1): osd name set to 'china')
DEBUG - Main::onCecLogMessage(417 [D]Recorder 1 (1): menu language set to 'eng')
DEBUG - Main::onCecLogMessage(417 [D]GetPhysicalAddress - trying to get the physical address via ADL)
[... 100 lines cut ...]
DEBUG - Main::onCecLogMessage(27607 [T]>> 0f:80:30:00:10:00)
DEBUG - Main::onCecLogMessage(27607 [D]>> TV (0) -> Broadcast (F): routing change (80))
DEBUG - Main::onCecCommand(Command 0->15[AA] 128)
DEBUG - Main::onCecLogMessage(27607 [D]Recorder 1 (1) was already marked as active source)
DEBUG - Main::onCecLogMessage(27607 [N]>> source activated: Recorder 1 (1))
DEBUG - Main::onCecSourceActivated(logicalAddress 1 = 1)
DEBUG - Main::onCecLogMessage(27607 [D]sending active source message for 'Recorder 1')
DEBUG - Main::onCecLogMessage(27607 [N]<< powering on 'TV' (0))
DEBUG - Main::onCecLogMessage(27607 [T]<< 10:04)
DEBUG - Main::onCecLogMessage(27664 [N]<< Recorder 1 (1) -> broadcast (F): active source (1000))
DEBUG - Main::onCecLogMessage(27664 [T]<< 1f:82:10:00)
DEBUG - Main::onCecLogMessage(27773 [D]<< Recorder 1 (1) -> TV (0): menu state 'activated')
DEBUG - Main::onCecLogMessage(27773 [T]<< 10:8e:00)
DEBUG - Main::onCecLogMessage(34896 [T]>> 01)
DEBUG - Main::onCecLogMessage(34896 [D]<< POLL: TV (0) -> Recorder 1 (1))
DEBUG - Main::onCecLogMessage(34985 [T]>> 02)
DEBUG - Main::onCecLogMessage(34985 [D]<< POLL: TV (0) -> Recorder 2 (2))
DEBUG - Main::onCecLogMessage(35073 [T]>> 03)
DEBUG - Main::onCecLogMessage(35073 [D]<< POLL: TV (0) -> Tuner 1 (3))
DEBUG - Main::onCecLogMessage(35157 [T]>> 04)
DEBUG - Main::onCecLogMessage(35157 [D]<< POLL: TV (0) -> Playback 1 (4))
DEBUG - Main::onCecLogMessage(35245 [T]>> 05)
DEBUG - Main::onCecLogMessage(35245 [D]<< POLL: TV (0) -> Audio (5))
DEBUG - Main::onCecLogMessage(35334 [T]>> 06)
DEBUG - Main::onCecLogMessage(35334 [D]<

opdenkamp, 2015-05-03:
iirc this was on purpose because the cec source isn't deactivated. have to check the spec as i don't remember that part exactly

mk01, 2015-06-07:
@opdenkamp 
you remember fine, source can be changed only by TV (or CEC switch) by changing route directly <set stream path>/<routing change>, or by cec device asking to be active source <as>, or by (again) fallback to TV after active source sends inactive source command.
the specs also say (1.3a version) - if non-HDMI source is selected, TV nor switch should generate above messages (I would assume non-HDMI be also HDMI non-CEC). 

@yoyoma2 
this is a "feature" which does ugly things like for instance XBMC turning off TV even if another device (non-CEC aware) is already active.

opdenkamp, 2015-07-02:
yup, that's what the docs say, so we can't include this, sorry.
we could add another callback for routing changes that we detect.