Pulse-Eight/libcec #77: Installing libcec on Raspberry Pi

mattbue, 2015-04-08:
Hi,

i installed libcec on a Raspberry Pi (Raspbian) a few weeks ago by using these commands:

git clone git://github.com/Pulse-Eight/libcec.git
cd libcec
./bootstrap
./configure --with-rpi-include-path=/opt/vc/include --with-rpi-lib-path=/opt/vc/lib --enable-rpi
make
sudo make install
sudo ldconfig

Today i wanted to install libcec on another Raspberry Pi. It seems there have been some changes on the current version of it. I tried to compile be the steps described in the README.md but doesn't work.

Any ideas?
Thanks!

opdenkamp, 2015-04-08:
the build system changed from autotools to cmake. please provide a better description of what's not working exactly. "doesn't work" can be caused by many things ;-)

mattbue, 2015-04-08:
I just figured out that some dependencies weren't installed.

I did the following:
sudo apt-get install cmake

git clone git://github.com/Pulse-Eight/platform.git
cd platform/
mkdir build
cd build/
cmake ..
make
sudo make install

cd  (go to home path)
git clone git://github.com/Pulse-Eight/libcec.git
cd libcec/
mkdir build
cd build/
cmake ..

I got the output:

-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found.
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Looking for initscr in curses
-- Looking for initscr in curses - not found
-- Looking for dlopen in dl
-- Looking for dlopen in dl - found
-- Looking for clock_gettime in rt
-- Looking for clock_gettime in rt - found
-- Could NOT find PythonLibs (missing:  PYTHON_INCLUDE_DIRS) 
-- Could NOT find PythonLibs (missing:  PYTHON_INCLUDE_DIRS) 
-- Looking for include file lockdev.h
-- Looking for include file lockdev.h - found
-- Looking for dev_unlock in lockdev
-- Looking for dev_unlock in lockdev - found
-- Looking for udev_new in udev
-- Looking for udev_new in udev - found
-- Looking for 3 include files X11/Xlib.h, ..., X11/extensions/Xrandr.h
-- Looking for 3 include files X11/Xlib.h, ..., X11/extensions/Xrandr.h - not found.
-- Looking for XRRGetScreenResources in Xrandr
-- Looking for XRRGetScreenResources in Xrandr - not found
-- Looking for vchi_initialise in bcm_host
-- Looking for vchi_initialise in bcm_host - not found
-- Looking for include files tda998x_ioctl.h, comps/tmdlHdmiCEC/inc/tmdlHdmiCEC_Types.h
-- Looking for include files tda998x_ioctl.h, comps/tmdlHdmiCEC/inc/tmdlHdmiCEC_Types.h - not found.
-- Could NOT find PythonLibs (missing:  PYTHON_INCLUDE_DIRS) 
-- Could NOT find PythonLibs (missing:  PYTHON_INCLUDE_DIRS) 
-- Could NOT find SWIG (missing:  SWIG_EXECUTABLE SWIG_DIR) 
-- Configured features:
-- Pulse-Eight CEC Adapter:                yes
-- Pulse-Eight CEC Adapter detection:      yes
-- lockdev support:                        yes
-- xrandr support:                         no
-- Raspberry Pi support:                   no
-- TDA995x support:                        no
-- Exynos support:                         no
-- Python support:                         no
-- lib info: compiled on Linux-3.18.7-v7+, features: P8_USB, P8_detect
-- Configuring done
-- Generating done
-- Build files have been written to: /home/adscreen/libcec/build

Since there is no Raspberry Pi support i think something went wrong...

opdenkamp, 2015-04-08:
you have to provide the path to bcm_host and the development headers to cmake if they're not in the standard library search path on your system

raspiuser, 2015-04-08:
I also can't do make sucessful. I get:

cmake ..
CMake Error at src/cec-client/CMakeLists.txt:21 (find_package):
  By not providing "Findplatform.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "platform",
  but CMake did not find one.

  Could not find a package configuration file provided by "platform" with any
  of the following names:

    platformConfig.cmake
    platform-config.cmake

  Add the installation prefix of "platform" to CMAKE_PREFIX_PATH or set
  "platform_DIR" to a directory containing one of the above files.  If
  "platform" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!

mattbue, 2015-04-09:
@opdenkamp 
since i'm using a system which i just wrote on the raspberrys SD card, i think bcm_host and the development headers are in the standard library search path on my system...

@raspiuser 
you need to install 'platform' from Pulse-Eight first. (https://github.com/Pulse-Eight/platform)

raspiuser, 2015-04-09:
thanks @mattbue 
now im at the same point like you

opdenkamp, 2015-04-10:
jenkins can find the deps, headers and libs just fine:
```
-- Pulse-Eight CEC Adapter:                no
-- Pulse-Eight CEC Adapter detection:      no
-- lockdev support:                        no
-- xrandr support:                         no
-- Raspberry Pi support:                   yes
-- TDA995x support:                        no
-- Exynos support:                         no
-- Python support:                         no
```
so you have to provide the path to vcos and vchiq_arm to cmake

closed 2015-04-10 by opdenkamp

mattbue, 2015-04-10:
Can you provide an example how to call cmake with providing the paths for vcos and vchiq_arm?

Snyder1977, 2015-04-10:
Providing an example would be great. I had the same problems when I tried to build the latest release of libCEC on my PI (Rasbian).

raspiuser, 2015-04-13:
seams not so easy to build at raspberry if nobody knows how.
@opdenkamp Please give a short example

spectrokid, 2015-05-03:
Used an afternoon combining this thread with this:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=104797&start=5
Got the whole thing to compile without complaints on latest raspbian, but I still get the
-- Raspberry Pi support:                   no
And I couldn't get it to run scripts which worked before.  I will try to switch to a distro that supports it out of the box.

opdenkamp, 2015-05-03:
You can use Kodi's `Toolchain.cmake` to set the correct paths:
```
cmake \
 -DCMAKE_TOOLCHAIN_FILE=/path/to/kodi/tools/depends/target/Toolchain.cmake \
 -DCMAKE_INSTALL_PREFIX=/path/to/installdir -DBUILD_SHARED_LIBS=1 \
 -DCMAKE_INSTALL_LIBDIR=/path/to/installdir/lib \
 -DSKIP_PYTHON_WRAPPER:STRING=1 ..
```

spectrokid, 2015-05-05:
Spent over 10 hours on this now.  Tried to download the old version and follow this thread to the letter: https://drgeoffathome.wordpress.com/2014/08/08/cec-adventures-on-the-raspberry-pi/
Nothing works.  Everytime it compiles without errors and then refuses to open the connection.
The frustrating thing is this is a project which used to work fine and which I f*cked to oblivion with a single apt-get upgrade.  That will teach me about whole-disk backups!!!  The weardest thing is that raspbian has an official cecclient package but it just doesn't work (and that is not just for me!)
I dig that it's all volunteer work, but who uses time to make a package specifically for the raspberry which doesn't support the raspberry???

opdenkamp, 2015-05-05:
it's not a project specifically for the raspberry pi, and those instructions are outdated. just follow the instructions in the readme file and use the cmake command listed above

spectrokid, 2015-05-10:
I used this:
```cmake \
 -DCMAKE_TOOLCHAIN_FILE= ~/xbmc/tools/depends/target/libcec \
 -DCMAKE_INSTALL_PREFIX=/usr -DBUILD_SHARED_LIBS=1 \
 -DCMAKE_INSTALL_LIBDIR=/usr/lib \
 -DSKIP_PYTHON_WRAPPER:STRING=1 ..```
both on the Platform package and on the libcec.
(as I could not find a folder called toolchain.cmake in the XMBC repository)

This got me to``` -- Raspberry Pi support:                   yes```
Hurray!
But then I type make, and I get this:
``` 
pi@raspberrypi ~/libcec/build $ sudo make
Scanning dependencies of target cec
[  2%] Building CXX object src/libcec/CMakeFiles/cec.dir/CECClient.cpp.o
In file included from /home/pi/libcec/src/libcec/CECClient.cpp:35:0:
/home/pi/libcec/src/libcec/CECClient.h:48:11: error: ‘shared_ptr’ in namespace ‘std’ does not name a type
In file included from /home/pi/libcec/src/libcec/CECClient.cpp:37:0:
/home/pi/libcec/src/libcec/CECProcessor.h:59:11: error: ‘shared_ptr’ in namespace ‘std’ does not name a type
/home/pi/libcec/src/libcec/CECProcessor.h:63:5: error: ‘CECClientPtr’ does not name a type
/home/pi/libcec/src/libcec/CECProcessor.h:71:58: error: ‘CECClientPtr’ has not been declared
/home/pi/libcec/src/libcec/CECProcessor.h:76:5: error: ‘CECClientPtr’ does not name a type
In file included from /home/pi/libcec/src/libcec/CECClient.cpp:37:0:
/home/pi/libcec/src/libcec/CECProcessor.h:90:27: error: ‘CECClientPtr’ has not been declared
/home/pi/libcec/src/libcec/CECProcessor.h:92:29: error: ‘CECClientPtr’ has not been declared
/home/pi/libcec/src/libcec/CECProcessor.h:95:7: error: ‘CECClientPtr’ does not name a type
/home/pi/libcec/src/libcec/CECProcessor.h:96:7: error: ‘CECClientPtr’ does not name a type
/home/pi/libcec/src/libcec/CECProcessor.h:126:29: error: ‘CECClientPtr’ has not been declared
/home/pi/libcec/src/libcec/CECProcessor.h:163:37: error: ‘CECClientPtr’ has not been declared
/home/pi/libcec/src/libcec/CECProcessor.h:191:37: error: ‘CECClientPtr’ was not declared in this scope
/home/pi/libcec/src/libcec/CECProcessor.h:191:49: error: template argument 2 is invalid
/home/pi/libcec/src/libcec/CECProcessor.h:191:49: error: template argument 4 is invalid
In file included from /home/pi/libcec/src/libcec/CECClient.cpp:38:0:
/home/pi/libcec/src/libcec/LibCEC.h:47:11: error: ‘shared_ptr’ in namespace ‘std’ does not name a type
In file included from /home/pi/libcec/src/libcec/CECClient.cpp:38:0:
/home/pi/libcec/src/libcec/LibCEC.h:132:7: error: ‘CECClientPtr’ does not name a type
/home/pi/libcec/src/libcec/LibCEC.h:133:19: error: ‘CECClientPtr’ was not declared in this scope
/home/pi/libcec/src/libcec/LibCEC.h:133:31: error: template argument 1 is invalid
/home/pi/libcec/src/libcec/LibCEC.h:133:31: error: template argument 2 is invalid
/home/pi/libcec/src/libcec/LibCEC.h:151:7: error: ‘CECClientPtr’ does not name a type
/home/pi/libcec/src/libcec/LibCEC.h:152:19: error: ‘CECClientPtr’ was not declared in this scope
/home/pi/libcec/src/libcec/LibCEC.h:152:31: error: template argument 1 is invalid
/home/pi/libcec/src/libcec/LibCEC.h:152:31: error: template argument 2 is invalid
In file included from /home/pi/libcec/src/libcec/devices/CECPlaybackDevice.h:36:0,
                 from /home/pi/libcec/src/libcec/CECClient.cpp:40:
/home/pi/libcec/src/libcec/devices/CECBusDevice.h:52:11: error: ‘shared_ptr’ in namespace ‘std’ does not name a type
In file included from /home/pi/libcec/src/libcec/devices/CECPlaybackDevice.h:36:0,
                 from /home/pi/libcec/src/libcec/CECClient.cpp:40:
/home/pi/libcec/src/libcec/devices/CECBusDevice.h:193:5: error: ‘CECClientPtr’ does not name a type
src/libcec/CMakeFiles/cec.dir/build.make:54: recipe for target 'src/libcec/CMakeFiles/cec.dir/CECClient.cpp.o' failed
make[2]: *** [src/libcec/CMakeFiles/cec.dir/CECClient.cpp.o] Error 1
CMakeFiles/Makefile2:190: recipe for target 'src/libcec/CMakeFiles/cec.dir/all' failed
make[1]: *** [src/libcec/CMakeFiles/cec.dir/all] Error 2
Makefile:113: recipe for target 'all' failed
make: *** [all] Error 2
```

opdenkamp, 2015-05-11:
looks like your compiler doesn't support c++11

spectrokid, 2015-05-11:
yup: "At the time of this writing Raspbian is based on Debian Wheezy, which comes with the stable but outdated GCC 4.6 as the default C and C++ compiler. If you want to write modern C++ code you are stuck with a partial implementation of the C++11 standard."

opdenkamp, 2015-05-11:
Kodi 15 (Isengard) requires c++11 too, so it looks like it's about time 
Raspbian upgrades to something more recent :)

popcornmix, 2015-05-11:
You can get gcc 4.8 fairly easily under raspbian:
https://somewideopenspace.wordpress.com/2014/02/28/gcc-4-8-on-raspberry-pi-wheezy/

Raspbian is moving to jessie soon...

spectrokid, 2015-05-11:
Looks like I will have to live with 2 remotes for the next couple of months. :-(

spectrokid, 2015-05-12:
I tried, but I just move on to the next set of error messages.  I think it
is time to pull the plug, I also have 2 kids to feed...
I will try again when either Raspbian moves to Jessie or when Microsoft
launches windows 10 on the Pi, whichever happens first.

Snyder1977, 2015-05-12:
I´m just wondering why this issue is closed.
I would defintley like to use the lastest libcec release on my Pi because of the python wrapper which is supposed to work since the latest release. I am still waiting for the maintainers to build it and to integrate it in their images. I'm running kodi on Rasbian/Wheezy and I´m still not able to use libcec 3.0.

malard, 2015-05-12:
you can't on Wheezy, it's not compatible

Snyder1977, 2015-05-12:
Which Linux distribution would you recommend? I would like to run kodi and I also want to use the python Interface for libcec.

malard, 2015-05-12:
you will need to move away from the Pi until they make a new build using Jessie or when Windows 10 is released, as @spectrokid said

opdenkamp, 2015-05-12:
as @popcornmix pointed out in https://github.com/Pulse-Eight/libcec/issues/77#issuecomment-101028358 you need an updated gcc to build c++11 code. the issue has been closed because this is not an issue with libCEC or the Pi, but just an issue with the (very old) gcc version used by your distribution

Snyder1977, 2015-05-12:
ok, many thanks. I got it now.... then I´ll keep on waiting for Jessie or Windows 10.

conect, 2015-08-03:
I have updated to Jessie as suggested here: (https://somewideopenspace.wordpress.com/2014/02/28/gcc-4-8-on-raspberry-pi-wheezy/ )  but nevertheless I receive the same errors as @spectrokid said.
g++ --version says: 
g++ (Raspbian 4.8.4-1) 4.8.4
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

opdenkamp, 2015-08-03:
if you get the same errors, then cmake doesn't use the correct compiler most likely. your system may come with multiple versions of gcc installed (not unusual) and cmake may use the wrong one.

if your system can build and install a recent kodi version, then it supports c++11 (it requires the same stuff as libCEC to compile)

conect, 2015-08-03:
I placed:
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")
in the CMakeLists.txt file afterwards it compiled and I could execute sudo make install.
But now when I execute cec-client i get:
libcec.so.3.0: cannot open shared object file: No such file or directory
Cannot load libcec.so

conect, 2015-08-03:
Ok summary for those who struggle to get current libcec running on raspberry pi (so far tested on raspberry 2):
#update your raspberry to jessie (needed für c++11)
update your raspberry:
sudo apt-get update
sudo apt-get upgrade
sudo apt-get dist-upgrade
sudo sed -i /deb/s/wheezy/jessie/g /etc/apt/sources.list
sudo sed -i /deb/s/wheezy/jessie/g /etc/apt/sources.list.d/*.list
sudo apt-get update
sudo apt-get upgrade
sudo apt-get dist-upgrade

#get libcec:
git clone git://github.com/Pulse-Eight/libcec.git

#force libcec to use c++11
edit the CMakeLists.txt
add following line:
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")

#install
mkdir build
cd build
cmake -DRPI_INCLUDE_DIR=/opt/vc/include -DRPI_LIB_DIR=/opt/vc/lib -DCMAKE_INSTALL_PREFIX:PATH=/usr ..
make
sudo make install
sudo ldconfig

#check if installation worked
cec-client -l

cerw, 2017-07-21:
Hi

tried all that, but still getting this: 

```
/usr/bin/ld: cannot find -lpthreads
```

opdenkamp, 2017-07-21:
I suppose that your build environment is incomplete. Try `sudo apt-get update; sudo apt-get install build-essential ; sudo apt-get upgrade`, remove the build directory and try again.

The instructions from https://github.com/Pulse-Eight/libcec/blob/master/docs/README.raspberrypi.md work on a fresh installation of Raspbian.

cerw, 2017-07-21:
@opdenkamp I did followed the https://github.com/Pulse-Eight/libcec/blob/master/docs/README.raspberrypi.md 

```
raspberrypi:/home/homeassistant/.homeassistant# apt-get install build-essential
Reading package lists... Done
Building dependency tree
Reading state information... Done
build-essential is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
raspberrypi:/home/homeassistant/.homeassistant#
```

It's today fresh new install only has [HASS](home-assistant.io) on it and nginx 

thanks