Pulse-Eight/libcec #80: libcec stops getting input after source change & switch back

LongChair, 2015-04-11:
I am running an RPI2 directly connected to a TV thru an HDMI cable.
The RPi runs Openelec with libcec 3

He is what happens:
- Boot up and start cecc-client
- press a few keys : everything works as expected
- switch sources from HDMI1 (connected to RPI) to HDM2 (connected to nothing
- switch back to HDMI1

at this point cecc-client still seems to see the switch, but pressing any key will seem to trigger any key event.

Here is a log of cecc-client with the sequence above :

```

No device type given. Using 'recording device'
CEC Parser created - libCEC version 3.0.0
no serial port given. trying autodetect:
 path:     Raspberry Pi
 com port: RPI

opening a connection to the CEC adapter...
DEBUG:   [              62]	Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [              62]	InitHostCEC - vchiq_initialise succeeded
DEBUG:   [              62]	InitHostCEC - vchi_initialise succeeded
DEBUG:   [              62]	InitHostCEC - vchi_connect succeeded
DEBUG:   [              62]	Open - vc_cec initialised
DEBUG:   [              63]	logical address changed to Free use (e)
NOTICE:  [              63]	connection opened
DEBUG:   [              63]	<< Broadcast (F) -> TV (0): POLL
DEBUG:   [              63]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [              63]	<< e0
DEBUG:   [              63]	processor thread started
DEBUG:   [             124]	>> POLL sent
DEBUG:   [             124]	TV (0): device status changed into 'present'
DEBUG:   [             124]	<< requesting vendor ID of 'TV' (0)
TRAFFIC: [             124]	<< e0:8c
TRAFFIC: [             318]	>> 0f:87:00:10:39
DEBUG:   [             319]	>> TV (0) -> Broadcast (F): device vendor id (87)
DEBUG:   [             319]	TV (0): vendor = Unknown (001039)
DEBUG:   [             319]	expected response received (87: device vendor id)
NOTICE:  [             319]	registering new CEC client - v3.0.0
DEBUG:   [             319]	detecting logical address for type 'recording device'
DEBUG:   [             319]	trying logical address 'Recorder 1'
DEBUG:   [             319]	<< Recorder 1 (1) -> Recorder 1 (1): POLL
TRAFFIC: [             319]	<< 11
TRAFFIC: [             410]	<< 11
DEBUG:   [             500]	>> POLL not sent
DEBUG:   [             500]	using logical address 'Recorder 1'
DEBUG:   [             500]	Recorder 1 (1): device status changed into 'handled by libCEC'
DEBUG:   [             500]	Recorder 1 (1): power status changed from 'unknown' to 'on'
DEBUG:   [             500]	Recorder 1 (1): vendor = Pulse Eight (001582)
DEBUG:   [             501]	Recorder 1 (1): CEC version 1.4
DEBUG:   [             501]	AllocateLogicalAddresses - device '0', type 'recording device', LA '1'
DEBUG:   [             501]	logical address changed to Recorder 1 (1)
DEBUG:   [             501]	Recorder 1 (1): osd name set to 'CEC tester'
DEBUG:   [             502]	Recorder 1 (1): menu language set to 'eng'
DEBUG:   [             502]	GetPhysicalAddress - physical address = 1000
DEBUG:   [             502]	AutodetectPhysicalAddress - autodetected physical address '1000'
DEBUG:   [             502]	Recorder 1 (1): physical address changed from ffff to 1000
DEBUG:   [             502]	<< Recorder 1 (1) -> broadcast (F): physical adddress 1000
TRAFFIC: [             502]	<< 1f:84:10:00:01
NOTICE:  [             683]	CEC client registered: libCEC version = 3.0.0, client version = 3.0.0, firmware version = 1, logical address(es) = Recorder 1 (1) , physical address: 1.0.0.0, compiled on Linux-3.13.0-32-generic, features: P8_USB, P8_detect, randr, 'RPi'
DEBUG:   [             683]	<< Recorder 1 (1) -> TV (0): OSD name 'CEC tester'
TRAFFIC: [             683]	<< 10:47:43:45:43:20:74:65:73:74:65:72
TRAFFIC: [             684]	>> 01:83
DEBUG:   [             684]	>> TV (0) -> Recorder 1 (1): give physical address (83)
DEBUG:   [            1014]	<< requesting power status of 'TV' (0)
TRAFFIC: [            1014]	<< 10:8f
DEBUG:   [            1135]	<< Recorder 1 (1) -> broadcast (F): physical adddress 1000
[... 109 lines cut ...]
NOTICE:  [           32472]	physical address changed to 1000
DEBUG:   [           32472]	physical address unchanged (1000)
TRAFFIC: [           32472]	>> 01:83
DEBUG:   [           32473]	<< Recorder 1 (1) -> broadcast (F): physical adddress 1000
TRAFFIC: [           32473]	<< 1f:84:10:00:01
DEBUG:   [           32473]	>> TV (0) -> Recorder 1 (1): give physical address (83)
TRAFFIC: [           32668]	>> 01:46
DEBUG:   [           32668]	<< Recorder 1 (1) -> TV (0): OSD name 'CEC tester'
DEBUG:   [           32668]	>> TV (0) -> Recorder 1 (1): give osd name (46)
TRAFFIC: [           32668]	<< 10:47:43:45:43:20:74:65:73:74:65:72
signal caught: 2 - exiting
DEBUG:   [           47818]	unregistering all CEC clients
NOTICE:  [           47818]	unregistering client: libCEC version = 3.0.0, client version = 3.0.0, firmware version = 1, logical address(es) = Recorder 1 (1) , physical address: 1.0.0.0, compiled on Linux-3.13.0-32-generic, features: P8_USB, P8_detect, randr, 'RPi'
DEBUG:   [           47818]	Recorder 1 (1): power status changed from 'on' to 'unknown'
DEBUG:   [           47818]	Recorder 1 (1): vendor = Unknown (000000)
DEBUG:   [           47818]	Recorder 1 (1): CEC version unknown
DEBUG:   [           47818]	Recorder 1 (1): osd name set to 'Recorder 1'
DEBUG:   [           47818]	marking Recorder 1 (1) as inactive source
DEBUG:   [           47818]	Recorder 1 (1): device status changed into 'unknown'
DEBUG:   [           47818]	unregistering all CEC clients
DEBUG:   [           48069]	UnregisterLogicalAddress - releasing previous logical address
DEBUG:   [           48071]	logical address changed to Broadcast (f)
```

Let me know if you need anything else for troubleshooting

noben78, 2015-05-03:
Same issue here, but it started with libcec 2.2.0. (Using Openelec 5.0.7 os Pi 2, TV: LG 42LH3000)
Worked great on libcec 2.1.4!

opdenkamp, 2015-07-03:
@LongChair the log you pasted does not indicate any issues on libCEC's side. libCEC's sends all the correct responses in here, which should have made things work correctly after the switch. this is either a bug in your TV's or in the Pi's low level CEC implementation.

@noben78 unlikely that your issue is related. LG is always "special" because of the tricks that LG uses to prevent it to work with anything that LG didn't make themselves. I'll test LG before releasing 3.0.1 later today

opdenkamp, 2015-07-06:
Can you try again once 3.0.1 has been picked up by your distribution and pastebin a full log if it still doesn't work.

trusnak, 2015-07-06:
Hi, I tested this setup Pulse-Eight Adapter-> Onkyo -> LG TV with latest 3.0.1. 
Full log: http://pastebin.com/34Ju7ZRL

This happened when I choose Simplink -> Recorder on the TV:
TRAFFIC: [           20428]	>> 50:83
TRAFFIC: [           20553]	>> 0f:84:00:00
DEBUG:   [           20553]	>> TV (0) -> Broadcast (F): report physical address (84)
TRAFFIC: [           21790]	>> 5e
DEBUG:   [           21790]	<< POLL: Audio (5) -> Free use (e)
TRAFFIC: [           21837]	>> 5e
DEBUG:   [           21837]	<< POLL: Audio (5) -> Free use (e)
TRAFFIC: [           21989]	>> 5f:84:10:00:05
DEBUG:   [           21989]	>> Audio (5) -> Broadcast (F): report physical address (84)
TRAFFIC: [           22135]	>> 5f:87:00:09:b0
DEBUG:   [           22135]	>> Audio (5) -> Broadcast (F): device vendor id (87)
TRAFFIC: [           25951]	>> 01:8f
DEBUG:   [           25951]	<< Recorder 1 (1) -> TV (0): on
TRAFFIC: [           25951]	<< 10:90:00
DEBUG:   [           25951]	>> TV (0) -> Recorder 1 (1): give device power status (8F)
TRAFFIC: [           26150]	>> 01:1a:01
DEBUG:   [           26150]	>> Recorder 1 (1): deck status changed from 'stop' to 'LG other'
DEBUG:   [           26150]	<< Recorder 1 (1) -> TV (0): deck status 'LG other'
DEBUG:   [           26150]	>> TV (0) -> Recorder 1 (1): give deck status (1A)
TRAFFIC: [           26150]	<< 10:1b:20
TRAFFIC: [           26349]	>> 01:89:04:01
NOTICE:  [           26349]	SL initialised
TRAFFIC: [           26349]	<< 10:89:05:01
DEBUG:   [           26349]	>> TV (0) -> Recorder 1 (1): vendor command (89)
TRAFFIC: [           26521]	>> 01:83
DEBUG:   [           26521]	<< Recorder 1 (1) -> broadcast (F): physical adddress 1100
TRAFFIC: [           26521]	<< 1f:84:11:00:01
DEBUG:   [           26521]	>> TV (0) -> Recorder 1 (1): give physical address (83)
TRAFFIC: [           26760]	>> 01:83
DEBUG:   [           26760]	<< Recorder 1 (1) -> broadcast (F): physical adddress 1100
TRAFFIC: [           26760]	<< 1f:84:11:00:01
DEBUG:   [           26760]	>> TV (0) -> Recorder 1 (1): give physical address (83)
TRAFFIC: [           33270]	>> 05:89:01
TRAFFIC: [           33390]	>> 50:00:89:03
TRAFFIC: [           40897]	>> 0f:87:00:e0:91
DEBUG:   [           40897]	<< Recorder 1 (1) -> Broadcast (F): vendor id LG (e091)
TRAFFIC: [           40897]	<< 1f:87:00:e0:91
DEBUG:   [           40897]	>> TV (0) -> Broadcast (F): device vendor id (87)
TRAFFIC: [           41096]	>> 01:00
DEBUG:   [           41096]	>> TV (0) -> Recorder 1 (1): feature abort ( 0)
TRAFFIC: [           41248]	>> 05:00
TRAFFIC: [           41342]	>> 05:89:01
TRAFFIC: [           41389]	>> 55
DEBUG:   [           41389]	<< POLL: Audio (5) -> Audio (5)
TRAFFIC: [           41436]	>> 55
DEBUG:   [           41436]	<< POLL: Audio (5) -> Audio (5)
TRAFFIC: [           41556]	>> 50:00:89:03
DEBUG:   [           41598]	<< POLL: TV (0) -> Recorder 2 (2)

It's still not working with message "Please try again" on the TV.

attaxia, 2015-11-05:
I have the same issue. I updated the firmware of my TV. I have the issue with a Raspberry Pi 1 and 2. I've tried libcec 3.0.0 and 3.0.1. The connection works after a boot of the system. A quick source change doesn't affect anything. Watching a different channel for a longer period of time does however disable the CEC connection.

If I switch back to HDMI I can then not use the remote. However, if I go to the simplink menu and select the "HDD Recorder" (which is the RPi) I will get the "Please try again error". It does work again though! I would like however that it works always when switching to HDMI.

attaxia, 2015-11-21:
Actually I paid more attention and my issue is slightly different:

For my TV (LG 49LB550V) this is a 100% way to reproduce the issue:

1. Use OpenELEC/Rasplex with HDMI-CEC
2. Switch to TV channel (issue does not occur if TV is left on HDMI channel)
3. Switch TV off (standby)
4. Wait 15 minutes
5. Turn on TV
6. Switch to HDMI channel

Now HDMI-CEC remote control forwarding will not work. Turning Simplink on and off in the settings will fix the issue without reboot. Or using the method described in my previous post.

opdenkamp, 2015-11-25:
What happens when you select the source via the SimpLink menu (red button in the sources list on my model, but may be different on yours)

attaxia, 2015-11-25:
Yes, but this is only possible when the CEC connection is still/already working

opdenkamp, 2015-11-25:
do you also have the device connected through an AVR to the TV?

attaxia, 2015-11-25:
I'm not sure what an AVR is, but probably no. I have a direct HDMI cable from Raspberry Pi to my TV.

opdenkamp, 2015-11-25:
then the answer is no indeed, an AVR is an amplifier

attaxia, 2015-11-29:
Are there any logs I can provide that would help you debug the issue?

opdenkamp, 2015-12-01:
I cannot reproduce any of these issues with current master. The only thing that I've seen is that I got a "please try again" message once after attempting to switch after keeping the TV off for 15+ minutes. When I did try again, it worked.

attaxia, 2015-12-01:
Can it be that the bug was fixed since the last release? If so I might see if I can manage to compile it myself on openelec.

opdenkamp, 2015-12-01:
Nothing changed related to LG since 3.0.1. But I only noticed now that in the log of @trusnak the AVR is not compatible with SimpLink (LG's custom CEC). LG TVs don't support CEC when a device is connected that's made by another vendor. libCEC works around this issue, by fooling the TV into thinking that it's talking to another LG device. The AVR doesn't do this and if it's connected between the PC and TV, then it's likely causing CEC to break.

Nothing that we can do about it except for recommending you to buy another brand next time if you want CEC to work.

If one of you here doesn't have any other brand connected and is still getting issues, let me know by posting a log here, then I can check if it's actually caused by or fixable by libCEC and re-open this ticket.

closed 2015-12-01 by opdenkamp

attaxia, 2015-12-01:
Then maybe I posted in an incorrect thread, as I do not use an AVR. I have a direct connection from the PC to the TV.

opdenkamp, 2015-12-01:
Log please :)

attaxia, 2015-12-01:
Log can be found here.
http://pastebin.com/twNBm8QH

I did the following:
1. Run cec-client > libcec.log (I didn't find a way to enable logging without running an extra client)
2. Play around with the menu a bit, everything works
3. Switch to TV channel (this is important, if I leave the TV on hdmi and execute the next steps there usually isn't a problem)
4. Turn off the TV for about 5 minutes
5. Turn on the TV
6. Switch to HDMI
7. Try to reconnect the simplink, nothing works.

Usually I can reconnect the simplink manually, but I guess having two instances of the cec-client interfered with this.

opdenkamp, 2015-12-01:
this is a patched version according to the log. can you try with a build from the master branch or the last release please

attaxia, 2015-12-01:
Thanks again for the reply. I was not aware that this Rasplex (built on OpenELEC) distribution didn't come with a standard 3.0.1. I really hope that this is the issue. However, compiling a new version on Raspbian for testing failed, for which I've created issue #164