Pulse-Eight/libcec #81: Compile error due to undefined XCASE on Linux with musl toolchain

bkuhls, 2015-04-11:
Hi,

[  2%] Building CXX object src/libcec/CMakeFiles/cec.dir/platform/posix/serialport.cpp.o
/home/fli4l/br3/output/build/libcec-d2f8cf729686dc4d7cc7980cfe6979e2ee1d9c94/src/libcec/platform/posix/serialport.cpp: In member function 'virtual bool PLATFORM::CSerialSocket::Open(uint64_t)':
/home/br/br3/output/build/libcec-d2f8cf729686dc4d7cc7980cfe6979e2ee1d9c94/src/libcec/platform/posix/serialport.cpp:177:57: error: 'XCASE' was not declared in this scope
   m_options.c_lflag &= ~(ICANON | ECHO | ECHOE | ISIG | XCASE | ECHOK | ECHONL | ECHOCTL | ECHOPRT | ECHOKE | TOSTOP);
                                                         ^
make[3]: *** [src/libcec/CMakeFiles/cec.dir/platform/posix/serialport.cpp.o] Error 1

According to https://dev.openwrt.org/ticket/18186 XCASE is not supported on Linux at all.

opdenkamp, 2015-04-14:
well, it used to be supported but got marked as deprecated and removed from whatever version that you're using, same as on bsd. will update the backward compat check

opdenkamp, 2015-07-03:
see https://github.com/Pulse-Eight/libcec/commit/bff9c9b64e4db42420268fcde157b88a7cd485c1

closed 2015-07-03 by opdenkamp