Pulse-Eight/libcec #83: p8usb-cec.inf (Windows Driver) does not install on Windows 7 64 Bit

gpost, 2015-04-14:
Hello,

I have been attempting to install the Windows Version of libCEC tray on a win 7 64bit machine for use with an USB-CEC adapter .  I have tried both versions (2.1.3 & 2.1.4) available on pulse-eight.com under the "downloads" section of the HDMI-CEC Control.  I have also tried Windows version 2.2.0 found under the Pulse-Eight libCEC packages (http://packages.pulse-eight.net/windows/) 

The installer(s) reports success, However when I connect the USB-CEC adapter, Windows is unable to find the device driver, an entry is added tot he device manager group of "Other devices" with the name "USB-CEC Adapter", and the libCEC tray reports "Could not connect to any CEC adapter"

I have tried manually updating the driver for the listed device by right clicking the "USB-CEC Adapter", selecting update driver and pointing it to the driver folder containing p8usb-cec.inf.  This however always fails.  

I also tried manually adding p8usb-cec.inf to the driver store using PnPutil.  I then tried selecting the pulse-eight Manufacturer from the ports device group.  This also claimed to install successfully, hover the device still did not work with the libCEC tray.

Any guidance for getting the libCEC driver working under windows 7 64 bit would be appreciated.

opdenkamp, 2015-04-14:
you can't manually install the .inf file because it's not signed. a signed version of the file is included in http://packages.pulse-eight.net/windows/libcec-2.2.0.exe

the inf file just points to microsoft's own driver for usb serial ports, and if that's failing, then your system is most likely missing the drivers that get installed by default when you install windows. check the dialog in the first tab in the device manager for the "other device", it'll say what's wrong.

closing this ticket because it's not a bug in libCEC. please contact support@pulse-eight.com for further assistance.

closed 2015-04-14 by opdenkamp

gpost, 2015-04-14:
Hi Lars,

Thank you for your reply.

I have installed the exact same driver successfully on a Windows XP machine.  I have verified that all MS drivers listed in the INF are installed.

I can’t point at a specific problem, but it seems like there may be an issue on the NTamd64 section of the INF file.  Windows does not detect that p8usb-cec.inf contains a valid hardware driver for the USB-CEC device.   (This is a brand new device just purchased from Pulse Eight, tested as working on Linux and XP)

The only information I can provide from the first tab of the device manager is the following device status:

The drivers for this device are not installed (Code 28)
There is no driver selected for the device information set or element.

I have tried the 2.2.0 driver with the  same issue.

Thanks for your assistance.

-Greg

closing this ticket because it's not a bug in libCEC. please contact support@pulse-eight.com<mailto:support@pulse-eight.com> for further assistance.

opdenkamp, 2015-04-14:
We have many users running the same version as yours, including one of my own systems. So the only reason that I can think of is that your installation is missing the driver files that should have been installed with the system's installation. Try restoring the system via the installation dvd.

karbowiak, 2015-07-27:
gpost did you ever figure out this issue?

Cause i'm running into it aswell with Windows 8.1 Pro 64bit

Even tried downloading the libcec 3.0.0 version taht was shared here: https://github.com/Pulse-Eight/libcec/issues/88#issuecomment-98553530

So no avail. just goes
```
The drivers for this device are not installed. (Code 28)

A service installation section in this INF is invalid.

To find a driver for this device, click Update Driver.
```

And everytime i try and install a driver, it increments the com port a number up (from COM5 to COM19 now)

gpost, 2015-07-27:
I was not able to resolve the issue on one machine.  I think the problem was a conflict with another USB driver.  Seemed to install fine on a fresh OS.

Even tried downloading the libcec 3.0.0 version taht was shared here: #88 (comment)<https://github.com/Pulse-Eight/libcec/issues/88#issuecomment-98553530>