Pulse-Eight/libcec #84: Complete installation instructions on Raspbian

andreikun, 2015-04-17:
I finally managed to install libCEC within Raspbian, after hours and hours of searching, hitting different errors and so on. Below is the complete installation process required in order to have libcec for your Raspberry Pi.

1. Install required tools : sudo apt-get install cmake g++-4.8

2. Get development headers
   sudo apt-get install liblockdev1-dev libudev-dev libxrandr-dev

3. Python & SWIG
   sudo apt-get install python-dev swig

4. Get all the latest code
   git clone --recursive git://github.com/Pulse-Eight/libcec.git

No need here to install libplatform separately. --recursive will do the trick.

5. Modify ~/libcec/src/libcec/cmake/CheckPlatformSupport.cmake
   change
     check_library_exists(bcm_host vchi_initialise "" HAVE_RPI_API)
   to
     check_library_exists(bcm_host bcm_host_init "" HAVE_RPI_API)

   and below 'set(LIB_INFO "${LIB_INFO}, 'RPi'")' add
     include_directories(/opt/vc/include /opt/vc/include/interface/vcos/pthreads /opt/vc/include/interface/vmcs_host/linux)
6. Build platform package.
   cd ~/libcec/src/platform
   mkdir build
   cd build
   cmake -DCMAKE_CXX_COMPILER=g++-4.8 ..
   make
   sudo make install
7. Build libcec library
   cd ~/libcec
   mkdir build
   cd build
   export LIBRARY_PATH=/opt/vc/lib
   cmake -DCMAKE_CXX_COMPILER=g++-4.8 -DCMAKE_INSTALL_PREFIX=/usr -DBUILD_SHARED_LIBS=1 ..
   make
   sudo -E bash -c 'make install'

Enjoy it! :D

rasp-presentator, 2015-04-29:
I had a couple of problems by following your instructions.

1. - Build platform package.
1.1 after creating the build dir running the cmake-DCMAKE... resulted in the errormsg. 
      "CMAKE Error. The source directory "/home/pi/libcec/src/platform/build" does not appear to contain
       CMakeLists.txt."
1.2 Fixed that error by coping the CMakeLists.txt from "/home/pi/libcec/src/platform" to "/home/pi/libcec
       /src/platform/build"
1.3 excecuted cmake -DCMAKE... again. Result

opdenkamp, 2015-05-03:
Thanks for this, will add these to the readme file

avrelio, 2015-05-05:
Instructions followed setp by step in a fresh wheezy installation and worked like a charm. Thanks loads

joerifeyen, 2015-05-10:
Hi andreikun,

I followed your instructions on the latest version of Raspbian and everything works fine!

Thanks a lot ! 

Kr,

Joeri

coopak, 2015-06-06:
andreikun you're amazing. I spent so long trying to compile this before I came across your edit instructions. To readers: the above instructions works on up-to-date Raspbian as of 6/6/15.

ffranco-daitan, 2015-06-09:
I just followed your instructions and, although they were the only ones that came close to working, they unfortunately did not work on raspbian jessie.
My first indication that something was wrong was:
```
-- Found SWIG: /usr/bin/swig2.0 (found version "2.0.12")
-- Configured features:
-- Pulse-Eight CEC Adapter:                yes
-- Pulse-Eight CEC Adapter detection:      yes
-- lockdev support:                        yes
-- xrandr support:                         yes
-- Raspberry Pi support:                   no
-- TDA995x support:                        no
-- Exynos support:                         no
-- Python support:                         version 2.7.9 (2.7)
-- lib info: compiled on Linux-3.18.11+, features: P8_USB, P8_detect, randr
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/libcec/build
```
Notice: ```-- Raspberry Pi support:                   no```

Then trying to run cec-client gives me:

```
$ cec-client
No device type given. Using 'recording device'
CEC Parser created - libCEC version 3.0.0
no serial port given. trying autodetect: FAILED
```

Any help would be much appreciated!

guaycuru, 2015-06-09:
I just followed your instructions and, although they were the only ones that came close to working, they unfortunately did not work on raspbian jessie.

```
-- Found SWIG: /usr/bin/swig2.0 (found version "2.0.12")
-- Configured features:
-- Pulse-Eight CEC Adapter:                yes
-- Pulse-Eight CEC Adapter detection:      yes
-- lockdev support:                        yes
-- xrandr support:                         yes
-- Raspberry Pi support:                   no
-- TDA995x support:                        no
-- Exynos support:                         no
-- Python support:                         version 2.7.9 (2.7)
-- lib info: compiled on Linux-3.18.11+, features: P8_USB, P8_detect, randr
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/libcec/build
```
Notice: ```-- Raspberry Pi support:                   no```

Then trying to run cec-client gives me:

```
$ cec-client
No device type given. Using 'recording device'
CEC Parser created - libCEC version 3.0.0
no serial port given. trying autodetect: FAILED
```

So what I did was to comment out the IFs that detect RPI support on ~/libcec/src/libcec/cmake/CheckPlatformSupport.cmake:

```
#if (HAVE_RPI_API)
```
and
```
#endif()
```

Horrible hack, I know, but it worked! :D

opdenkamp, 2015-07-02:
I've updated the cmake files and README.md file and tested this with the firmware files in /opt like this distribution does. the changes will be included in tomorrow's 3.0.1 update

closed 2015-07-02 by opdenkamp

kernelsmith, 2015-07-06:
@opdenkamp if you put "Fixes #issue_num" in the commit message (in the short title or the expanded commit message) for the code fix then the issue will automatically be closed and linked to the commit w/the actual fix.  Makes it easy for people to correlate the code change to the issue/bug

opdenkamp, 2015-07-10:
yes I'm aware of the triggers, just forgot to update the commit message before closing these and didn't want to spend time finding them all again ;-)

Isaacpm, 2015-07-18:
Hi,

I'm following the instructions and get the following errors when building the libcec library, with the following command
cmake -DCMAKE_CXX_COMPILER=g++-4.8 -DCMAKE_INSTALL_PREFIX=/usr -DBUILD_SHARED_LIBS=1 ..

The errors are:
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
RPI_BCM_HOST
    linked by target "cec" in directory /root/distr/libcec/src/libcec
RPI_VCHIQ_ARM
    linked by target "cec" in directory /root/distr/libcec/src/libcec
RPI_VCOS
    linked by target "cec" in directory /root/distr/libcec/src/libcec

Not sure what I'm missing as I followed the steps as per the post.
Any ideas?

Thanks,
Isaac

opdenkamp, 2015-07-19:
see https://github.com/Pulse-Eight/libcec/blob/master/README.md for instructions on how to build libCEC