Pulse-Eight/libcec #85: Enhancement, Extend KeyPress Callback to report device.

UEckhardt, 2015-04-18:
Hello,

I have an enhancement request. It would be nice to have the keypress callback to report also the device, which sends the key press. 

The background is, that several devices does not send some keys via CEC, for example the menu key is mostly not forwarded. Therefore my plugin for the VDR (http://www.uli-eckhardt.de/vdr/cec.en.shtml) allows to redefine keys. This can currently be done only global. But since different devices have different remotes with different keys, it would be nice to give the user the opportunity to redefine keys per device.

opdenkamp, 2015-05-03:
the only devices that support sending keys to libCEC are TVs. some AVRs send key presses too, but they're directed at the TV from the logs that I've seen. do you have any logs with different situations?

UEckhardt, 2015-06-23:
I have tried to find some devices, which sends key presses to the device controlled by libcec, but I until now I haven't found any. So I think this enhancement request has a really low priority. 

But in theory it should be allowed from any device to send key presses to the active cec device. So I am not sure if I should close this report or should leave it open e.g. for a major releases which already includes API changes required for other things, so that libcec may be capable of handling such devices.

closed 2025-02-27 by github-actions[bot]