Pulse-Eight/libcec #86: KeyPress callback not working reliable in 3.0.0

UEckhardt, 2015-04-18:
After upgrading to libCEC 3.0.0 the keypress callback seems not working reliable anymore. Sometimes i don't get any keypresses back from libCEC. I had to reconnect several times to libCEC.

Is there a way to debug keypresses via cec-client, to get a better description and probably a way to reproduce this better?

opdenkamp, 2015-05-03:
if you just pastebin the output of cec-client, then I should be able to see what's going on

UEckhardt, 2015-06-23:
This was not a problem of libcec. I had plugged the cec adapter to an USB3.0 port, and my motherboard/kernel drivers had problems, which leads to a strange behaviour of a lot of devices which where connected to this USB3.0 port.

closed 2015-06-23 by UEckhardt