Pulse-Eight/libcec #88: cec-tray hangs on windows

gorrus, 2015-05-02:
After some time, cec-tray process starts to consume too much cpu. When I open tray app and go to log output - it hangs completely, so I expect this is connected to logging somehow.
I tried restarting it and unchecking all checkboxes in log output (errors, traffic, etc.), after that it does not hang. Unfortunately, after restart this is not preserved and it logs everything again. 
v2.2.0

opdenkamp, 2015-05-03:
should be fixed in the next version of libCEC, which will be on our website soon. you can download the beta build here: https://dl.dropboxusercontent.com/u/59212404/libcec-3.0.0.exe

gorrus, 2015-05-04:
Great, thanks!
I've installed it, there were some errors during the installation
[image]
will report how it works

opdenkamp, 2015-05-04:
ok thanks, that must be the check for the new plugin that doesn't work correctly, I'll get that fixed.

gorrus, 2015-05-05:
it is all the same in 3.0.0,
here is a stack trace from process explorer
[image]

gorrus, 2015-05-26:
I have to restart tray app every day :(
Is there any way I could help with the issue?

opdenkamp, 2015-07-02:
I'll have to reproduce this in the debugger, that stack points to something external crashing for some reason. may be the log tab filling up.

opdenkamp, 2015-12-02:
log tab didn't have an upper limit. fixed in master (will be committed in a bit)

closed 2015-12-02 by opdenkamp

gorrus, 2015-12-04:
Super, thanks! Can I have a nightly build bits or something?

gorrus, 2016-03-30:
It seems that I can download only 3.0.0 windows binary that does not contain the fix. Where can I get 3.1.0 for windows?

opdenkamp, 2017-07-13:
The latest version of the software can be found on https://www.pulse-eight.com/p/104/usb-hdmi-cec-adapter