Pulse-Eight/libcec #89: Compile ubunut

jteeuw, 2015-05-05:
Hi i try to compile libcec , but get the following error :

CMake Error at src/cec-client/CMakeLists.txt:21 (find_package):
  By not providing "Findplatform.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "platform",
  but CMake did not find one.

  Could not find a package configuration file provided by "platform" with any
  of the following names:

    platformConfig.cmake
    platform-config.cmake

  Add the installation prefix of "platform" to CMAKE_PREFIX_PATH or set
  "platform_DIR" to a directory containing one of the above files.  If
  "platform" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!

jteeuw, 2015-05-05:
forgot to cmake te platform, compiling worked

closed 2015-05-05 by jteeuw

appup93, 2016-05-09:
Hi

Could you just ellaborate on what you did to get rid of the error? I am stuck with the same issue!

Lcpuche, 2019-01-28:
Same issue on Ubuntu. Can someone explain what this means?