Pulse-Eight/libcec #90: cec-client becomes unresponsive after 'active source' timeout

aedomsan, 2015-05-10:
Raspberry Pi 1, using cec-client hangs and becomes non responsive after successfully changing the active source by using "tx 4f 82 XX XX" or using txn, this happens regardless of using broadcast address or not. By non responsive I mean that all new commands entered are ignored, and logging to stdout ceases. After sending a sigterm, backlogged output finally comes out

osmc@osmc:~$ cec-client 
No device type given. Using 'recording device'
CEC Parser created - libCEC version 2.2.0
no serial port given. trying autodetect: 
 path:     Raspberry Pi
 com port: RPI

opening a connection to the CEC adapter...
DEBUG:   [             104]	unregistering all CEC clients
DEBUG:   [             106]	Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [             108]	InitHostCEC - vchiq_initialise succeeded
DEBUG:   [             109]	InitHostCEC - vchi_initialise succeeded
DEBUG:   [             110]	InitHostCEC - vchi_connect succeeded
DEBUG:   [             117]	logical address changed to Free use (e)
DEBUG:   [             118]	Open - vc_cec initialised
NOTICE:  [             119]	connection opened
DEBUG:   [             122]	<< Broadcast (F) -> TV (0): POLL
DEBUG:   [             123]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [             124]	<< e0
DEBUG:   [             127]	processor thread started
DEBUG:   [             186]	>> POLL sent
DEBUG:   [             187]	TV (0): device status changed into 'present'
DEBUG:   [             188]	<< requesting vendor ID of 'TV' (0)
TRAFFIC: [             190]	<< e0:8c
TRAFFIC: [             386]	>> 0f:87:00:00:f0
DEBUG:   [             386]	>> TV (0) -> Broadcast (F): device vendor id (87)
DEBUG:   [             387]	TV (0): vendor = Samsung (0000f0)
DEBUG:   [             388]	expected response received (87: device vendor id)
DEBUG:   [             389]	replacing the command handler for device 'TV' (0)
NOTICE:  [             390]	registering new CEC client - v2.2.0
DEBUG:   [             390]	detecting logical address for type 'recording device'
DEBUG:   [             392]	trying logical address 'Recorder 1'
DEBUG:   [             394]	<< Recorder 1 (1) -> Recorder 1 (1): POLL
TRAFFIC: [             395]	<< 11
WARNING: [             544]	unhandled response received: opcode=87 initiator=e destination=f response=0
TRAFFIC: [             634]	<< 11
DEBUG:   [             725]	>> POLL not sent
DEBUG:   [             726]	using logical address 'Recorder 1'
DEBUG:   [             726]	Recorder 1 (1): device status changed into 'handled by libCEC'
DEBUG:   [             726]	Recorder 1 (1): power status changed from 'unknown' to 'on'
DEBUG:   [             726]	Recorder 1 (1): vendor = Pulse Eight (001582)
DEBUG:   [             726]	Recorder 1 (1): CEC version 1.4
DEBUG:   [             726]	AllocateLogicalAddresses - device '0', type 'recording device', LA '1'
DEBUG:   [             728]	logical address changed to Recorder 1 (1)
DEBUG:   [             729]	Recorder 1 (1): osd name set to 'CECTester'
DEBUG:   [             730]	Recorder 1 (1): menu language set to 'eng'
DEBUG:   [             732]	GetPhysicalAddress - physical address = 1100
DEBUG:   [             733]	AutodetectPhysicalAddress - autodetected physical address '1100'
DEBUG:   [             734]	Recorder 1 (1): physical address changed from ffff to 1100
DEBUG:   [             735]	<< Recorder 1 (1) -> broadcast (F): physical adddress 1100
TRAFFIC: [             736]	<< 1f:84:11:00:01
NOTICE:  [             887]	CEC client registered: libCEC version = 2.2.0, client version = 2.2.0, firmware version = 1, logical address(es) = Recorder 1 (1) , physical address: 1.1.0.0, host: armv7l-unknown-linux-gnueabihf, features: 'P8 USB' 'P8 USB detect' 'RPi' 'EXYNOS', compiled on: Thu Dec 11 02:38:54 UTC 2014 by root@compiler2 on Linux 3.2.0-4-amd64 (armv7l)
DEBUG:   [             888]	<< Recorder 1 (1) -> TV (0): OSD name 'CECTester'
TRAFFIC: [             888]	<< 10:47:43:45:43:54:65:73:74:65:72
DEBUG:   [            1189]	<< requesting power status of 'TV' (0)
TRAFFIC: [            1189]	<< 10:8f
TRAFFIC: [            1335]	>> 01:90:00
DEBUG:   [            1335]	>> TV (0) -> Recorder 1 (1): report power status (90)
DEBUG:   [            1335]	TV (0): power status changed from 'unknown' to 'on'
DEBUG:   [            1337]	expected response received (90: report power status)
waiting for input

tx 4f 82 11 00

TRAFFIC: [           26959]	<< 4f:82:11:00
WARNING: [           27081]	unhandled response received: opcode=82 initiator=1 destination=f response=0
TRAFFIC: [           27155]	>> 01:8d:02
DEBUG:   [           27155]	>> TV (0) -> Recorder 1 (1): menu request (8D)
DEBUG:   [           27961]	command 'active source' timeout

h
help
//This command was successful, inputs changed, but as you can see, cec-client no longer responds

^Csignal caught: 2 - exiting
DEBUG:   [           47455]	<< Recorder 1 (1) -> TV (0): menu state 'activated'
TRAFFIC: [           47455]	<< 10:8e:00
DEBUG:   [           47457]	unregistering all CEC clients
NOTICE:  [           47458]	unregistering client: libCEC version = 2.2.0, client version = 2.2.0, firmware version = 1, logical address(es) = Recorder 1 (1) , physical address: 1.1.0.0, host: armv7l-unknown-linux-gnueabihf, features: 'P8 USB' 'P8 USB detect' 'RPi' 'EXYNOS', compiled on: Thu Dec 11 02:38:54 UTC 2014 by root@compiler2 on Linux 3.2.0-4-amd64 (armv7l)
DEBUG:   [           47546]	Recorder 1 (1): power status changed from 'on' to 'unknown'
DEBUG:   [           47547]	Recorder 1 (1): vendor = Unknown (000000)
DEBUG:   [           47547]	Recorder 1 (1): CEC version unknown
DEBUG:   [           47547]	Recorder 1 (1): osd name set to 'Recorder 1'
DEBUG:   [           47547]	Recorder 1 (1): device status changed into 'unknown'
DEBUG:   [           47547]	unregistering all CEC clients
DEBUG:   [           48139]	UnregisterLogicalAddress - releasing previous logical address
DEBUG:   [           48140]	logical address changed to Broadcast (f)
osmc@osmc:~$

mkreisl, 2015-05-20:
@aedomsan 
Maybe the same issue like mine #94
You can check this by just waiting 500s

opdenkamp, 2015-07-02:
that fix will be included in tomorrow's 3.0.1 update

closed 2015-07-02 by opdenkamp