Pulse-Eight/libcec #93: standby command toggles power for Samsung UE46F6470

CSchulz, 2015-05-16:
Executing `echo 'standby 0' | cec-client -s /dev/ttyACM0` toggles the power state for a Samsung UE46F6470 TV

opdenkamp, 2015-07-02:
can you pastebin the full log output please. looks like a problem on samsung's side as I've never heard of anything like this

opdenkamp, 2015-07-03:
I cannot reproduce this with the latest code:
```
echo "standby 0" | cec-client -s
opening a connection to the CEC adapter...
...
DEBUG:   [             808]	<< requesting power status of 'TV' (0)
TRAFFIC: [             808]	<< 10:8f
TRAFFIC: [             958]	>> 01:90:01
DEBUG:   [             958]	TV (0): power status changed from 'unknown' to 'standby'
DEBUG:   [             958]	expected response received (90: report power status)
DEBUG:   [             959]	>> TV (0) -> Recorder 1 (1): report power status (90)
NOTICE:  [             959]	<< putting 'TV' (0) in standby mode
TRAFFIC: [             959]	<< 10:36
...
```

CSchulz, 2015-07-04:
Perhaps it took some time before Debian got the newest version.

I can't reproduce it either.

closed 2015-07-04 by CSchulz