Pulse-Eight/libcec #95: Repeat functionality philips remote

Kroellie, 2015-05-21:
Hi,

I want to re-open issue: https://github.com/Pulse-Eight/libcec/issues/66 since I'm seeing behaviour in the cec-client that's not in line with the statement that it is not supported.

I've noticed with the cec-client, that my Philips remote does in fact send repeated commands, I can see this scroll by when holding a button (i.e. down arrow) down. Though Kodi does not accept this input. What can I do to further troubleshoot this?

Debug log cec-client:
DEBUG: [ 31510] >> TV (0) -> Recorder 1 (1): user control pressed (44)
TRAFFIC: [ 31789] >> 01:44:02
DEBUG: [ 31789] >> TV (0) -> Recorder 1 (1): user control pressed (44)
TRAFFIC: [ 32067] >> 01:44:02
DEBUG: [ 32067] >> TV (0) -> Recorder 1 (1): user control pressed (44)
TRAFFIC: [ 32346] >> 01:44:02
DEBUG: [ 32346] >> TV (0) -> Recorder 1 (1): user control pressed (44)
TRAFFIC: [ 32625] >> 01:44:02
DEBUG: [ 32625] >> TV (0) -> Recorder 1 (1): user control pressed (44)
TRAFFIC: [ 32897] >> 01:44:02
DEBUG: [ 32897] >> TV (0) -> Recorder 1 (1): user control pressed (44)

Thanks.

opdenkamp, 2015-07-02:
can you paste the full log please. looks like the repeated key may not be forwarded

Nico5206, 2015-07-03:
Hi,

I have the same issue as Kroellie.
Here are my logs (hopefully complete):

On single press:
TRAFFIC: [           98137]     >> 01:44:02
DEBUG:   [           98137]     >> TV (0) -> Recorder 1 (1): user control pressed (44)
DEBUG:   [           98137]     SetCurrentButton down (2) D:0ms cur:2
DEBUG:   [           98137]     key pressed: down (2) current(ff) duration(0)
DEBUG:   [           98137]     Changed key down (2) D:0ms cur:ff
DEBUG:   [           98137]     key pressed: down (2, 0)
DEBUG:   [           98137]     CheckKeypressTimeout T:1172.869
DEBUG:   [           98138]     Key down: idle (duration:0) (2) timeout:500ms (rel:500,rep:0,prs:1,rel:0)
TRAFFIC: [           98366]     >> 01:8b:02
DEBUG:   [           98366]     >> TV (0) -> Recorder 1 (1): vendor remote button up (8B)
DEBUG:   [           98367]     key released: down (2) D:230ms

On long press:
TRAFFIC: [          134435]     >> 01:44:02
DEBUG:   [          134436]     >> TV (0) -> Recorder 1 (1): user control pressed (44)
DEBUG:   [          134436]     SetCurrentButton down (2) D:0ms cur:2
DEBUG:   [          134436]     key pressed: down (2) current(ff) duration(0)
DEBUG:   [          134436]     Changed key down (2) D:0ms cur:ff
DEBUG:   [          134436]     key pressed: down (2, 0)
DEBUG:   [          134436]     CheckKeypressTimeout T:1209.168
DEBUG:   [          134436]     Key down: idle (duration:0) (2) timeout:500ms (rel:500,rep:0,prs:1,rel:0)
DEBUG:   [          134938]     CheckKeypressTimeout T:1209.670
DEBUG:   [          134938]     Key unknown: idle (duration:502) (ff) timeout:1000ms (rel:0,rep:0,prs:0,rel:0)
TRAFFIC: [          135135]     >> 01:44:02
DEBUG:   [          135135]     >> TV (0) -> Recorder 1 (1): user control pressed (44)
TRAFFIC: [          135584]     >> 01:44:02
DEBUG:   [          135585]     >> TV (0) -> Recorder 1 (1): user control pressed (44)
TRAFFIC: [          136035]     >> 01:44:02
DEBUG:   [          136035]     >> TV (0) -> Recorder 1 (1): user control pressed (44)
TRAFFIC: [          136485]     >> 01:44:02
DEBUG:   [          136485]     >> TV (0) -> Recorder 1 (1): user control pressed (44)
TRAFFIC: [          136935]     >> 01:44:02
DEBUG:   [          136936]     >> TV (0) -> Recorder 1 (1): user control pressed (44)
TRAFFIC: [          137386]     >> 01:44:02
DEBUG:   [          137387]     >> TV (0) -> Recorder 1 (1): user control pressed (44)
TRAFFIC: [          137836]     >> 01:44:02
DEBUG:   [          137836]     >> TV (0) -> Recorder 1 (1): user control pressed (44)
TRAFFIC: [          138404]     >> 01:44:02
DEBUG:   [          138404]     >> TV (0) -> Recorder 1 (1): user control pressed (44)
TRAFFIC: [          138479]     >> 01:8b:02
DEBUG:   [          138479]     >> TV (0) -> Recorder 1 (1): vendor remote button up (8B)

Do you think something can be done?

Thanks.

opdenkamp, 2015-07-03:
Looks like it. Will check this issue before the next update. I have a 
Philips TV here that I can use to test it.

Kroellie, 2015-07-05:
Awesome... Thanks. Any guesstimate date for updated version?

opdenkamp, 2015-07-06:
Can't provide an ETA, sorry.

I've just tested this on two Philips TVs that I have here, both MY2012, and it doesn't send repeats correctly (it only sends key pressed and released, and is pretty slow in doing that). So it looks like something that they've only fixed recently and that they didn't bother to update the software in previous models.

I'll check some other brands that I have here.

Kroellie, 2015-07-06:
If there is anything I can do please let me know...

opdenkamp, 2015-07-10:
https://github.com/Pulse-Eight/libcec/pull/111 will improve things for you

closed 2015-07-10 by opdenkamp

mrgreywater, 2015-08-04:
I don't think #111 will fix the issue, since the button is auto released for some reason. The input is not repeated at all, even when holding it.

To clarify: When holding the button, only a single input action is executed, even though libcec receives multiple "user control pressed (44)" commands.

This are some more some more logs:
*SINGLE PRESS ARROW-DOWN*
    16:27:46 2462.345215 T:1867510848   DEBUG: CecLogMessage - >> 01:44:02
    16:27:46 2462.345459 T:1867510848   DEBUG: CecLogMessage - key pressed: down (2)
    16:27:46 2462.345703 T:1867510848   DEBUG: PushCecKeypress - received key a7 duration 0
    16:27:46 2462.345703 T:1867510848   DEBUG: CecLogMessage - >> TV (0) -> Recorder 1 (1): user control pressed (44)
    16:27:46 2462.351562 T:1967886336   DEBUG: OnKey: 167 (0xa7) pressed, action is Down
    16:27:47 2462.594971 T:1867510848   DEBUG: CecLogMessage - >> 01:8b:02
    16:27:47 2462.595215 T:1867510848   DEBUG: CecLogMessage - key released: down (2)
    16:27:47 2462.595459 T:1867510848   DEBUG: PushCecKeypress - received key a7 duration 249
    16:27:47 2462.595459 T:1867510848   DEBUG: CecLogMessage - >> TV (0) -> Recorder 1 (1): vendor remote button up (8B)
    
    
*SINGLE PRESS ARROW-UP*
    16:27:24 2440.355469 T:1867510848   DEBUG: CecLogMessage - >> 01:44:01
    16:27:24 2440.355713 T:1867510848   DEBUG: CecLogMessage - key pressed: up (1)
    16:27:24 2440.355713 T:1867510848   DEBUG: PushCecKeypress - received key a6 duration 0
    16:27:24 2440.355713 T:1867510848   DEBUG: CecLogMessage - >> TV (0) -> Recorder 1 (1): user control pressed (44)
    16:27:25 2440.390381 T:1967886336   DEBUG: OnKey: 166 (0xa6) pressed, action is Up
    16:27:25 2440.697510 T:1867510848   DEBUG: CecLogMessage - >> 01:8b:01
    16:27:25 2440.697754 T:1867510848   DEBUG: CecLogMessage - key released: up (1)
    16:27:25 2440.697754 T:1867510848   DEBUG: PushCecKeypress - received key a6 duration 342
    16:27:25 2440.697998 T:1867510848   DEBUG: CecLogMessage - >> TV (0) -> Recorder 1 (1): vendor remote button up (8B)
    
*HOLD ARROW-UP*
    16:29:28 2564.316162 T:1867510848   DEBUG: CecLogMessage - >> 01:44:01
    16:29:28 2564.316650 T:1867510848   DEBUG: CecLogMessage - key pressed: up (1)
    16:29:28 2564.316650 T:1867510848   DEBUG: PushCecKeypress - received key a6 duration 0
    16:29:28 2564.316895 T:1867510848   DEBUG: CecLogMessage - >> TV (0) -> Recorder 1 (1): user control pressed (44)
    16:29:28 2564.340576 T:1967886336   DEBUG: OnKey: 166 (0xa6) pressed, action is Up
    16:29:29 2564.781250 T:1867510848   DEBUG: CecLogMessage - >> 01:44:01
    16:29:29 2564.781738 T:1867510848   DEBUG: CecLogMessage - >> TV (0) -> Recorder 1 (1): user control pressed (44)
    16:29:29 2565.243896 T:1867510848   DEBUG: CecLogMessage - >> 01:44:01
    16:29:29 2565.244141 T:1867510848   DEBUG: CecLogMessage - key auto-released: up (1)
    16:29:29 2565.244141 T:1867510848   DEBUG: PushCecKeypress - received key a6 duration 927
    16:29:29 2565.244385 T:1867510848   DEBUG: CecLogMessage - >> TV (0) -> Recorder 1 (1): user control pressed (44)
    16:29:30 2565.701416 T:1867510848   DEBUG: CecLogMessage - >> 01:44:01
    16:29:30 2565.701660 T:1867510848   DEBUG: CecLogMessage - >> TV (0) -> Recorder 1 (1): user control pressed (44)
    16:29:30 2566.160645 T:1867510848   DEBUG: CecLogMessage - >> 01:44:01
    16:29:30 2566.160889 T:1867510848   DEBUG: CecLogMessage - >> TV (0) -> Recorder 1 (1): user control pressed (44)
    16:29:31 2566.621094 T:1867510848   DEBUG: CecLogMessage - >> 01:44:01
    16:29:31 2566.621338 T:1867510848   DEBUG: CecLogMessage - >> TV (0) -> Recorder 1 (1): user control pressed (44)
    16:29:31 2567.083496 T:1867510848   DEBUG: CecLogMessage - >> 01:44:01
    16:29:31 2567.083740 T:1867510848   DEBUG: CecLogMessage - >> TV (0) -> Recorder 1 (1): user control pressed (44)
    16:29:32 2567.540039 T:1867510848   DEBUG: CecLogMessage - >> 01:44:01
    16:29:32 2567.540527 T:1867510848   DEBUG: CecLogMessage - >> TV (0) -> Recorder 1 (1): user control pressed (44)
    16:29:32 2568.002686 T:1867510848   DEBUG: CecLogMessage - >> 01:44:01
    16:29:32 2568.002930 T:1867510848   DEBUG: CecLogMessage - >> TV (0) -> Recorder 1 (1): user control pressed (44)
    16:29:33 2568.462158 T:1867510848   DEBUG: CecLogMessage - >> 01:44:01
    16:29:33 2568.462646 T:1867510848   DEBUG: CecLogMessage - >> TV (0) -> Recorder 1 (1): user control pressed (44)
    16:29:33 2568.550049 T:1867510848   DEBUG: CecLogMessage - >> 01:8b:01
    16:29:33 2568.550293 T:1867510848   DEBUG: CecLogMessage - >> TV (0) -> Recorder 1 (1): vendor remote button up (8B)
    
*HOLD ARROW-DOWN*
    16:29:06 2541.447266 T:1867510848   DEBUG: CecLogMessage - >> 01:44:02
    16:29:06 2541.447510 T:1867510848   DEBUG: CecLogMessage - key pressed: down (2)
    16:29:06 2541.447754 T:1867510848   DEBUG: PushCecKeypress - received key a7 duration 0
    16:29:06 2541.447754 T:1867510848   DEBUG: CecLogMessage - >> TV (0) -> Recorder 1 (1): user control pressed (44)
    16:29:06 2541.466064 T:1967886336   DEBUG: OnKey: 167 (0xa7) pressed, action is Down
    16:29:06 2541.900146 T:1867510848   DEBUG: CecLogMessage - >> 01:44:02
    16:29:06 2541.900391 T:1867510848   DEBUG: CecLogMessage - >> TV (0) -> Recorder 1 (1): user control pressed (44)
    16:29:06 2542.360840 T:1867510848   DEBUG: CecLogMessage - >> 01:44:02
    16:29:06 2542.361084 T:1867510848   DEBUG: CecLogMessage - key auto-released: down (2)
[... cut ...]
    16:29:08 2544.201172 T:1867510848   DEBUG: CecLogMessage - >> TV (0) -> Recorder 1 (1): user control pressed (44)
    16:29:09 2544.659912 T:1867510848   DEBUG: CecLogMessage - >> 01:44:02
    16:29:09 2544.660400 T:1867510848   DEBUG: CecLogMessage - >> TV (0) -> Recorder 1 (1): user control pressed (44)
    16:29:09 2544.728271 T:1867510848   DEBUG: CecLogMessage - >> 01:8b:02
    16:29:09 2544.728516 T:1867510848   DEBUG: CecLogMessage - >> TV (0) -> Recorder 1 (1): vendor remote button up (8B)

All using libcec-3.0.0

mrgreywater, 2015-08-04:
Exerpt from the CEC Specs 13.13.2

> The initiator will send a <User Control Pressed> message when the remote control button is pressed. When
the button is released a <User Control Released> message should be sent by the initiator. The initiator
should not send repeated <User Control Pressed> messages for the same button press.
> ...
> If a follower has received a <User Control Pressed> message and it did not receive a <User Control
Released> message (or another <User Control Pressed> message with a different [UI Command] ) within
500ms, then it is recommended that the receiving device should assume that the button has been released
and act accordingly. 

It looks like libCEC does it correctly after the specs, but Phillips doesn't send the correct commands. I would suggest a vendor specific workaround.

mrgreywater, 2015-08-04:
Is this enough info to reopen the issue, @opdenkamp?