Pulse-Eight/libcec #97: CECTray wakes TV when it shouldn't.

amosstorkey, 2015-05-23:
I use libCEC and libCEC CECtray on Windows 7 with a Pulse Eight CEC Adapter.

I notice a bug in CECTray - even when the option is unchecked in CECTray, the CECTray sends a message to the TV to wake up when the windows machine comes out of standby. As a result the TV turns on when the computer comes out of standby automatically for guide updates etc. I would prefer CECTray to never wake the TV.

opdenkamp, 2015-07-02:
I've checked the code and it should just check the same settings that are used when the app starts. can you enable advanced mode, reproduce the issue, go to the log output tab and hit "save" or copy+paste it all to pastebin.com

lcasale, 2015-08-28:
I'm having the same issue with 2.2.0. Below is a log with me putting my computer to sleep and then waking it. It turns on my TV when it shouldn't.

http://pastebin.com/faUjms6P

lcasale, 2015-11-24:
This is still an issue as of 3.0.1

opdenkamp, 2015-12-02:
should be fixed after this commit

closed 2015-12-02 by opdenkamp

lcasale, 2016-07-23:
I'm still seeing this issue for 3.1.0