Pulse-Eight/libcec #99: Making libcec-3.0.0 works with Exynos.

YamashitaRen, 2015-05-29:
Hello,
Current git doesn't work with Exynos. Here is a (small) patch solving the issue : https://gist.github.com/YamashitaRen/8bb06aa2951030b4b1e9
Moreover, the Exynos support doesn't seem to be autodetected, I had to pass -DHAVE_EXYNOS_API=1 to cmake. I think it should be specified somewhere...

opdenkamp, 2015-07-02:
I've included this in the README.md file and included you c+p monster fix. It'll be included in tomorrow's 3.0.1 update. Thanks!

closed 2015-07-02 by opdenkamp

marlemion, 2016-02-16:
When I try this with 3.1.0:

```
cd build
cmake .. \
        -DCMAKE_BUILD_TYPE=Release \
        -DBUILD_SHARED_LIBS=1 \
        -DCMAKE_INSTALL_PREFIX=/usr \
        -DCMAKE_INSTALL_LIBDIR=/usr/lib \
        -DCMAKE_INSTALL_LIBDIR_NOARCH=/usr/lib \
        -DHAVE_EXYNOS_API=1 \
        -DHAVE_TDA995X_API=0 \
        -DHAVE_RPI_API=0
```

I get:
```

-- The C compiler identification is GNU 5.1.0
-- The CXX compiler identification is GNU 5.1.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test SUPPORTS_CXX11
-- Performing Test SUPPORTS_CXX11 - Success
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Looking for initscr in curses
-- Looking for initscr in curses - found
-- Looking for dlopen in dl
-- Looking for dlopen in dl - found
-- Looking for clock_gettime in rt
-- Looking for clock_gettime in rt - found
-- Found PythonLibs: /usr/lib/libpython3.5m.so (found version "3.5.1") 
-- git found: <unknown>
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29") 
-- Looking for include file sys/file.h
-- Looking for include file sys/file.h - found
-- Looking for flock
-- Looking for flock - found
-- Checking for module 'udev'
--   Found udev, version 228
-- Looking for 3 include files X11/Xlib.h, ..., X11/extensions/Xrandr.h
-- Looking for 3 include files X11/Xlib.h, ..., X11/extensions/Xrandr.h - found
-- Looking for XRRGetScreenResources in Xrandr
-- Looking for XRRGetScreenResources in Xrandr - found
-- Looking for bcm_host_init in bcm_host
-- Looking for bcm_host_init in bcm_host - not found
-- Found SWIG: /usr/bin/swig (found version "3.0.8") 
-- Configured features:
-- Pulse-Eight CEC Adapter:                yes
-- Pulse-Eight CEC Adapter detection:      yes
-- xrandr support:                         yes
-- Raspberry Pi support:                   no
-- TDA995x support:                        no
-- Exynos support:                         no
-- DRM support:                            yes
-- Python support:                         version 3.5.1 (3.5)
-- lib info: git revision: <unknown>, compiled on Di 16. Feb 11:39:37 UTC 2016 by eger@odroid on Linux 3.8.13.30-3-ARCH (armv7l), features: P8_USB, DRM, P8_detect, randr
-- Configuring done
-- Generating done
-- Build files have been written to: /home/eger/PKGBUILD/libcec-u2/src/libcec-libcec-3.1.0/build
```
Do I miss something?