Pulse-Eight/libcec #101: Supporting "pc standby" without having to run xbmc/kodi

mattiasost, 2015-06-07:
Chir21 did earlier a libcec tray app since it was not possible to enable the option "put pc in standby when tv is put to standby" in default app. 

I also think that it would be very good if you did not have to run xbmc just to get that function, I just want to be able to poweron TV/PC/Audio and Standby TV/PC/Audio with my TV remote. With only libcec tray I only get the poweron function, not the standby function (for the pc).

opdenkamp, 2015-07-02:
unfortunately this is not going to work without running a background service. like libCEC Tray, chir21's app, Kodi or EventGhost.

closed 2015-07-02 by opdenkamp

mattiasost, 2015-12-03:
Yes of course. But it does not work with libCEC Tray. Chir21's appp is a little unstable an unsupported.

opdenkamp, 2015-12-03:
You can do it with eventghost now: http://www.eventghost.org/
After (re)installing, install libCEC, then you'll get a plugin in there