Pulse-Eight/libcec #103: Additional CEC command needed to make active source audio work on my system

adq, 2015-06-15:
Hi. just upgraded my system to a Samsung TV (CEC 0.0.0.0), a Yamaha AV RVX377 receiver (CEC 1.0.0.0), and a pulse eight usb/hdmicec adaptor (CEC 1.1.0.0) attached to my media pc running xbmc under Arch linux. I'm using external 5.1 speakers on the yamaha, and never use the TV speakers.

Problem: video works fine, but I can't heard any sound on initial boot! Only way to force it to work is to manually turn the Yamaha AV off and on again once everything has started up. Then I can hear sound perfectly. 

Playing around, I found that the Yamaha appears to be routing sound to the TV speakers by default rather than its own speakers, and none of the "set active source" commands XBMC sends makes it change it.

I've been playing around this evening, and found if I send the command: 15:70:11:00, the yamaha switches its output from the TV to the speakers attached to it and all is good. (this is the CEC System Audio Mode Request command, telling it to use 1.1.0.0 (the cec adapter) as the source.

I don't know if this is a libcec or an xbmc problem (or just some weird combination of equipment). I did look at the XBMC source, but it doesn't seem to send raw CEC commands; just uses the C++ API exposed by this library, hence I'm posting here to start with.

I can obviously send this to my receiver manually on startup (I do now!), but thought I should open a ticket to report this corner case in case it is useful.

opdenkamp, 2015-07-02:
the problem is actually caused by the tv, which should send this command. are you sure that you've enabled the external speakers in the TV's settings?

jabbas, 2015-07-22:
I have exactly same issue, but with RX-V477 with libcec-3.0.1 and kodi 15rc2.

External speakers are enabled in the TV settings, volume change is working but there is no sound :(

opdenkamp, 2015-07-22:
Can you pastebin Kodi's debug log with component logging for the CEC 
component enabled.

jabbas, 2015-07-22:
I found an "temporary" solution: Set TV Speakers to the Receiver, but this makes sound routed from kodi to tv and from tv to the avr and it's reencoded by the TV, so even when i'm listenening to the stereo material i'm getting 5.1 stream to the avr.

Here's a log (with tv speakers set to external) - i've injected lines with additional information to easier find things (search for ^KODI): http://pastebin.com/ySNKvi96

jabbas, 2015-07-22:
Uhm... here's an correct logfile: http://pastebin.com/zX9V9AfM - i've added manually one line that shows when i'm powering off and on the avr.

jabbas, 2015-09-22:
Sending '15:70:11:00' is needed (at least in Samsung+Yamaha environment). Setting speakers to 'external' doesn't change anything...

malard, 2015-09-24:
it seems the AVR doesnt care which is the sender, given your sending it as LA1 does it work also if you send it as 05:70:11:00 ?

jabbas, 2015-09-24:
Yep, it doesn't care.

malard, 2015-09-24:
okay so perhaps we should add this @opdenkamp to our list of quirks modes?

jabbas, 2015-09-24:
Anyway... i've tested this with Kodi and the whole block which should powerup avr explicitly (and my code) isn't executed by kodi...
when doing `as` in cec-client it works like a charm... And i can't find any 'scheduling' in kodi logs.

jabbas, 2015-09-24:
Okie, i've checked... the AVR Isn't available while we are doing both things. Here's a few lines of log with log message added by me:

```
19:28:20 T:140577223993088   DEBUG: CecLogMessage - Recorder 1 (1) was already marked as active source
19:28:20 T:140577223993088   DEBUG: CecLogMessage - >> source activated: Recorder 1 (1)
19:28:20 T:140577223993088   DEBUG: CecLogMessage - JABBAS: audioSystem->IsPresent()? 0
19:28:20 T:140577223993088   DEBUG: CecLogMessage - sending active source message for 'Recorder 1'
19:28:20 T:140577223993088   DEBUG: CecLogMessage - << powering on 'TV' (0)
19:28:20 T:140577223993088   DEBUG: CecLogMessage - << 10:04
19:28:20 T:140577223993088   DEBUG: CecLogMessage - << Recorder 1 (1) -> broadcast (F): active source (1100)
19:28:20 T:140577223993088   DEBUG: CecLogMessage - << 1f:82:11:00
19:28:20 T:140577223993088   DEBUG: CecLogMessage - << Recorder 1 (1) -> TV (0): menu state 'activated'
19:28:20 T:140577223993088   DEBUG: CecLogMessage - << 10:8e:00
19:28:20 T:140577223993088   DEBUG: CecLogMessage - << powering on 'TV' (0)
19:28:20 T:140577223993088   DEBUG: CecLogMessage - << 10:04
19:28:20 T:140577223993088   DEBUG: CecLogMessage - CEC transmission - received response - TRANSMIT_FAILED_ACK
19:28:20 T:140577223993088   DEBUG: CecLogMessage - << 10:04
19:28:21 T:140577223993088 WARNING: CecLogMessage - HIGH_ERROR line:189 time:1871
19:28:21 T:140577223993088   DEBUG: CecLogMessage - << Recorder 1 (1) -> Audio (5): POLL
19:28:21 T:140577223993088   DEBUG: CecLogMessage - << 15
19:28:21 T:140577223993088   DEBUG: CecLogMessage - >> POLL sent
19:28:21 T:140577223993088   DEBUG: CecLogMessage - Audio (5): device status changed into 'present'
```

mk01, 2015-10-18:
@jabbas 

libCEC marks foreign device as 'present' only after a message is handled from it (or getstatus() was called, POLL sent and ACK received).

change audioSystem->IsPresent() for (audioSystem->GetStatus() == CEC_DEVICE_STATUS_PRESENT)

also comment out the whole block above (unconditional poweron of AVR).
```
/* * some AVRs fail to be powered up by the TV  ..
```

closed 2025-02-27 by github-actions[bot]