Pulse-Eight/libcec #104: cec doesn't work with no tv attached

djbijo, 2015-06-20:
i have a cec enabled avr connected to cubox-i with kodi.
i can see the key presses in kodi's log but they are ignored and i also can't control the avr's volume.
 i also see the error:

[code]
CecLogMessage - << Recorder 1 (1) -> TV (0): POLL
00:42:52 T:1487889472   DEBUG: CecLogMessage - << 10
00:42:52 T:1487889472   ERROR: CecLogMessage - Write: sent command error !
00:42:52 T:1487889472   DEBUG: CecLogMessage - << 10
00:42:52 T:1487889472   ERROR: CecLogMessage - Write: sent command error !
00:42:52 T:1487889472   DEBUG: CecLogMessage - >> POLL not sent
[/code]

all the time

opdenkamp, 2015-07-02:
the CEC spec says that the TV should support CEC too, and the code will keep trying to poll the TV for that reason.

I've changed the code to not treat a failed ack from the TV as error when sending an active source message, which will be included in tomorrow's 3.0.1 update.

no idea what xbian is using or if they've changed anything.

closed 2015-07-02 by opdenkamp

djbijo, 2015-07-04:
i realize that this is the spec but there is no reason why it shouldn't work without cec enabled tv.
you have the power to overcome the spec's bugs and at least add the option to disable tv polling so this will still work with only CEC enabled AVR.