Pulse-Eight/libcec #110: build issue on Linux, Cannot load libcec.so

kernelsmith, 2015-07-02:
What am I missing here?

```
git clone https://github.com/Pulse-Eight/platform.git
cd platform
mkdir build && cd build
cmake ..
make
sudo make install
cd ../../
git clone https://github.com/Pulse-Eight/libcec.git
cd libcec
mkdir build && cd build
cmake ..
make
sudo make install

$ cec-client
No device type given. Using 'recording device'
libcec.so.3.0: cannot open shared object file: No such file or directory
Cannot load libcec.so

$uname -a
# Linux 3.18.0-kali3-amd64 #1 SMP Debian 3.18.6-1~kali2(2015-03-02) x86_64 GNU/Linux

opdenkamp, 2015-07-02:
the problem is that cmake builds a static lib by default here. I've changed the default in the code that'll be used for tomorrow's 3.0.1 update

closed 2015-07-02 by opdenkamp

kernelsmith, 2015-07-02:
thanks @opdenkamp