Pulse-Eight/libcec #112: can't compile libcec 3.0.1 on mac os x yosemite

nijikon, 2015-07-09:
https://gist.github.com/nijikon/931cf1577fc79f1ecb2c

balrog:Formula n$ brew config
HOMEBREW_VERSION: 0.9.5
ORIGIN: git@github.com:polishgeeks/homebrew.git
HEAD: f762f8de97be31495326e100e8e0ad7f8bacd086
Last commit: 56 minutes ago
HOMEBREW_PREFIX: /usr/local
HOMEBREW_CELLAR: /usr/local/Cellar
HOMEBREW_BOTTLE_DOMAIN: https://homebrew.bintray.com
CPU: quad-core 64-bit haswell
OS X: 10.10.4-x86_64
Xcode: 6.4
CLT: 6.4.0.0.1.1435007323
Clang: 6.1 build 602
X11: 2.7.7 => /opt/X11
System Ruby: 2.0.0-p481
Perl: /usr/bin/perl
Python: /usr/local/bin/python => /usr/local/Cellar/python/2.7.10_1/Frameworks/Python.framework/Versions/2.7/bin/python2.7
Ruby: /Users/n/.rvm/rubies/ruby-2.2.2/bin/ruby
Java: 1.8.0_45

I can compile 3.0.0-repack without issues.

nijikon, 2015-07-10:
/cc @opdenkamp

opdenkamp, 2015-07-10:
some generated string is messed up from the looks of it. can you show me the contents of `/tmp/libcec20150709-57124-110w21t/libcec-libcec-3.0.1/src/libcec/env.h`

nijikon, 2015-07-10:
```
bash-3.2$ cat src/libcec/env.h
#pragma once
/*
 * WARNING: Auto-generated file from env.h.in
 *
 * This file is part of the libCEC(R) library.
 *
 * libCEC(R) is Copyright (C) 2011-2015 Pulse-Eight Limited.  All rights reserved.
 * libCEC(R) is an original work, containing original code.
 *
 * libCEC(R) is a trademark of Pulse-Eight Limited.
 *
 * This program is dual-licensed; you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation; either version 2 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program; if not, write to the Free Software
 * Foundation, Inc., 59 Temple Place - Suite 330, Boston, MA 02111-1307, USA.
 *
 *
 * Alternatively, you can license this library under a commercial license,
 * please contact Pulse-Eight Licensing for more information.
 *
 * For more information contact:
 * Pulse-Eight Licensing       <license@pulse-eight.com>
 *     http://www.pulse-eight.com/
 *     http://www.pulse-eight.net/
 */

#include "cectypes.h"
#include "platform/os.h"

#ifdef UNUSED
#elif defined(__GNUC__)
#define UNUSED(x) UNUSED_ ## x __attribute__((unused))
#elif defined(__LCLINT__)
#define UNUSED(x) /**/ x
#else
#define UNUSED(x) x
#endif

/* Define to 1 for xrandr support */
/* #undef HAVE_RANDR */

/* Define to 1 for lockdev support */
/* #undef HAVE_LOCKDEV */

/* Define to 1 for udev support */
/* #undef HAVE_LIBUDEV */

/* Define to 1 for Pulse-Eight CEC Adapter detection support */
/* #undef HAVE_P8_USB_DETECT */

/* Define to 1 for Raspberry Pi support */
/* #undef HAVE_RPI_API */

/* Define to 1 for TDA995x support */
/* #undef HAVE_TDA995X_API */

/* Define to 1 for Exynos support */
/* #undef HAVE_EXYNOS_API */

/* Define to 1 for nVidia EDID parsing support (on selected models) */
/* #undef HAVE_NVIDIA_EDID_PARSER */

/* Define to 1 for Python support */
#define HAVE_PYTHON 1

/* information about how libCEC was compiled */
#define LIB_INFO ("git revision: fatal: Not a git repository (or any of the parent directories): .git
-n, compiled on Fri Jul 10 15:04:22 UTC 2015 by n@balrog.local on Darwin 14.4.0 (x86_64), features: P8_USB")
```

opdenkamp, 2015-07-10:
`#define LIB_INFO ("git revision: fatal: Not a git repository (or any of the parent directories): .git
-n, compiled on Fri Jul 10 15:04:22 UTC 2015 by n@balrog.local on Darwin 14.4.0 (x86_64), features: P8_USB")`
guess I'll need to check for that ;-)
change your build script to not delete the .git directory, then it'll work until i've added a check for it

commit 2f32a9de (2015-07-10): fixed: don't generate an invalid env.h when not built with the .git dir present. closes #112
closed 2015-07-10 by opdenkamp