Pulse-Eight/libcec #113: "Open()" in Python wrapper resulting in "Segmentation fault"

Forage, 2015-07-14:
Libcec, the Python wrapper in particular, is having issues since [build #0703](http://forum.kodi.tv/showthread.php?tid=231092&pid=2044074#pid2044074) of OpenElec.
Since #0703 "self.lib.Open()" give me a "Segmentation fault" (line 80 in /usr/bin/pyCecClient.py) when trying to open a connection to a USB-CEC adapter on a RPi.

This build included quite a few libcec updates, as you can see in the post above, but I can't spot an obvious one resulting in this issue. Do you have an idea which one could be the cause of the problem or is there any way to get some more debug info to narrow it down?

MilhouseVH, 2015-07-14:
Is this a Kodi segfault, if so you should have a crash log.

If it's some other non-Kodi process that is faulting try enabling core dumping (` ulimit -c unlimited`) then once you have a core, try running it through gdb (`gdb /path/to/your/executable --core="<your core>" --batch -ex "thread apply all bt" 2>/dev/null`)

opdenkamp, 2015-07-14:
i haven't tested if this displays a proper message and exits correctly if a second program tries to open a connection, but i think that's what happening here. Kodi will be having an open connection to the adapter here, and then the second instance (python) will fail.

you'll have to disable CEC in Kodi for this to work.

but still, I'd like to see a stack trace of this so it can complain without crashing. thanks

Forage, 2015-07-14:
It happens when I run e.g. "python /usr/bin/pyCecClient.py" in through ssh so I don't think this qualifies as a Kodi segfault.

It's not a problem of a second connection being opened. Kodi had one open with the RPi CEC adapter. I'm connecting to an attached USB-CEC adapter (/dev/ttyACM0) which is disabled in the Kodi settings.
This has worked without issues until the update I referred to.

Does the method described by MilhouseVH provide the desired stack trace? I'll give that a go tomorrow if needed.

For what it's worth: if I do steal Kodi's connection to the RPi's adapter on purpose then I didn't get a segfault in the past either. Kodi looses it's connection but python is using it happily from there on. I'll check if this is still the case after the update tomorrow as well.

MilhouseVH, 2015-07-14:
> Does the method described by MilhouseVH provide the desired stack trace? I'll give that a go tomorrow if needed.

Yes. If you run:
```
ulimit -c unlimited
python /usr/bin/pyCecClient.py
```
you should end up with a core dump in `/storage/.cache/cores` once the process has seg faulted, eg. `core.!usr!bin!python2.7.1436913318.8747` (your name will be similar, but different).

Then:
```
gdb python --core=/storage/.cache/cores/core.!usr!bin!python2.7.1436913318.8747 --batch -ex "thread apply all bt"
```
should get you a stacktrace - not sure how useful the stacktraces will be as you're not using a debug build, but it's better than nothing (if you need a debug build, let me know - your FAT partition will need to be at least 384MB to use a debug build).

By the way,  /usr/bin/pyCecClient.py isn't crashing for me (tested on both Pi1 and Pi2) with build [#0714](http://forum.kodi.tv/showthread.php?tid=231092&pid=2052953#pid2052953), but then I don't use CEC (it's disabled in Kodi), or have any additional USB CEC dongles connected (does it still crash with the dongle unplugged?)

opdenkamp, 2015-07-14:
it may be a bug in the detection code. stack trace will tell. never tested it with both the pi's internal wire and a usb-cec either ;-)

Forage, 2015-07-15:
Here's the output of the gdb command: http://pastebin.com/nSgw58ns
Let met know if it's useful this way or if you need any more information.

When I disconnect the USB-CEC adapter then the same `pyCecClient.py` script will happily connect to the RPi's CEC adapter.

P.s. does `ulimit -c unlimited` need to be undone or is it not persistent?

MilhouseVH, 2015-07-15:
> P.s. does `ulimit -c unlimited` need to be undone or is it not persistent?

It's not persistent and only applies to that ssh session.

opdenkamp, 2015-07-19:
it's a bug in https://github.com/Pulse-Eight/libcec/commit/2ac430f0f2d68a338f4f08b774482741436f23ce

Forage, 2015-07-30:
I bet it's not the typo in line 49 of  src/libcec/platform/drm/drm-edid.cpp ;-P

anaconda, 2015-09-01:
`opendir()` returns `NULL` if the directory doesn't exist.
This should fix it:
https://clbin.com/sEYl5

MilhouseVH, 2015-09-02:
@Forage: I have included @anaconda's patch in the latest [#0901 test build](http://forum.kodi.tv/showthread.php?tid=231092&pid=2095660#pid2095660) - perhaps you can give it a try and report?

Forage, 2015-09-02:
@MilhouseVH thanks for including it to be able to take it for a spin.
@anaconda your fix works like a charm! Thank you!

opdenkamp, 2015-09-25:
@anaconda could you create a PR from that patch please, and sign the dev agreement, then we can include it: http://www.pulse-eight.net/contributors

thanks!

anaconda, 2015-09-25:
It seems you've merged a similar fix in #133, so this can be closed.

opdenkamp, 2015-09-25:
yeah sorry, same bug got fixed in there ;-)

closed 2015-09-25 by opdenkamp