Pulse-Eight/libcec #117: RPI: Won't compile with GCC 5.1.0, cmake 3.2.3

moonman, 2015-07-24:
Trying to compile v3.0.1 on Arch for RPI (all deps installed):
```
mkdir build
  cd build
  cmake -DCMAKE_INSTALL_PREFIX=/usr \
        -DBUILD_SHARED_LIBS=1 \
        -DHAVE_RPI_API=1 \
        -DRPI_INCLUDE_DIR=/opt/vc/include \
        -DRPI_LIB_DIR=/opt/vc/lib \
        -DRPI_BCM_HOST=0 \
        -DRPI_VCHIQ_ARM=1 \
        -DRPI_VCOS=1 \
        ..
  make
```
results in:

```
- The C compiler identification is GNU 5.1.0
-- The CXX compiler identification is GNU 5.1.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test SUPPORTS_CXX11
-- Performing Test SUPPORTS_CXX11 - Success
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Looking for initscr in curses
-- Looking for initscr in curses - found
-- Looking for dlopen in dl
-- Looking for dlopen in dl - found
-- Looking for clock_gettime in rt
-- Looking for clock_gettime in rt - found
-- Found PythonLibs: /usr/lib/libpython3.4m.so (found version "3.4.3") 
-- git found: <unknown>
-- Looking for include file lockdev.h
-- Looking for include file lockdev.h - found
-- Looking for dev_unlock in lockdev
-- Looking for dev_unlock in lockdev - found
-- Looking for udev_new in udev
-- Looking for udev_new in udev - found
-- Looking for 3 include files X11/Xlib.h, ..., X11/extensions/Xrandr.h
-- Looking for 3 include files X11/Xlib.h, ..., X11/extensions/Xrandr.h - found
-- Looking for XRRGetScreenResources in Xrandr
-- Looking for XRRGetScreenResources in Xrandr - found
-- Looking for include files tda998x_ioctl.h, comps/tmdlHdmiCEC/inc/tmdlHdmiCEC_Types.h
-- Looking for include files tda998x_ioctl.h, comps/tmdlHdmiCEC/inc/tmdlHdmiCEC_Types.h - not found
-- Found SWIG: /usr/bin/swig (found version "3.0.6") 
-- Configured features:
-- Pulse-Eight CEC Adapter:                yes
-- Pulse-Eight CEC Adapter detection:      yes
-- lockdev support:                        yes
-- xrandr support:                         yes
-- Raspberry Pi support:                   yes
-- TDA995x support:                        no
-- Exynos support:                         no
-- Python support:                         version 3.4.3 (3.4)
-- lib info: git revision: <unknown>, compiled on Fri Jul 24 07:37:07 UTC 2015 by oleg@alarmcbi on Linux 3.14.48-1-ARCH (armv7l), features: P8_USB, P8_detect, randr, 'RPi'
-- Configuring done
-- Generating done
-- Build files have been written to: /home/oleg/alarm/PKGBUILDs/alarm/libcec-rpi/src/libcec-libcec-3.0.1/build
Scanning dependencies of target cec
[  5%] [  5%] [  7%] [ 10%] Building CXX object src/libcec/CMakeFiles/cec.dir/CECProcessor.cpp.o
Building CXX object src/libcec/CMakeFiles/cec.dir/CECClient.cpp.o
Building CXX object src/libcec/CMakeFiles/cec.dir/LibCEC.cpp.o
Building CXX object src/libcec/CMakeFiles/cec.dir/LibCECC.cpp.o
[ 12%] Building CXX object src/libcec/CMakeFiles/cec.dir/adapter/RPi/RPiCECAdapterDetection.cpp.o
[ 15%] Building CXX object src/libcec/CMakeFiles/cec.dir/adapter/RPi/RPiCECAdapterCommunication.cpp.o
[ 17%] Building CXX object src/libcec/CMakeFiles/cec.dir/adapter/RPi/RPiCECAdapterMessageQueue.cpp.o
[ 20%] Building CXX object src/libcec/CMakeFiles/cec.dir/platform/drm/drm-edid.cpp.o
[ 23%] [ 25%] Building CXX object src/libcec/CMakeFiles/cec.dir/adapter/Pulse-Eight/USBCECAdapterMessage.cpp.o
Building CXX object src/libcec/CMakeFiles/cec.dir/adapter/AdapterFactory.cpp.o
[ 28%] Building CXX object src/libcec/CMakeFiles/cec.dir/adapter/Pulse-Eight/USBCECAdapterCommands.cpp.o
[ 30%] Building CXX object src/libcec/CMakeFiles/cec.dir/adapter/Pulse-Eight/USBCECAdapterCommunication.cpp.o
[ 33%] Building CXX object src/libcec/CMakeFiles/cec.dir/adapter/Pulse-Eight/USBCECAdapterMessageQueue.cpp.o
[ 35%] Building CXX object src/libcec/CMakeFiles/cec.dir/adapter/Pulse-Eight/USBCECAdapterDetection.cpp.o
[ 38%] Building CXX object src/libcec/CMakeFiles/cec.dir/devices/CECAudioSystem.cpp.o
[ 41%] Building CXX object src/libcec/CMakeFiles/cec.dir/devices/CECBusDevice.cpp.o
[ 43%] Building CXX object src/libcec/CMakeFiles/cec.dir/devices/CECDeviceMap.cpp.o
[ 46%] Building CXX object src/libcec/CMakeFiles/cec.dir/devices/CECPlaybackDevice.cpp.o
[ 48%] Building CXX object src/libcec/CMakeFiles/cec.dir/devices/CECRecordingDevice.cpp.o
[ 51%] Building CXX object src/libcec/CMakeFiles/cec.dir/devices/CECTuner.cpp.o
[ 53%] [ 56%] Building CXX object src/libcec/CMakeFiles/cec.dir/devices/CECTV.cpp.o
Building CXX object src/libcec/CMakeFiles/cec.dir/implementations/ANCommandHandler.cpp.o
[ 58%] Building CXX object src/libcec/CMakeFiles/cec.dir/implementations/CECCommandHandler.cpp.o
[ 61%] Building CXX object src/libcec/CMakeFiles/cec.dir/implementations/SLCommandHandler.cpp.o
[ 64%] Building CXX object src/libcec/CMakeFiles/cec.dir/implementations/VLCommandHandler.cpp.o
[ 66%] Building CXX object src/libcec/CMakeFiles/cec.dir/implementations/RLCommandHandler.cpp.o
[ 69%] Building CXX object src/libcec/CMakeFiles/cec.dir/implementations/PHCommandHandler.cpp.o
[ 71%] Building CXX object src/libcec/CMakeFiles/cec.dir/implementations/RHCommandHandler.cpp.o
[ 74%] Building CXX object src/libcec/CMakeFiles/cec.dir/implementations/AQCommandHandler.cpp.o
[ 76%] Building CXX object src/libcec/CMakeFiles/cec.dir/platform/adl/adl-edid.cpp.o
[ 79%] Building CXX object src/libcec/CMakeFiles/cec.dir/platform/nvidia/nv-edid.cpp.o
[ 82%] Building CXX object src/libcec/CMakeFiles/cec.dir/platform/posix/os-edid.cpp.o
[ 84%] Building CXX object src/libcec/CMakeFiles/cec.dir/platform/posix/serialport.cpp.o
make[2]: *** No rule to make target '1', needed by 'src/libcec/libcec.so.3.0.1'.  Stop.
make[2]: *** Waiting for unfinished jobs....
[ 87%] Building CXX object src/libcec/CMakeFiles/cec.dir/platform/X11/randr-edid.cpp.o
CMakeFiles/Makefile2:227: recipe for target 'src/libcec/CMakeFiles/cec.dir/all' failed
make[1]: *** [src/libcec/CMakeFiles/cec.dir/all] Error 2
Makefile:116: recipe for target 'all' failed
make: *** [all] Error 2
```

gutopa, 2015-07-31:
Any solution on that? I have the same problem.

[ 87%] Building CXX object src/libcec/CMakeFiles/cec.dir/platform/X11/randr-edid.cpp.o
make[2]: *** No rule to make target '1', needed by 'src/libcec/libcec.so.3.0.1'.  Stop.
CMakeFiles/Makefile2:227: recipe for target 'src/libcec/CMakeFiles/cec.dir/all' failed
make[1]: *** [src/libcec/CMakeFiles/cec.dir/all] Error 2
Makefile:113: recipe for target 'all' failed
make: *** [all] Error 2

moonman, 2015-07-31:
Take a look here: https://github.com/archlinuxarm/PKGBUILDs/blob/master/alarm/libcec-rpi/PKGBUILD

Unset ldflags, and get rid of some options

closed 2015-07-31 by moonman

EMJB, 2015-08-02:
I have just started to have the same problem when compiling on a clean build Pi1 or Pi2 using a script that I had used successfully a few days ago.  Don't think I have changed the script accidentally, so am assuming one of the libraries has been updated or something vital had been installed previously. Any suggestions as to how to proceed?

EMJB, 2015-08-17:
Rather than continue fighting this, I am using the ready-made deb at https://drgeoffathome.wordpress.com/2015/08/09/a-premade-libcec-deb/. Hope this helps someone else.