Pulse-Eight/libcec #123: Running cec-client disconnects tv/kodi

Holland1, 2015-08-07:
Hi 

I would like to use cec-client to start and shutdown the tv, but the problem is when I issue the cec-client command the link to kodi is broken, and therefore I can no longer use the remote control to control OSMC/Kodi. Only solution is to restart the kodi service.

These are the commands that I issue from within a script;
- 'echo "on 0" | cec-client -s'
- 'echo "standby 0" | cec-client -s'

Running OSMC July 2015 version (based on Kodi 15 and therefore libcec 3.0 ) raspberry 2B
TV is a samsung year 2014 model. UE50HU6900

mrDach, 2015-08-10:
Got the exact same issue. Also a RPi 2 with the same version of OSMC. My TV is a Samsung D7000 series.

patlux, 2015-08-18:
Use the cec-client in monitor mode (-m):

'echo "on 0" | cec-client -s -m'

Holland1, 2015-08-18:
Hi Patlux. I immediately tried you suggestion, but unfortunately the result is the same. 

osmc@osmc:~$ echo "standby 0" | cec-client -s -m
starting a monitor-only client. use 'mon 0' to switch to normal mode
opening a connection to the CEC adapter...
DEBUG:   [              69]     Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [              69]     InitHostCEC - vchiq_initialise succeeded
DEBUG:   [              69]     InitHostCEC - vchi_initialise succeeded
DEBUG:   [              69]     InitHostCEC - vchi_connect succeeded
DEBUG:   [              70]     Open - vc_cec initialised
DEBUG:   [              71]     logical address changed to Free use (e)
NOTICE:  [              71]     connection opened
DEBUG:   [              72]     processor thread started
DEBUG:   [              73]     << Broadcast (F) -> TV (0): POLL
DEBUG:   [              73]     initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [              73]     << e0
DEBUG:   [             116]     >> POLL sent
DEBUG:   [             116]     TV (0): device status changed into 'present'
DEBUG:   [             116]     << requesting vendor ID of 'TV' (0)
DEBUG:   [             117]     initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [             117]     << e0:8c
TRAFFIC: [             395]     >> 0f:87:00:00:f0
DEBUG:   [             395]     >> TV (0) -> Broadcast (F): device vendor id (87)
DEBUG:   [             395]     TV (0): vendor = Samsung (0000f0)
DEBUG:   [             396]     expected response received (87: device vendor id)
DEBUG:   [             396]     replacing the command handler for device 'TV' (0)
NOTICE:  [             396]     << putting 'TV' (0) in standby mode
DEBUG:   [             397]     initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [             397]     << e0:36
WARNING: [             545]     unhandled response received: opcode=87 initiator=e destination=f response=0
DEBUG:   [             605]     unregistering all CEC clients
DEBUG:   [             606]     unregistering all CEC clients
TRAFFIC: [            1014]     >> 0f:36
DEBUG:   [            1014]     TV (0): power status changed from 'unknown' to 'standby'
DEBUG:   [            1015]     >> TV (0) -> Broadcast (F): standby (36)
DEBUG:   [            1073]     UnregisterLogicalAddress - releasing previous logical address
DEBUG:   [            1074]     logical address changed to Broadcast (f)

osmc@osmc:~$ echo "on 0" | cec-client -s -m
starting a monitor-only client. use 'mon 0' to switch to normal mode
opening a connection to the CEC adapter...
DEBUG:   [              69]     Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [              69]     InitHostCEC - vchiq_initialise succeeded
DEBUG:   [              70]     InitHostCEC - vchi_initialise succeeded
DEBUG:   [              70]     InitHostCEC - vchi_connect succeeded
DEBUG:   [              71]     Open - vc_cec initialised
DEBUG:   [              71]     logical address changed to Free use (e)
NOTICE:  [              72]     connection opened
DEBUG:   [              72]     processor thread started
DEBUG:   [              73]     << Broadcast (F) -> TV (0): POLL
DEBUG:   [              73]     initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [              73]     << e0
DEBUG:   [             134]     >> POLL sent
DEBUG:   [             134]     TV (0): device status changed into 'present'
DEBUG:   [             134]     << requesting vendor ID of 'TV' (0)
DEBUG:   [             134]     initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [             134]     << e0:8c
TRAFFIC: [             339]     >> 0f:87:00:00:f0
DEBUG:   [             339]     TV (0): vendor = Samsung (0000f0)
DEBUG:   [             339]     expected response received (87: device vendor id)
DEBUG:   [             339]     replacing the command handler for device 'TV' (0)
NOTICE:  [             339]     << powering on 'TV' (0)
DEBUG:   [             339]     initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [             339]     << e0:04
DEBUG:   [             339]     >> TV (0) -> Broadcast (F): device vendor id (87)
DEBUG:   [             430]     TV (0): power status changed from 'unknown' to 'in transition from standby to on'
DEBUG:   [             430]     unregistering all CEC clients
DEBUG:   [             430]     unregistering all CEC clients
DEBUG:   [            1074]     UnregisterLogicalAddress - releasing previous logical address
DEBUG:   [            1075]     logical address changed to Broadcast (f)

opdenkamp, 2015-08-20:
This is not possible. Only one program can use the CEC wire at a time. to support this kind of thing, you'd have to write a "cec server" and "cec client", with Kodi and cec-client being clients for the server running libCEC. I had started preparations to support this kind of thing, but then decided that this is not something that belongs in libCEC.

closed 2015-08-20 by opdenkamp

Holland1, 2015-08-20:
One followup question. Just thinking out of the box. Would it be possible to add the CEC USB adapter to the setup. This way Kodi has a CEC link to both the Raspberry and the CEC USB adapter. I could then use the cec-client to send the startup and close down commands to the CEC USB adapter and I guess that the link between kodi and Raspberry remains intact.

opdenkamp, 2015-08-20:
this could work depending on what you want to do, but because only one logical address will be the active source and both devices get their own address, things like r/c passthrough will not work.

Holland1, 2015-08-20:
I would like to use both my tv remote control, to control OSMC (Kodi 15), and I also would to use the cec-client to be able to start and turn off the TV. I can use my remote control within OSMC (Kodi 15) up to the moment I fire a cec-client command. But that is, as you explained above not supported, given that only 1 program can use the CEC wire at the time.

So the setup is as follows: I would use two cec clients (raspberry/Kodi and the CEC USB adapter), one dedicated for the remote control function and the other only for cec-client requests.  Given that it was not completely clear what I wanted to achieve, and with the additional info I have just given, do you think this could work?

Alvast bedankt.

opdenkamp, 2015-08-20:
You should be using Kodi's API instead, calling this command: https://github.com/xbmc/xbmc/blob/master/xbmc/interfaces/Builtins.cpp#L227
There's a eventclient implementation on OpenELEC if you're using that, or you could use something like this: https://github.com/joshjowen/script.json-cec

Relys, 2015-10-14:
I also ran into this same issue. My TV remote works fine with OSMC's CEC support, but I also wanted to issue standby/on commands from my phone. I ended up installing dropbear ssh client on my phone and setting up a Tasker shortcut to issue an cec-client command over ssh to the kodi host which works fine (was a bit of a pain to setup). However, this kills the existing cec-client connection and renders the TV remote useless until reboot.

I will try using the script posted above that exposes cec commands from Kodi API through JASON-RPC API tonight. A much more eloquent solution. I would really like to see the CEC JASON-RPC script included in Kodi distribution. Then apps like Yatse could integrate by default. :)

Thanks for all your hard work opdenkamp & other devs! =D

cayfordb, 2016-02-12:
Hi.  I too want to do more with CEC commands from my Pi with Kodi/OpenElec.  I'm thinking, disable CEC on Kodi, and do everything from a script.service plugin that issues cec-client commands.   Is this reasonable?    In this case, the "everything" means based on keymap.xml buttons, sending "deck" commands (pause/play) to the current input device, Blueray player or FireTV, and "OSD Text" or "OSD Device Name" messages to the TV.  Crazy, or does this stand a chance?

mattie47, 2017-08-10:
Hi @opdenkamp 

Sorry to bring up an old topic but you seem to be the best one to ask this.

Re. your comment https://github.com/Pulse-Eight/libcec/issues/123#issuecomment-132986039

Is this still the case today with Kodi? Or is there now possibly a way to send cec-client commands without it killing the functionality within kodi?

I ask because I've been trying to remotely turn my TV on/off with hdmi-cec called via a node app, however as soon as I start the node app the tv remote stops working within kodi. Kodi log looks like the cec support is being terminated.

This is with kodi 17 on rpi3 running osmc with libCEC version: 4.0.2, git revision: libcec-4.0.2+30-8adc786~dirty,

Some people seem to be able to get both working together, yet I've sadly been unsuccessful :-( 

Thanks!

Matt

natterstefan, 2017-11-18:
Hi @mattie47, were you successfull in the meantime? I was not able to so far...

mattie47, 2017-11-19:
Hi @stweil,

Yes - I've had some (limited) success.

I did some further analysis into this problem and came up with a working solution (for me).

In short - Kodi takes control of the CECAdapter using LibCEC. When you use cec-client while Kodi is running, kodi no longer has control of the CEC Adapter. 

As you probably know, Kodi needs to be restarted to resume control of the adapter. 

The good news is Kodi has 3 built-in functions allowing us to control the CEC adapter. 

These are:

Function | Description
-- | --
CECActivateSource | Wake up playing device via a CEC peripheral
CECStandby | Put playing device on standby via a CEC peripheral
CECToggleState | Toggle state of playing device via a CEC peripheral

You can see other built in functions here - http://kodi.wiki/view/list_of_built-in_functions

So while these aren't exposed through a specific Kodi API as such, there is another Kodi plugin which allows you to send these through JSON-RPC:

https://github.com/joshjowen/script.json-cec

So with that plugin, you can go to a web address to Turn your TV on or off :-) 

For me (and a lot of users of Home Assistant and Home Bridge etc) the one killer thing really missing from the built-in functions is support for the cec-client "pow" command, which returns the TV's status. 

Hope that helps.

Thanks,

Matt

natterstefan, 2017-11-19:
@mattie47 thanks for the answer. Looks promising. I will check it out and try it, even though the first attempts did not work. The Pi answers (get the following response) but the TV doesn't act (eg. turn off).

[image]

Am I missing any setting (firewall) or something? Or should it work kind of out of the box?

mattie47, 2017-11-20:
Hi @natterstefan,

Will reply further on https://github.com/joshjowen/script.json-cec/issues/8  

Thanks,

Matt