Pulse-Eight/libcec #126: Sony KD-55W895 not seeing RPi as CEC Device

DaveTBlake, 2015-08-09:
Having problems with CEC from Sony TV (KD-55W895). Running Kodi 15.0 for OE 5.95.3 on RPi2 (hence libcec 3.0.0), but have also tried test builds with libecec 3.0.1

Kodi/OE sees the TV and displays a "Pulsecode 8 Adapter: TV connected" message when TV is powered up, but when my TV is searching for CEC enabled devices the RPi is not detected.

Connecting to a different HDMI port makes no difference.
New CEC enabled cable makes no difference.
Power cycling makes no difference, neither does trying different HDMI ports.
RPi as only HDMI devive makes no difference, no devices shown in list.
The TV does detect the presence of other possible CEC devices e,g when a Chromcast device is plugged in.

The test builds with libcec 3.0.1 were even worse - Pulsecode8 adpater connection message did not come up.
What information can I provide to help solve this?

opdenkamp, 2015-08-20:
can you pastebin a debug log please. this page explains how to generate it: http://kodi.wiki/view/Log_file/Easy

DaveTBlake, 2015-08-22:
Lars, a debug log is here http://goo.gl/x4rn18 .
I have tried 4 different HDMI cables now, but don't have another CEC TV to test the RPi hardware. Pulsecode8 adpater TV connection message comes up, but log reads like it does not see the TV at all, so confused. Thnaks for looking at this, would be so convenient to get CEC working.

opdenkamp, 2015-11-25:
I know it's been a while, but I'm checking the open reports now and the log has been deleted. Can you re-upload a log please. I've just tested master on a recent Sony and a Pi2 and I'm not seeing any issues.

DaveTBlake, 2015-11-26:
I will have to regenerate the logs, and I'm away from my system for a few days. But CEC still doesn't work so I would appreciate you looking at it. The Sony firmware has updated a few times, and I retest then just in case it has helped - it didn't. I have a new PVR connected to the Sony and CEC works, so it would seem the fault is not the Sony. That leaves the Pi or Kodi/OpenELEC

opdenkamp, 2016-02-26:
Since I can't reproduce this issue and I'm not seeing any problems with Sony, I'm closing this issue.

closed 2016-02-26 by opdenkamp

DaveTBlake, 2016-02-26:
Fair enough Lars. I have not had the chance to recreate the debug etc. and if I can't be bothered why should you! I have kind of given up on CEC, I use an android app as remote instead.  But if you ever get others with a Sony issue I would appreciate a heads up.

opdenkamp, 2016-02-26:
Well I've tested a recent (2015) model and an older model (2012 if I'm correct) with the code from Kodi's master and libCEC's master, and I'm not seeing any issues with this combination. The issue might have been fixed by now. Sony are not known for changing too many things in CEC implementations between various models.

DaveTBlake, 2016-02-26:
Yes maybe. I will eventually update when OE gets to Jarvis. But since everything else is working well I am not in a rush to change things. CEC to other devices from TV is fine, just RPi and Kodi. I will open a new thread if I eventually find a problem with the new version too, with fresh debug etc. Thanks.