Pulse-Eight/libcec #127: Guide for raspberry compilation

teopost, 2015-08-10:
Please,
   add a quick guide for raspberry pi compilation !

mrgreywater, 2015-08-10:
```
sudo apt-get install cmake g++-4.8 liblockdev1-dev libudev-dev libxrandr-dev python-dev swig
git clone --recursive git://github.com/Pulse-Eight/libcec.git
cd ~/libcec/src/platform
mkdir build
cd build
cmake -DCMAKE_CXX_COMPILER=g++-4.8 ..
make
sudo make install
cd ~/libcec
mkdir build
cd build
export LIBRARY_PATH=/opt/vc/lib
cmake -DCMAKE_CXX_COMPILER=g++-4.8 -DRPI_INCLUDE_DIR=/opt/vc/include -DRPI_LIB_DIR=/opt/vc/lib -DBUILD_SHARED_LIBS=1 -DCMAKE_INSTALL_PREFIX=/usr -DBUILD_SHARED_LIBS=1 ..
make -j4
sudo -E bash -c 'make install -j4'

```

conect, 2015-08-16:
Would be nice if somone could include this into the readme.

Aupajo, 2015-10-13:
I've added this as a wiki page for the mean-time: https://github.com/Pulse-Eight/libcec/wiki/Raspberry-Pi-set-up

opdenkamp, 2015-11-25:
I've updated the README.md file with more detailed instructions: https://github.com/Pulse-Eight/libcec/commit/2e880768f33b6b40b518ecc23fe9da1bb6d43bc3
Also updated the wiki page with these instructions

closed 2015-11-25 by opdenkamp