Pulse-Eight/libcec #128: STOP key not forwarded to eventghost.

yasuoiwakura, 2015-08-13:
hi,

im using libcec3 on my LG tv (simplink)
Keys working:
up, down, left, right, back/exit (same), play, pause, fwd, backward.

working in debug window but not in EventGhost:
stop
when i press stop, nothing happens
when i press stop+other key i get 
CEC..

now its getting interesting:
stop+pause=CEC.root.menu
stop+Play=CEC.. (is this even an allowed syntax?)
stop+other key=other key

well its interesting to have a 2nd binding for play+pause, but...
is it not possible to "just" use the stop-key?

when pressing stop, that happens in Debug(EventGhost closed):
[code]
TRAFFIC: [           17391]     >> 01:44:45
DEBUG:   [           17397]     >> TV (0) -> Recorder 1 (1): user control pressed (44)
TRAFFIC: [           17461]     >> 01:45
DEBUG:   [           17465]     >> TV (0) -> Recorder 1 (1): user control release (45)
DEBUG:   [           18461]     key auto-released: stop (45)
[/code]

compare PLAY button (working):
[code]
TRAFFIC: [           62211]     >> 01:44:44
DEBUG:   [           62211]     key pressed: play (44)
DEBUG:   [           62214]     >> TV (0) -> Recorder 1 (1): user control pressed (44)
TRAFFIC: [           62281]     >> 01:45
DEBUG:   [           62281]     key released: play (44)
DEBUG:   [           62283]     >> TV (0) -> Recorder 1 (1): user control release (45)
[/code]

yasuoiwakura, 2015-08-13:
[image]
[image]

opdenkamp, 2015-11-26:
there's a bug in the EventGhost plugin. it will not process the "combo key", which is used for remotes with a limited set of keys. it defaults to the stop button.

bug will be fixed in the 3.1.0 update

commit a39a5599 (2015-11-27): fixed: combo key (stop by default) in the EventGhost plugin. fixes #128
closed 2015-11-27 by opdenkamp