Pulse-Eight/libcec #129: support\build.cmd fails badly under Windows 10

ScottPKuehn, 2015-08-14:
There are multiple problems:

%BUILDDIR% is double quoted on each end of the string without the fix in generate-cmake.cmd
rem BUG FIX Below //IF NOT EXIST "%BUILDDIR%" MKDIR "%BUILDDIR%"
IF NOT EXIST %BUILDDIR% MKDIR %BUILDDIR%

Once that was fixed the cmake command fails in generate-cmake.cmd:
[image]

Obviously, the source directory path is really screwed up:
CMake Error: The source directory "C:/_Dev/Wellbeats 2.0/CECDeviceDriver/libcec-master/cmake-build/buildplatform64/2.0/CECDeviceDriver/libcec-master/build/x64" does not exist.

As you can see I added some echo statements to dump incoming parameters, etc.  This is as far as I got so far, there may be more issues.

I am on a very time-critical project.  A fix would be very much appreciated.

malard, 2015-08-14:
Hi, you can use the prebuilt binaries, what project is this for?

ScottPKuehn, 2015-08-14:
I am on a project that has an Intel Stick (http://www.intel.com/content/www/us/en/compute-stick/intel-compute-stick.html which connects directly to a TV on the HDMI port) driving a TV.  I need to control On/OFF, volume up/down, get current volume, get current on/off state, get active port, set active port.  The Stick will present scheduled and on-demand video content.  The Stick will be controlled over the internet (via SignalR) from virtually any device be it IOS, Windows 10, Windows 8.1, or android (PCs, tablets, cell phones).

 

Where are the prebuilt binaries?

 

Scott Kuehn

ScottPKuehn, 2015-08-14:
C# Windows 10 platform

malard, 2015-08-14:
Is this a commercial project? The windows binaries are available on our website.

ScottPKuehn, 2015-08-14:
Yes, it is a commercial project – a small start-up company.  I found the binaries, are there instructions on how to use them?  Installation?

ScottPKuehn, 2015-08-14:
Will your library work with the Intel Stick http://www.intel.com/content/www/us/en/compute-stick/intel-compute-stick.html ??

malard, 2015-08-14:
No, the stick does not have the CEC wire connected, commercial projects require a licence which comes with engineering support, please email us for details

ScottPKuehn, 2015-08-14:
I thought that your software was able to send CEC commands over HDMI... am I not understanding correctly?  Would the USB port on the Stick be useful for this purpose?

malard, 2015-08-14:
Yes, if the appropriate hardware connections are made and there is hardware to drive the pin. Our CEC adapter hardware products provide this capability, there are few products that provide this natively

ScottPKuehn, 2015-08-14:
We should be able to send CEC commands over HDMI... or am I wrong?

malard, 2015-08-14:
If you have the right hardware, Pin 13 of the HDMI socket must be connected to something!

ScottPKuehn, 2015-08-14:
So, what you are saying is that if the display supports CEC commands (pin 13) over HDMI, then libcec will do what I need?

malard, 2015-08-14:
the TV will support CEC, its your choice of hardware that wont! (i.e. the intel stick)

ScottPKuehn, 2015-08-14:
I have not seen the low level hardware specs of the Intel Stick.  Are you saying that the HDMI on the Stick does not have a pin 13?

malard, 2015-08-14:
yes, you need our hardware additionally, no normal PC hardware supports CEC natively

ScottPKuehn, 2015-08-14:
What additional hardware of yours are you referring to?

malard, 2015-08-14:
https://www.pulse-eight.com/

https://www.pulse-eight.com/p/104/usb-hdmi-cec-adapter

ScottPKuehn, 2015-08-17:
Can I get details on a commercial license?

malard, 2015-08-19:
yes, what is your email address, sorry i was out yesterday

ScottPKuehn, 2015-08-19:
ScottK@IntelligentSolutions.com

Slion, 2015-09-24:
Has anyone managed to build it on Windows 10? Would be useful for debugging libCECTray on Windows 10.

Slion, 2015-09-25:
Build on Windows 10 is working just fine following instructions from readme.

opdenkamp, 2015-11-25:
Am I correct that the issues have been resolved @Slion @ScottPKuehn ? Can I close this issue?

ScottPKuehn, 2015-11-25:
Issue was never resolved, but we are not going to use the library either.

 

Scott Kuehn

Text/Cell 952-884-0200

 <mailto:ScottK@IntelligentSolutions.com> ScottK@IntelligentSolutions.com

opdenkamp, 2015-11-25:
Okay, thanks for reporting back @ScottPKuehn 

@Slion in your last message you said `Build on Windows 10 is working just fine here. I just followed the instructions from readme.` so I'm closing this issue. Let me know if there are still any issues please.

closed 2015-11-25 by opdenkamp