Pulse-Eight/libcec #130: Compatibility Windows 10 Universal (Windows 10 Iot Core)

marnixcox, 2015-08-20:
SetupDiGetClassDevs is only supported on Windows 10 Desktop. 

For Windows 10 Universal (Windows 10 Iot Core -> Raspberry Pi) CM_Get_Device_ID_List / CM_Get_Device_Interface_List needs to be used.

https://msdn.microsoft.com/zh-tw/library/windows/hardware/ff551069(v=vs.85).aspx

opdenkamp, 2015-11-26:
oh lovely, it looks like the two may not be compatible with each other.

```
SetupDiGetClassDevs function
Target platform: Desktop
For universal, call CM_Get_Device_ID_List
For universal, call CM_Get_Device_Interface_List
```

```
CM_Get_Device_Interface_List function
Remarks: Use SetupDiGetClassDevs instead of CM_Get_Device_Interface_List
Target platform: Universal
```

opdenkamp, 2015-11-27:
luckily they are compatible, but I haven't found out how to read the registry key with the com port yet with one of the CM_* methods

opdenkamp, 2015-12-01:
this is going to be something for the next major update

davidGoff, 2016-01-26:
Hello, 
I'm a bit lost with these comments about windows iot.
Is there a work around to use libcec on this os? if not do you know if it is planned in a near future? 

I'm a C# dev with no specific knowledge in CEC, but I'll be glad to help you if there is something to test/ adapt. 

Best regards, 
David

opdenkamp, 2016-01-26:
I've replaced a couple of the winapi calls by the iot core compatible ones, but I couldn't find a method to read the registry entry that contains the com port that the system allocated for the device that's compatible with iot core when I looked at it earlier. That's in the C++ part: https://github.com/Pulse-Eight/libcec/blob/master/src/libcec/adapter/Pulse-Eight/USBCECAdapterDetection.cpp

I'll look into it before the next major update, but I can't give an ETA on that one at this time.

AndreaSpan, 2016-08-22:
Hi,

Anything new about using libcec on Windows 10 Universal (Windows 10 Iot Core)? 

Thank you!

DaveVoyles, 2016-10-05:
Following along, because I'd love to get this working on Win 10 IoT as well.

naice, 2016-10-20:
Also following. My MagicMirror is in Heavy need of switching On/Off its display. 

https://www.hackster.io/Emmuss/magicmirror-cb222b

gerritsen, 2017-03-20:
Any news on this? Waiting a long time on it. Hope it will be fixed soon!

jacobp85, 2017-03-28:
Any ETA?

opdenkamp, 2017-07-13:
Sorry, we haven't had time for this yet. It's a not a bug, but more like supporting a new operating system in this case.

opdenkamp, 2020-04-27:
.net core / uwp is now supported in master. it will be included in the next release.
we only provide installers for x86/x64 on Windows. I haven't looked at cross compiling for ARM yet

closed 2020-04-27 by opdenkamp