Pulse-Eight/libcec #132: Libcec 3.0.1 segfault on Ubuntu 15.04

rome1983, 2015-09-05:
libcec segfaults when cec adapter plugged in on boot  and kodi doesn't start. If I unplug cec adapter kodi starts. If I plug the cec adapter in after kodi starts it works, but the first time i did it kodi crashed. I have 2 pulse-eight adapters one internal and the other external getting the same results. All was fine before update.
dmesg http://paste.ubuntu.com/12266130/
kodi log http://paste.ubuntu.com/12266142/
crash log http://paste.ubuntu.com/12266171/

rome1983, 2015-09-05:
Just installed openelec 5.95.4 and it works as expected. Openelec version of libcec is 3.0.0.

rome1983, 2015-09-08:
Any help would be appreciated.

ravens, 2015-09-20:
I have the same issue since 3 weeks on my Kodi running on a x86 14.04 box. Kodi is crashing in a loop following a segfault in libcec when the dongle is connected.

Sep 20 20:56:39 nirvana kernel: [103438.581489] usb 2-1.3: new full-speed USB device number 10 using ehci-pci
Sep 20 20:56:39 nirvana kernel: [103438.676927] usb 2-1.3: New USB device found, idVendor=2548, idProduct=1002
Sep 20 20:56:39 nirvana kernel: [103438.676935] usb 2-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep 20 20:56:39 nirvana kernel: [103438.676939] usb 2-1.3: Product: USB-CEC Adapter
Sep 20 20:56:39 nirvana kernel: [103438.676942] usb 2-1.3: Manufacturer: Pulse-Eight
Sep 20 20:56:39 nirvana kernel: [103438.676946] usb 2-1.3: SerialNumber: v2 r8
Sep 20 20:56:39 nirvana kernel: [103438.677865] cdc_acm 2-1.3:1.0: ttyACM0: USB ACM device
Sep 20 20:56:39 nirvana kernel: [103438.680700] input: Pulse-Eight USB-CEC Adapter as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.3/2-1.3:1.2/0003:2548:1002.0006/input/input14
Sep 20 20:56:39 nirvana kernel: [103438.681111] hid-generic 0003:2548:1002.0006: input,hidraw0: USB HID v1.10 Mouse [Pulse-Eight USB-CEC Adapter] on usb-0000:00:1d.0-1.3/input2
Sep 20 20:56:43 nirvana kernel: [103442.386885] CECAdapter[20762]: segfault at 7f3e06ff6000 ip 00007f3e5dce05f9 sp 00007f3e06ff3460 error 4 in libcec.so.3.0.1[7f3e5dc98000+8c000]
Sep 20 20:56:43 nirvana lircd-0.9.0[794]: removed client
Sep 20 20:56:43 nirvana kernel: [103442.569889] [drm:cpt_set_fifo_underrun_reporting] *ERROR* uncleared pch fifo underrun on pch transcoder B
Sep 20 20:56:43 nirvana kernel: [103442.571658] [drm:cpt_serr_int_handler] *ERROR* PCH transcoder B FIFO underrun
Sep 20 20:56:43 nirvana acpid: client 20663[0:0] has disconnected
Sep 20 20:56:44 nirvana kernel: [103443.427704] init: kodi main process ended, respawning
Sep 20 20:56:44 nirvana console-kit-daemon[806]: GLib-CRITICAL: Source ID 718 was not found when attempting to remove it
Sep 20 20:56:44 nirvana acpid: client connected from 20792[0:0]
Sep 20 20:56:44 nirvana acpid: 1 client rule loaded
Sep 20 20:56:45 nirvana lircd-0.9.0[794]: accepted new client on /run/lirc/lircd
Sep 20 20:56:45 nirvana kernel: [103444.159930] CECAdapter[20809]: segfault at 7ff531311000 ip 00007ff53276b5f9 sp 00007ff53130e460 error 4 in libcec.so.3.0.1[7ff532723000+8c000]
Sep 20 20:56:45 nirvana lircd-0.9.0[794]: removed client
Sep 20 20:56:46 nirvana kernel: [103445.189506] init: kodi main process ended, respawning
Sep 20 20:56:46 nirvana console-kit-daemon[806]: GLib-CRITICAL: Source ID 734 was not found when attempting to remove it
Sep 20 20:56:46 nirvana acpid: client 20792[0:0] has disconnected
Sep 20 20:56:46 nirvana acpid: client connected from 20842[0:0]
Sep 20 20:56:46 nirvana acpid: 1 client rule loaded
Sep 20 20:56:46 nirvana lircd-0.9.0[794]: accepted new client on /run/lirc/lircd
Sep 20 20:56:46 nirvana kernel: [103445.901735] CECAdapter[20859]: segfault at 7f98dedf9000 ip 00007f98ec39e5f9 sp 00007f98dedf6460 error 4 in libcec.so.3.0.1[7f98ec356000+8c000]
Sep 20 20:56:46 nirvana lircd-0.9.0[794]: removed client
Sep 20 20:56:47 nirvana kernel: [103446.929092] init: kodi main process ended, respawning
Sep 20 20:56:47 nirvana console-kit-daemon[806]: GLib-CRITICAL: Source ID 751 was not found when attempting to remove it
Sep 20 20:56:48 nirvana acpid: client 20842[0:0] has disconnected
Sep 20 20:56:48 nirvana acpid: client connected from 20892[0:0]
Sep 20 20:56:48 nirvana acpid: 1 client rule loaded
Sep 20 20:56:48 nirvana lircd-0.9.0[794]: accepted new client on /run/lirc/lircd
Sep 20 20:56:48 nirvana kernel: [103447.622815] CECAdapter[20909]: segfault at 7fde6c89e000 ip 00007fde6dcf85f9 sp 00007fde6c89b460 error 4 in libcec.so.3.0.1[7fde6dcb0000+8c000]
Sep 20 20:56:48 nirvana lircd-0.9.0[794]: removed client

rome1983, 2015-09-20:
After a few fresh installs it finally started working again. Still don't know what the problem was.

ravens, 2015-09-20:
Reinstalled the libcec package from the PPA, no effect.

rome1983, 2015-09-20:
I was reinstalling the whole os. I use Ubuntu 15.04 server and the xbmc stable ppa.

opdenkamp, 2015-09-25:
fix can be found here and will be merged in in the next update: https://github.com/Pulse-Eight/libcec/pull/133

closed 2015-09-25 by opdenkamp