Pulse-Eight/libcec #138: TV not powering on

Slion, 2015-09-25:
I'm trying to get my Philips TV 55PFL7606K to power on from standby but it's not been working.
I've just been reusing the code from CecSharpClient and I simply did:
`Lib.PowerOnDevices(CecLogicalAddress.Broadcast);`

Here are the logs I get from the power on request:
http://pastebin.com/wi9Lcpy8
Those logs also include myself power on the TV manually since it did not work through CEC :)
Is power on not working with this TV hardware?
Do I need to specify the HDMI port?
When I call `PowerOnDevices` and the TV is already on it switches to another HDMI input. Possibly HDMI 1. The HTPC is connected on HDMI 2.

opdenkamp, 2015-09-25:
You indeed have to pass the correct HDMI port number libCEC by setting it in `LibCECConfiguration` when initialising.

For powering on the TV and switching, use `SetActiveSource()` instead

closed 2015-09-25 by opdenkamp

Slion, 2015-09-25:
I did try with HDMI port set to 2 for HDMI 2 like on my libCECTray setup but even then calling PowerOnDevices when already on keept switching to some other input.

opdenkamp, 2015-09-25:
`PowerOnDevices()` will not do any source switching. It will just send a power on command to the specified devices. Because no device is marked as active source when you start libCEC and don't use `SetActiveSource()`, the TV will switch to whatever the devs of the TV's firmware implemented.

So, simply put: you're using the wrong API command

Slion, 2015-09-25:
Thanks for helping out that got me on the right track.
`CecSharpClient` sets the default config device type to "recording", switching it TV fixed it all.
`Config.DeviceTypes.Types[0] = CecDeviceType.Tv;`
Now `PowerOnDevices` does stay on HDMI 2 as specified and it actually brings back the power from standby.
Awesome!

opdenkamp, 2015-09-25:
That's not right, that makes libCEC allocate a logical address for TV, which will fail as there can be only one TV, and then libCEC likely sends a command with an "unregistered" address. Use `SetActiveSource()` instead of `PowerOnDevices()` if you want the TV to switch to the correct input.

Slion, 2015-09-25:
Let me explain what I'm implementing.
I want the TV to turn on and off as Windows turns the monitor on and off.
Without CEC support although Windows turns off the monitor the TV remains on, displaying a message saying there is no video signal.

I don't even want to change the TV source, however even when just starting libCECTray it would switch to HDMI 1 until I specified the correct port which in my case is HDMI 2. My application would behave the same when issuing `PowerOnDevices` until I fixed the device type in my config and set the proper HDMI port. I do not know if `PowerOnDevices` with proper TV device type and wrong HDMI port 1 would also switch to that port. I'll have to test that.

Now the only reason I was calling `PowerOnDevices` while the TV was already ON was that when you register for Windows monitor power notifications it first sends you one with the current state of the monitor which in my case was ON and that would result in me calling `PowerOnDevices`.
See my code there:
http://hg.slions.net/hgweb.cgi/SharpDisplayManager/file/de942d321cfb/PowerManager/PowerManager.cpp#l47
and around there:
http://hg.slions.net/hgweb.cgi/SharpDisplayManager/file/7b19bea5d73a/Server/MainForm.Hid.cs#l101

All I intend to right now is just power on and off the TV without changing the source.
Now with my fixed up config `PowerOnDevices` and `StandbyDevices` seems to be doing the job just fine. Are you saying I should really use `SetActiveSource` rather than `PowerOnDevices` for my use case? 

Next I'll need to try and make my config adjustable by the user so that the app will work in different setup. That will involve removing the hard coded HDMI port, I guess I could leave the hard coded TV device type since anything else probably does not make sense for my use case.

Thanks for your support.