Pulse-Eight/libcec #141: Please replace use of lockdev with flock

rbalint, 2015-09-28:
Lockdev is being removed from Debian unstable and Ubuntu will stop shipping it eventually, too, while flock is the recommended replacement.
For more information please see:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=728023

Packages available in Debian are built without liblockdev.

opdenkamp, 2015-11-25:
ugh, then current debian packages are broken. do _not_ ship without locking in place please as it will lead to issues when someone uses Kodi and cec-client at the same time, breaking both (which a lot of people tried in the past)

as for the change, I'll take a look at getting it replaced for the upcoming release, thanks

opdenkamp, 2015-11-25:
Is `flock` available on all Unix flavours, so not just Linux?

opdenkamp, 2015-11-25:
nvm, seems to be in libc ;-)

rbalint, 2015-11-26:
Thanks for looking into this. If my understanding is correct using Kodi only works properly and most users only use libcec with Kodi.
I think in this situation (lockdev already scheduled for removal and flock is not yet used) the best short term solution is documenting the problem in Debian's BTS, too.
Which bug report can I use as reference for the Debian bug? I don't have any CEC compatible device thus I can't reproduce the problem myself.
I would happily apply the flock patch in Debian even before libCEC 3.1 is released.

opdenkamp, 2015-11-26:
> Thanks for looking into this. If my understanding is correct using Kodi only works properly and most users only use libcec with Kodi.

Most users yes, but there's always some users who try to access the adapter with cec-client while Kodi is running. There have been multiple (now closed as invalid) reports here of people trying it and complaining it doesn't work. Which is to be expected.

> I think in this situation (lockdev already scheduled for removal and flock is not yet used) the best short term solution is documenting the problem in Debian's BTS, too. Which bug report can I use as reference for the Debian bug?

Sure, feel free to document it there, but I'm not sure what reference you're asking for?

> I would happily apply the flock patch in Debian even before libCEC 3.1 is released.

There is no patch yet, but it'll be part of the 3.1 release, which should be ready this week or next week. But Debian should wait a little longer before removing things that are still in use instead of disabling it and shipping a broken lib. I thought Debian was all about not shipping the latest greatest untested and patched version, but actually shipping things that work and are tested?

closed 2015-11-27 by opdenkamp

opdenkamp, 2015-11-27:
patch is attached @rbalint

rbalint, 2015-11-28:
@opdenkamp Thank you. I have just updated the package with the patch in unstable and will update in jessie-backports when the package enters testing.

rbalint, 2015-12-08:
@opdenkamp libcec in testing and jessie-backports are fixed now, too

commit ff86c07a (2015-11-27): changed: replaced lockdev by flock. closes #141