Pulse-Eight/libcec #142: Suddenly TV can't be controled via CEC any more

mattbue, 2015-09-28:
Hi, i'm using libcec since about half a year to control a Samsung TV.
Since last Weekend, it suddenly stopped working.

There were no changes on the System (no updates).
On the TV i already made a factory reset without any changes...

Here the output of the try to switch the TV on:

root@raspberrypi:~# sudo echo "on 0" | cec-client -s
opening a connection to the CEC adapter...
DEBUG:   [              82]     Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [              83]     InitHostCEC - vchiq_initialise succeeded
DEBUG:   [              83]     InitHostCEC - vchi_initialise succeeded
DEBUG:   [              83]     InitHostCEC - vchi_connect succeeded
DEBUG:   [              85]     Open - vc_cec initialised
DEBUG:   [              85]     logical address changed to Free use (e)
NOTICE:  [              86]     connection opened
DEBUG:   [              87]     processor thread started
DEBUG:   [              87]     << Broadcast (F) -> TV (0): POLL
DEBUG:   [              88]     initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [              88]     << e0
DEBUG:   [             149]     >> POLL sent
DEBUG:   [             150]     TV (0): device status changed into 'present'
DEBUG:   [             150]     << requesting vendor ID of 'TV' (0)
TRAFFIC: [             151]     << e0:8c
DEBUG:   [            1212]     expected response not received (87: device vendor id)
TRAFFIC: [            1213]     << e0:8c
DEBUG:   [            2274]     expected response not received (87: device vendor id)
NOTICE:  [            2275]     registering new CEC client - v3.0.0
DEBUG:   [            2276]     detecting logical address for type 'recording device'
DEBUG:   [            2276]     trying logical address 'Recorder 1'
DEBUG:   [            2277]     << Recorder 1 (1) -> Recorder 1 (1): POLL
TRAFFIC: [            2277]     << 11
TRAFFIC: [            2368]     << 11
DEBUG:   [            2459]     >> POLL not sent
DEBUG:   [            2460]     using logical address 'Recorder 1'
DEBUG:   [            2461]     Recorder 1 (1): device status changed into 'handled by libCEC'
DEBUG:   [            2461]     Recorder 1 (1): power status changed from 'unknown' to 'on'
DEBUG:   [            2462]     Recorder 1 (1): vendor = Pulse Eight (001582)
DEBUG:   [            2462]     Recorder 1 (1): CEC version 1.4
DEBUG:   [            2463]     AllocateLogicalAddresses - device '0', type 'recording device', LA '1'
DEBUG:   [            2465]     logical address changed to Recorder 1 (1)
DEBUG:   [            2465]     Recorder 1 (1): osd name set to 'CECTester'
DEBUG:   [            2466]     Recorder 1 (1): menu language set to 'eng'
DEBUG:   [            2467]     GetPhysicalAddress - physical address = 2000
DEBUG:   [            2467]     AutodetectPhysicalAddress - autodetected physical address '2000'
DEBUG:   [            2468]     Recorder 1 (1): physical address changed from ffff to 2000
DEBUG:   [            2469]     << Recorder 1 (1) -> broadcast (F): physical adddress 2000
TRAFFIC: [            2469]     << 1f:84:20:00:01
NOTICE:  [            2620]     CEC client registered: libCEC version = 3.0.0, client version = 3.0.0, firmware version = 1, logical address(es) = Recorder 1 (1) , physical address: 2.0.0.0, compiled on Linux-3.18.11-v7+, features: P8_USB, P8_detect, randr, 'RPi'
DEBUG:   [            2621]     << Recorder 1 (1) -> TV (0): OSD name 'CECTester'
TRAFFIC: [            2622]     << 10:47:43:45:43:54:65:73:74:65:72
DEBUG:   [            2923]     << requesting power status of 'TV' (0)
TRAFFIC: [            2923]     << 10:8f
DEBUG:   [            3984]     expected response not received (90: report power status)
TRAFFIC: [            3985]     << 10:8f
DEBUG:   [            5047]     expected response not received (90: report power status)
DEBUG:   [            5048]     << requesting vendor ID of 'TV' (0)
TRAFFIC: [            5049]     << 10:8c
DEBUG:   [            6111]     expected response not received (87: device vendor id)
TRAFFIC: [            6112]     << 10:8c
DEBUG:   [            7173]     expected response not received (87: device vendor id)
NOTICE:  [            7174]     << powering on 'TV' (0)
TRAFFIC: [            7175]     << 10:04
DEBUG:   [            7236]     TV (0): power status changed from 'unknown' to 'in transition from standby to on'
DEBUG:   [            7237]     unregistering all CEC clients
NOTICE:  [            7238]     unregistering client: libCEC version = 3.0.0, client version = 3.0.0, firmware version = 1, logical address(es) = Recorder 1 (1) , physical address: 2.0.0.0, compiled on Linux-3.18.11-v7+, features: P8_USB, P8_detect, randr, 'RPi'
DEBUG:   [            7238]     Recorder 1 (1): power status changed from 'on' to 'unknown'
DEBUG:   [            7239]     Recorder 1 (1): vendor = Unknown (000000)
DEBUG:   [            7240]     Recorder 1 (1): CEC version unknown
DEBUG:   [            7240]     Recorder 1 (1): osd name set to 'Recorder 1'
DEBUG:   [            7241]     Recorder 1 (1): device status changed into 'unknown'
DEBUG:   [            7241]     unregistering all CEC clients
DEBUG:   [            8089]     UnregisterLogicalAddress - releasing previous logical address
DEBUG:   [            8090]     logical address changed to Broadcast (f)

Does someone has an idea why it is not possible to control the TV anymore?

Slion, 2015-09-29:
Which API call are you using to turn on your TV?

mattbue, 2015-09-29:
Until now i used 'sudo echo "on 0" | cec-client -s' to turn on the TV.

Slion, 2015-09-29:
Alright I'm new here and actually never used cec-client command line but I did use libcec on Windows to turn on and off my Philips TV (2011).
Looking at the code it looks like "cec-client on 0" should do it.
Though you may have to specify the HDMI port you are connected on to on your TV.
Not how that is done using cec-client.

Slion, 2015-10-03:
I had a similar issue though in my case I could not establish a CEC connection at all.
Cold start both TV and computer somehow fixed the issue.
See https://github.com/Pulse-Eight/libcec/issues/144.

opdenkamp, 2015-11-27:
I don't see libCEC doing anything wrong here. It's sending the correct commands, but your TV doesn't seem to be responding. It's only responding to polls, not a single other command.

Try power cycling the TV, as @Slion suggested and, if that doesn't work, try replacing the HDMI cable or upgrading the TV's firmware to the latest version

closed 2015-11-27 by opdenkamp