Pulse-Eight/libcec #143: Onkyo Reciever Shuts Off a few seconds after being woken up from resume

josh4trunks, 2015-09-29:
**OS** - OpenELEC 5.95.5 x64
**CPU/Motherboard** - Asrock J1900
**CEC** - Pulse-Eight USB adapter with v4 firmware
**Receiver** - Onkyo TX-SR309
**TV** - Panasonic TC-50PU54

**Description**
When I wakeup my openelec box from suspend I wait a few seconds until I hear the click of the TV and reciever being woken from CEC. *But, after a few more seconds the receiever shuts off.*
This was not a problem with version 5.0.8

**Debug Log**
```﻿
19:59:53 T:140583711201152  NOTICE: special://profile/ is mapped to: special://masterprofile/
19:59:53 T:140583711201152  NOTICE: -----------------------------------------------------------------------
19:59:53 T:140583711201152  NOTICE: Starting Kodi (15.1 Git:f4dda26). Platform: Linux x86 64-bit
19:59:53 T:140583711201152  NOTICE: Using Release Kodi x64 build
19:59:53 T:140583711201152  NOTICE: Kodi compiled Sep  4 2015 by GCC 4.9.3 for Linux x86 64-bit version 4.1.6 (262406)
19:59:53 T:140583711201152  NOTICE: Running on OpenELEC (official) - Version: 5.95.5 6.0, kernel: Linux x86 64-bit version 4.1.6
19:59:53 T:140583711201152  NOTICE: FFmpeg version: 2.6.4
19:59:53 T:140583711201152  NOTICE: WARNING: unsupported ffmpeg version detected
19:59:53 T:140583711201152  NOTICE: Host CPU: Intel(R) Celeron(R) CPU J1800 @ 2.41GHz, 2 cores available
19:59:53 T:140583711201152  NOTICE: special://xbmc/ is mapped to: /usr/share/kodi
19:59:53 T:140583711201152  NOTICE: special://xbmcbin/ is mapped to: /usr/lib/kodi
19:59:53 T:140583711201152  NOTICE: special://masterprofile/ is mapped to: /storage/.kodi/userdata
19:59:53 T:140583711201152  NOTICE: special://home/ is mapped to: /storage/.kodi
19:59:53 T:140583711201152  NOTICE: special://temp/ is mapped to: /storage/.kodi/temp
19:59:53 T:140583711201152  NOTICE: The executable running is: /usr/lib/kodi/kodi.bin
19:59:53 T:140583711201152  NOTICE: Local hostname: OpenELEC
19:59:53 T:140583711201152  NOTICE: Log File is located: /storage/.kodi/temp/kodi.log
19:59:53 T:140583711201152  NOTICE: -----------------------------------------------------------------------
19:59:53 T:140583711201152   ERROR: DBus: Error org.freedesktop.DBus.Error.ServiceUnknown - The name org.freedesktop.UPower was not provided by any .service files
19:59:53 T:140583711201152  NOTICE: load settings...
19:59:53 T:140583711184640  NOTICE: Thread FDEventMonitor start, auto delete: false
19:59:53 T:140583711201152  NOTICE: Found 1 Lists of Devices
19:59:53 T:140583711201152  NOTICE: Enumerated ALSA devices:
19:59:53 T:140583711201152  NOTICE:     Device 1
19:59:53 T:140583711201152  NOTICE:         m_deviceName      : @
19:59:53 T:140583711201152  NOTICE:         m_displayName     : Default (HDA Intel PCH ALC662 rev1 Analog)
19:59:53 T:140583711201152  NOTICE:         m_displayNameExtra:
19:59:53 T:140583711201152  NOTICE:         m_deviceType      : AE_DEVTYPE_PCM
19:59:53 T:140583711201152  NOTICE:         m_channels        : FL,FR
19:59:53 T:140583711201152  NOTICE:         m_sampleRates     : 48000
19:59:53 T:140583711201152  NOTICE:         m_dataFormats     : AE_FMT_S32NE
19:59:53 T:140583711201152  NOTICE:     Device 2
19:59:53 T:140583711201152  NOTICE:         m_deviceName      : @:CARD=PCH,DEV=0
19:59:53 T:140583711201152  NOTICE:         m_displayName     : HDA Intel PCH
19:59:53 T:140583711201152  NOTICE:         m_displayNameExtra: ALC662 rev1 Analog
19:59:53 T:140583711201152  NOTICE:         m_deviceType      : AE_DEVTYPE_PCM
19:59:53 T:140583711201152  NOTICE:         m_channels        : FL,FR
19:59:53 T:140583711201152  NOTICE:         m_sampleRates     : 48000
19:59:53 T:140583711201152  NOTICE:         m_dataFormats     : AE_FMT_S32NE
19:59:53 T:140583711201152  NOTICE:     Device 3
19:59:53 T:140583711201152  NOTICE:         m_deviceName      : hdmi:CARD=PCH,DEV=0
19:59:53 T:140583711201152  NOTICE:         m_displayName     : HDA Intel PCH
19:59:53 T:140583711201152  NOTICE:         m_displayNameExtra: ONK TX-SR309 on HDMI
19:59:53 T:140583711201152  NOTICE:         m_deviceType      : AE_DEVTYPE_HDMI
19:59:53 T:140583711201152  NOTICE:         m_channels        : FL,FR,LFE,FC,BL,BR
19:59:53 T:140583711201152  NOTICE:         m_sampleRates     : 32000,44100,48000,88200,96000,176400,192000
19:59:53 T:140583711201152  NOTICE:         m_dataFormats     : AE_FMT_LPCM,AE_FMT_AC3,AE_FMT_DTS,AE_FMT_EAC3,AE_FMT_DTSHD,AE_FMT_TRUEHD,AE_FMT_S32NE,AE_FMT_S16NE,AE_FMT_S16LE,AE_FMT_AAC
19:59:53 T:140583711201152  NOTICE:     Device 4
19:59:53 T:140583711201152  NOTICE:         m_deviceName      : hdmi:CARD=PCH,DEV=1
19:59:53 T:140583711201152  NOTICE:         m_displayName     : HDA Intel PCH
19:59:53 T:140583711201152  NOTICE:         m_displayNameExtra: HDMI
19:59:53 T:140583711201152  NOTICE:         m_deviceType      : AE_DEVTYPE_HDMI
19:59:53 T:140583711201152  NOTICE:         m_channels        : FL,FR,LFE,FC,BC,BL,BR,FLOC,FROC,BLOC,BROC,TFC,TC,TFL,TFR,UNKNOWN1
19:59:53 T:140583711201152  NOTICE:         m_sampleRates     : 32000,44100,48000,88200,96000,176400,192000
19:59:53 T:140583711201152  NOTICE:         m_dataFormats     : AE_FMT_S32NE,AE_FMT_S16NE,AE_FMT_S16LE,AE_FMT_LPCM,AE_FMT_DTSHD,AE_FMT_TRUEHD,AE_FMT_EAC3,AE_FMT_DTS,AE_FMT_AC3,AE_FMT_AAC
19:59:53 T:140583711201152  NOTICE: Loaded settings file from special://xbmc/system/advancedsettings.xml
19:59:53 T:140583711201152  NOTICE: Contents of special://xbmc/system/advancedsettings.xml are...
                                            <?xml version="1.0" encoding="UTF-8" ?>
[... 1416 lines cut ...]
20:00:50 T:140583465117440  NOTICE: Previous line repeats 4 times.
20:00:50 T:140583465117440   DEBUG: CecLogMessage - >> 50:8f
20:00:50 T:140583465117440   DEBUG: CecLogMessage - >> 05:90:00
20:00:50 T:140582323193600  NOTICE: EMBY LibrarySync -> DB Version: 1.1.39
20:00:59 T:140582903166720  NOTICE: Previous line repeats 8 times.
20:00:59 T:140582903166720   DEBUG: Thread JobWorker 140582903166720 terminating (autodelete)
20:00:59 T:140582348371712   DEBUG: Thread JobWorker 140582348371712 terminating (autodelete)
20:00:59 T:140582877988608   DEBUG: Thread JobWorker 140582877988608 terminating (autodelete)
20:00:59 T:140582323193600  NOTICE: EMBY LibrarySync -> DB Version: 1.1.39
20:01:00 T:140583465117440   DEBUG: CecLogMessage - >> 50:8f
20:01:00 T:140583465117440   DEBUG: CecLogMessage - >> 05:90:00
20:01:00 T:140582323193600  NOTICE: EMBY LibrarySync -> DB Version: 1.1.39
20:01:01 T:140583711201152   DEBUG: SECTION:UnloadDelayed(DLL: special://xbmcbin/system/ImageLib-x86_64-linux.so)
20:01:01 T:140583711201152   DEBUG: Unloading: /ImageLib-x86_64-linux.so
20:01:01 T:140582323193600  NOTICE: EMBY LibrarySync -> DB Version: 1.1.39
20:01:09 T:140582483629824  NOTICE: Previous line repeats 7 times.
20:01:09 T:140582483629824  NOTICE: ES: Client  from 127.0.0.1 timed out
20:01:10 T:140582323193600  NOTICE: EMBY LibrarySync -> DB Version: 1.1.39
20:01:10 T:140583465117440   DEBUG: CecLogMessage - >> 50:8f
20:01:10 T:140583465117440   DEBUG: CecLogMessage - >> 05:90:00
20:01:11 T:140582323193600  NOTICE: EMBY LibrarySync -> DB Version: 1.1.39
20:01:15 T:140582483629824  NOTICE: Previous line repeats 4 times.
20:01:15 T:140582483629824  NOTICE: ES: Client  from 127.0.0.1 timed out
20:01:16 T:140582323193600  NOTICE: EMBY LibrarySync -> DB Version: 1.1.39
```

josh4trunks, 2015-09-29:
I also posted this here.
https://github.com/OpenELEC/OpenELEC.tv/issues/4346

Popsikill, 2015-11-07:
Just wanted to add I am getting this also!
HTPC: RPi2
OS: Openelec 6.0
TV: Sony KDL-55W800B
Soundplate: Sony HTXT1

SSuuuuuper annoying

opdenkamp, 2015-11-27:
@josh4trunks this is an issue with the EDID that's being read from your TV or with the OpenELEC build that you're using.
libCEC doesn't get a valid EDID from xrandr, and then defaults to HDMI 1 of the TV. Which your AVR is using too. This is confusing the devices.

Try upgrading to the latest OpenELEC build and, if it still occurs, go to system -> input devices -> peripherals -> CEC adapter and set "connected to HDMI device" to "AVR" and the correct HDMI port number. Then press ok.

closed 2015-11-27 by opdenkamp

josh4trunks, 2015-11-27:
@opdenkamp I am on the latest stable build I believe 6.0.0. I have changed that to AVR in my testing, which I do remember seeming to work temporarily. Then it usually resets itself and the option to choose connected device goes away after a few reboots/wakeups from suspend. But I don't remember exactly.

I'll try again and document my results and take pictures of the menu. Thanks

dsebastien, 2016-02-14:
@opdenkamp I wanted to try your solution, but I don't find "connected to HDMI device" anywhere. Has this changed in recent versions? I'm using OpenElec 6.0.1

opdenkamp, 2017-07-13:
Late response: if `connected to HDMI device` or `HDMI port number` is not displayed in Kodi, then it's auto-detected from the EDID, which contains the physical address. If this value is incorrect (e.g. TV switching to the wrong input when starting Kodi), then you can override the value, with the `physical address` setting (uint16_t)