Pulse-Eight/libcec #149: ubuntu 15.10

opdenkamp, 2015-10-26:
build packages for 15.10

spdr870, 2015-11-04:
+1

I just upated to Ubuntu 15.10 and found out that cec-client doesn't work. I could install cec-utils, but get the following error on executing cec-client:
cec-client: symbol lookup error: /usr/lib/x86_64-linux-gnu/libcec.so.3.0: undefined symbol: _ZN11StringUtils6FormatB5cxx11EPKcz

Deus-Rex, 2015-11-10:
I've recently upgraded too.
Executing cec-client gives me:
    
     libcec.so.3.0: cannot open shared object file: No such file or directory
     Cannot load libcec.so
    

Any solution, or will I need to wait for now?

RSully, 2015-11-11:
Related to #151?

rbalint, 2015-11-26:
Ubuntu already ships libcec 3.0.1:
https://launchpad.net/ubuntu/+source/libcec/3.0.1+dfsg2-4
Is this the version failing for you or a custom install?

opdenkamp, 2015-11-26:
I got asked by Team Kodi to provide an official build, since this one is patched (none of those patches have been reported back by the way)

We've talked about it a while ago via email, but since our release schedule doesn't match Debian's, I didn't have time to look into it in detail back then and I do now ;-)

opdenkamp, 2015-11-26:
"official" meaning one provided by us, without any patches applied

rbalint, 2015-11-26:
@opdenkamp
"I got asked by Team Kodi to provide an official build, since this one is patched (none of those patches have been reported back by the way)"
Actually I started reporting them back and I'm waiting for the merging of this to move on to the next patches https://github.com/Pulse-Eight/platform/pull/16

I have also raised a few issues which block further PRs:
https://github.com/Pulse-Eight/platform/issues

I would happily move forward with upstreaming patches when the issues above are solved.

opdenkamp, 2015-11-26:
Right, I'm picking all these thing up atm, but didn't get to the platform lib yet.

opdenkamp, 2015-11-26:
> I have also raised a few issues which block further PRs:

btw, I'd like to point out again that libCEC != platform lib. If there are license issues in that code which block something, then it's blocking XBMC/Kodi too and should have blocked it in the past, because that's where it's coming from (one of pvr add-ons I suppose). But that doesn't relate to the changes done to libCEC itself. The platform lib just originated from libCEC once, got forked in XBMC's PVR add-ons and got merged back together again.

rbalint, 2015-11-26:
I know, but there are two things connecting them. One is changing platform's name which induces several changes in libcec. The second is that there are similar code and changing them the same way would help future maintainers of the code and both projects are maintained by Pulse-Eight.
I'm trying to minimize the work on the changes for both of us. :-)

opdenkamp, 2016-01-20:
we'll push our own builds to our PPA too, and sync up with your patches later. thanks

closed 2016-01-20 by opdenkamp