Pulse-Eight/libcec #150: exc_bad_access on OSX

RSully, 2015-10-26:
Using 3.0.1 I get an EXC_BAD_ACCESS using `DetectAdapters` (with last param false) or `Open`. If I call `DetectAdapters` with last param true, it does not trigger it (but then as soon as I `Open` it does).

Same code works fine using 2.2.0. 

My app with the problem (commit where 3.0.1 was added) [here](https://github.com/RSully/mac-cec-volume/commit/6c66692746e55e197606c8e8e9cc860e8622c874). You'll have to modify the path to the dylib and the header/library search paths.

RSully, 2015-10-26:
Isolated the libcec code from my app into this single file: https://gist.github.com/RSully/ff40878b9b1311f20c49

Compiled/linked against 3.0.1's dylib. When ran it produces an exc_bad_access on line 50 in `DetectAdapters`.

opdenkamp, 2015-11-25:
can you run in a debugger like gdb and provide a backtrace please

opdenkamp, 2015-11-25:
reported on irc to be fixed by @RSully

closed 2015-11-25 by opdenkamp