Pulse-Eight/libcec #155: Sporadic hangs in DetectAdapters

UEckhardt, 2015-11-20:
DetectAdapters sporadically hangs. One special setup may be, that on my system on the initial start no TV is available (disconnected via switchable extension lead). I could get the following backtrace of my hanging system (OpenSUSE 13.2 64Bit, LibCEC compiled from latest git revision (commit 2c675dac48387c48c7f43c5d2547ef0c4ef5c7dd).
```
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
bt
Core was generated by `/video/bin/vdr -w 60 -l 3 --lirc -L /video/lib/ -s /video/bin/vdrpoweroff.sh -v'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007f21e90677bc in __lll_lock_wait () from /lib64/libpthread.so.0
Missing separate debuginfos, use: zypper install alsa-plugins-debuginfo-1.0.28-3.1.3.x86_64 fontconfig-debuginfo-2.11.1-2.1.2.x86_64 glibc-debuginfo-2.19-16.15.1.x86_64 glibc-locale-debuginfo-2.19-16.15.1.x86_64 imlib2-loaders-debuginfo-1.4.5-17.2.2.x86_64 libGLU1-debuginfo-9.0.0-9.1.3.x86_64
libImlib2-1-debuginfo-1.4.5-17.2.2.x86_64 libMagick++-6_Q16-5-debuginfo-6.8.9.8-12.1.x86_64 libMagickCore-6_Q16-2-debuginfo-6.8.9.8-12.1.x86_64 libMagickWand-6_Q16-2-debuginfo-6.8.9.8-12.1.x86_64 libX11-6-debuginfo-1.6.2-5.1.2.x86_64 libX11-xcb1-debuginfo-1.6.2-5.1.2.x86_64 libXau6-debuginfo-1.0.8-5.1.2.x86_64 libXext6-debuginfo-1.3.3-2.1.2.x86_64 libXrandr2-debuginfo-1.4.2-4.1.2.x86_64 libXrender1-debuginfo-0.9.8-4.1.2.x86_64 libasound2-debuginfo-1.0.28-6.1.4.x86_64 libavcodec56-debuginfo-2.8.1-2.1.x86_64
libavutil54-debuginfo-2.8.1-2.1.x86_64 libboost_regex1_54_0-debuginfo-1.54.0-10.1.3.x86_64 libboost_system1_54_0-debuginfo-1.54.0-10.1.3.x86_64 libboost_thread1_54_0-debuginfo-1.54.0-10.1.3.x86_64 libbz2-1-debuginfo-1.0.6-29.2.7.x86_64 libcap2-debuginfo-2.22-13.1.8.x86_64 libcddb2-debuginfo-1.3.2-24.1.2.x86_64 libcdio14-debuginfo-0.90-5.1.3.x86_64 libcdio_cdda1-debuginfo-10.2+0.90-4.1.2.x86_64 libcdio_paranoia1-debuginfo-10.2+0.90-4.1.2.x86_64 libcelt0-2-debuginfo-0.11.1-16.1.2.x86_64 
libexif12-debuginfo-0.6.21-9.1.3.x86_64 libexpat1-debuginfo-2.1.0-14.1.2.x86_64 libffi4-debuginfo-4.8.3+r212056-2.2.4.x86_64 libfftw3-3-debuginfo-3.3.4-2.1.5.x86_64 libfreetype6-debuginfo-2.5.3-2.4.1.x86_64 libgcc_s1-debuginfo-4.8.3+r212056-2.2.4.x86_64 libgcrypt20-debuginfo-1.6.1-8.10.1.x86_64 libgif6-debuginfo-5.0.5-4.1.2.x86_64 libglib-2_0-0-debuginfo-2.42.2-9.1.x86_64 libglibmm-2_4-1-debuginfo-2.42.0-1.1.x86_64 libgmodule-2_0-0-debuginfo-2.42.2-9.1.x86_64 libgobject-2_0-0-debuginfo-2.42.2-9.1.x86_64 libgomp1-debuginfo-4.8.3+r212056-2.2.4.x86_64 libgpg-error0-debuginfo-1.15-1.2.x86_64
libgsm1-debuginfo-1.0.13-25.1.2.x86_64 libgtop-2_0-10-debuginfo-2.30.0-2.1.3.x86_64 libicu53_1-debuginfo-53.1-2.3.2.x86_64 libid3tag0-debuginfo-0.15.1b-183.1.2.x86_64 libjbig2-debuginfo-2.1-2.1.2.x86_64 libjpeg62-debuginfo-62.1.0-30.5.1.x86_64 libjpeg8-debuginfo-8.0.2-30.5.1.x86_64 liblcms2-2-debuginfo-2.6-2.1.2.x86_64 liblqr-1-0-debuginfo-0.4.2-4.2.3.x86_64 libltdl7-debuginfo-2.4.2-15.2.2.x86_64 liblzma5-debuginfo-5.0.7-1.1.x86_64 libmp3lame0-debuginfo-3.99.5-1015.1.x86_64 libogg0-debuginfo-1.3.1-2.1.2.x86_64
libopencore-amrnb0-debuginfo-0.1.3-4.1.x86_64 libopencore-amrwb0-debuginfo-0.1.3-4.1.x86_64 libopus0-debuginfo-1.1-2.1.2.x86_64 liborc-0_4-0-debuginfo-0.4.22-2.1.x86_64 libpcre1-debuginfo-8.37-3.5.1.x86_64 libpng16-16-debuginfo-1.6.13-2.4.1.x86_64 libpugixml1_2-debuginfo-1.2-1.1.x86_64 libschroedinger-1_0-0-debuginfo-1.0.11-9.1.4.x86_64 libsigc-2_0-0-debuginfo-2.4.1-4.1.x86_64 libspeex1-debuginfo-1.1.999_1.2rc1-23.1.2.x86_64 libstdc++6-gcc49-debuginfo-4.9.0+r211729-2.1.7.x86_64 libswresample1-debuginfo-2.8.1-2.1.x86_64 libswscale3-debuginfo-2.8.1-2.1.x86_64 libtheoradec1-debuginfo-1.1.1-27.2.2.x86_64
libtheoraenc1-debuginfo-1.1.1-27.2.2.x86_64 libtiff5-debuginfo-4.0.4-10.10.1.x86_64 libtwolame0-debuginfo-0.3.13-2.1.x86_64 libudev1-debuginfo-210-25.16.1.x86_64 libva1-debuginfo-1.2.1-5.1.3.x86_64 libvdpau1-debuginfo-0.8-2.3.1.x86_64 libvo-aacenc0-debuginfo-0.1.2-3.1.x86_64 libvorbis0-debuginfo-1.3.4-3.1.4.x86_64 libvorbisenc2-debuginfo-1.3.4-3.1.4.x86_64 libvpx1-debuginfo-1.3.0-2.1.2.x86_64 libwebp5-debuginfo-0.4.1-1.2.x86_64 libx264-148-debuginfo-0.148svn20150804-1.1.x86_64 libx265-59-debuginfo-1.7-1.1.x86_64 
libxcb-dpms0-debuginfo-1.11-2.1.2.x86_64 libxcb-icccm4-debuginfo-0.4.1-2.1.2.x86_64 libxcb-screensaver0-debuginfo-1.11-2.1.2.x86_64 libxcb1-debuginfo-1.11-2.1.2.x86_64 libxml++-2_6-2-debuginfo-2.37.1-2.2.1.x86_64 libxml2-2-debuginfo-2.9.1-7.2.1.x86_64 libxvidcore4-debuginfo-1.3.3-1.1.x86_64 libz1-debuginfo-1.2.8-5.1.2.x86_64
(gdb) bt
#0  0x00007f21e90677bc in __lll_lock_wait () at /lib64/libpthread.so.0
#1  0x00007f21e90634a4 in _L_lock_952 () at /lib64/libpthread.so.0
#2  0x00007f21e9063306 in pthread_mutex_lock () at /lib64/libpthread.so.0
#3  0x00007f219e0aa31a in PLATFORM::CMutex::Lock() () at /usr/lib64/libcec.so.3.0
#4  0x00007f219e0aa438 in PLATFORM::CLockObject::CLockObject(PLATFORM::CMutex&, bool) ()
    at /usr/lib64/libcec.so.3.0
#5  0x00007f219e0c7761 in CEC::CCECAdapterMessageQueue::Write(CEC::CCECAdapterMessage*) ()
    at /usr/lib64/libcec.so.3.0
#6  0x00007f219e0c3853 in CEC::CUSBCECAdapterCommunication::SendCommand(CEC::cec_adapter_messagecode, CEC::CCECAdapterMessage&, bool) () at /usr/lib64/libcec.so.3.0
#7  0x00007f219e0c1570 in CEC::CUSBCECAdapterCommands::PingAdapter() () at /usr/lib64/libcec.so.3.0
#8  0x00007f219e0c401e in CEC::CUSBCECAdapterCommunication::PingAdapter() ()
    at /usr/lib64/libcec.so.3.0
#9  0x00007f219e0c3a22 in CEC::CUSBCECAdapterCommunication::CheckAdapter(unsigned int) ()
    at /usr/lib64/libcec.so.3.0
#10 0x00007f219e0c25f4 in CEC::CUSBCECAdapterCommunication::Open(unsigned int, bool, bool) ()
    at /usr/lib64/libcec.so.3.0
#11 0x00007f219e0ae8fb in CEC::CCECProcessor::OpenConnection(char const*, unsigned short, unsigned int, bool) () at /usr/lib64/libcec.so.3.0
#12 0x00007f219e0b07c3 in CEC::CCECProcessor::GetDeviceInformation(char const*, CEC::libcec_configuration*, unsigned int) () at /usr/lib64/libcec.so.3.0
#13 0x00007f219e0ba796 in CEC::CLibCEC::GetDeviceInformation(char const*, CEC::libcec_configuration*, unsigned int) () at /usr/lib64/libcec.so.3.0
#14 0x00007f219e0baa91 in CEC::CLibCEC::DetectAdapters(CEC::cec_adapter_descriptor*, unsigned char, char const*, bool) () at /usr/lib64/libcec.so.3.0
#15 0x00007f21d649950a in cCECRemote::Connect() (this=this@entry=0xd5e4290) at cecremote.cc:259
#16 0x00007f21d649c08c in cCECRemote::cCECRemote(cCECGlobalOptions const&, cPluginCecremote*) (this=0xd5e4290, options=..., plugin=<optimized out>) at cecremote.cc:181
#17 0x00007f21d649d8e2 in cPluginCecremote::Start() (this=0x24ee610) at cecremoteplugin.cc:131
#18 0x00000000004dff15 in cPluginManager::StartPlugins() (this=this@entry=0x7fff59374620)
    at plugin.c:376
#19 0x00000000004767d0 in main(int, char**) (argc=<optimized out>, argv=<optimized out>) at vdr.c:841
```

opdenkamp, 2015-11-25:
I'll need to see what the other threads are doing (as some other thread is locking things here).
In gdb: `thread apply all bt`

UEckhardt, 2015-11-25:
The output of thread apply all:
```
Thread 16 (Thread 0x7f21b9dea700 (LWP 1295)):
#0  0x00007f21e7a5be4d in nanosleep () at /lib64/libc.so.6
#1  0x00007f21e7a5bce4 in sleep () at /lib64/libc.so.6
#2  0x00007f21da2cb61d in WorkerThread::Action() (this=0x2f09280) at tools.c:377
#3  0x000000000051524f in cThread::StartThread(cThread*) (Thread=0x2f09280) at thread.c:262
#4  0x00007f21e90610a4 in start_thread () at /lib64/libpthread.so.0
#5  0x00007f21e7a8a08d in clone () at /lib64/libc.so.6

Thread 15 (Thread 0x7f21d2838700 (LWP 1278)):
#0  0x00007f21e7a81c5d in poll () at /lib64/libc.so.6
#1  0x00000000004f3002 in cSectionHandler::Action() (this=0x28f8e80) at sections.c:184
#2  0x000000000051524f in cThread::StartThread(cThread*) (Thread=0x28f8e80) at thread.c:262
#3  0x00007f21e90610a4 in start_thread () at /lib64/libpthread.so.0
#4  0x00007f21e7a8a08d in clone () at /lib64/libc.so.6

Thread 14 (Thread 0x7f219ffff700 (LWP 1301)):
#0  0x00007f21e9065408 in pthread_cond_timedwait@@GLIBC_2.3.2 () at /lib64/libpthread.so.0
#1  0x0000000000514cb2 in cCondVar::TimedWait(cMutex&, int) (this=this@entry=0x2f0a0a0, Mutex=..., TimeoutMs=TimeoutMs@entry=100) at thread.c:127
#2  0x00007f21d9dfb028 in cGraphTFTDisplay::wait(unsigned long) (this=this@entry=
    0x2f09c30, updateIn=updateIn@entry=29965) at display.c:592
#3  0x00007f21d9dff975 in cGraphTFTDisplay::Action() (this=0x2f09c30) at display.c:568
#4  0x000000000051524f in cThread::StartThread(cThread*) (Thread=0x2f09c48) at thread.c:262
#5  0x00007f21e90610a4 in start_thread () at /lib64/libpthread.so.0
#6  0x00007f21e7a8a08d in clone () at /lib64/libc.so.6

Thread 13 (Thread 0x7f21d483c700 (LWP 1268)):
#0  0x00007f21e9065408 in pthread_cond_timedwait@@GLIBC_2.3.2 () at /lib64/libpthread.so.0
#1  0x0000000000514cb2 in cCondVar::TimedWait(cMutex&, int) (this=this@entry=0x2884308, Mutex=..., TimeoutMs=TimeoutMs@entry=1000) at thread.c:127
#2  0x0000000000496366 in cDvbTuner::Action() (this=0x2883b80) at dvbdevice.c:1000
#3  0x000000000051524f in cThread::StartThread(cThread*) (Thread=0x2883b80) at thread.c:262
#4  0x00007f21e90610a4 in start_thread () at /lib64/libpthread.so.0
#5  0x00007f21e7a8a08d in clone () at /lib64/libc.so.6

Thread 12 (Thread 0x7f21b95e9700 (LWP 1296)):
#0  0x00007f21e7a832f3 in select () at /lib64/libc.so.6
#1  0x00007f21da079596 in cTBSelect::Select(unsigned int) (this=this@entry=0x7f21b95e8ca0, TimeoutMs=TimeoutMs@entry=400) at select.c:32
#2  0x00007f21da05c845 in cStreamdevServer::Action() (this=0x2f09790) at server.c:101
#3  0x000000000051524f in cThread::StartThread(cThread*) (Thread=0x2f09790) at thread.c:262
#4  0x00007f21e90610a4 in start_thread () at /lib64/libpthread.so.0
#5  0x00007f21e7a8a08d in clone () at /lib64/libc.so.6

Thread 11 (Thread 0x7f219d805700 (LWP 1302)):
#0  0x00007f21e9065408 in pthread_cond_timedwait@@GLIBC_2.3.2 () at /lib64/libpthread.so.0
#1  0x00007f219e0aa167 in PLATFORM::CConditionImpl::Wait(pthread_mutex_t&, unsigned int) ()
    at /usr/lib64/libcec.so.3.0
#2  0x00007f219e0ac570 in PLATFORM::CCondition<bool>::Wait(PLATFORM::CMutex&, bool (*)(void*), void*, unsigned int) () at /usr/lib64/libcec.so.3.0
#3  0x00007f219e0abf9b in PLATFORM::CCondition<bool>::Wait(PLATFORM::CMutex&, bool&, unsigned int) ()
    at /usr/lib64/libcec.so.3.0
#4  0x00007f219e0ac1e4 in PLATFORM::SyncedBuffer<CEC::CCallbackWrap*>::Pop(CEC::CCallbackWrap*&, int) () at /usr/lib64/libcec.so.3.0
#5  0x00007f219e0a920d in CEC::CCECClient::Process() () at /usr/lib64/libcec.so.3.0
#6  0x00007f219e0aa621 in PLATFORM::CThread::ThreadHandler(void*) () at /usr/lib64/libcec.so.3.0
#7  0x00007f21e90610a4 in start_thread () at /lib64/libpthread.so.0
#8  0x00007f21e7a8a08d in clone () at /lib64/libc.so.6

Thread 10 (Thread 0x7f21d3039700 (LWP 1277)):
#0  0x00007f21e9065408 in pthread_cond_timedwait@@GLIBC_2.3.2 () at /lib64/libpthread.so.0
#1  0x0000000000514cb2 in cCondVar::TimedWait(cMutex&, int) (this=this@entry=0x28902b8, Mutex=..., TimeoutMs=TimeoutMs@entry=1000) at thread.c:127
#2  0x0000000000496366 in cDvbTuner::Action() (this=0x288fb30) at dvbdevice.c:1000
#3  0x000000000051524f in cThread::StartThread(cThread*) (Thread=0x288fb30) at thread.c:262
#4  0x00007f21e90610a4 in start_thread () at /lib64/libpthread.so.0
#5  0x00007f21e7a8a08d in clone () at /lib64/libc.so.6
Thread 9 (Thread 0x7f21bb5f5700 (LWP 1291)):
#0  0x00007f21e9065408 in pthread_cond_timedwait@@GLIBC_2.3.2 () at /lib64/libpthread.so.0
#1  0x0000000000514ace in cCondWait::Wait(int) (this=this@entry=0x7f21bb5f4e60, TimeoutMs=TimeoutMs@entry=333) at thread.c:71
#2  0x0000000000514b46 in cCondWait::SleepMs(int) (TimeoutMs=TimeoutMs@entry=333) at thread.c:60
#3  0x00007f21dc2196c8 in cRadioImage::Action() (this=0x2f083e0) at radioaudio.c:174
#4  0x000000000051524f in cThread::StartThread(cThread*) (Thread=0x2f083e0) at thread.c:262
#5  0x00007f21e90610a4 in start_thread () at /lib64/libpthread.so.0
[... 85 lines cut ...]
#2  0x00007f21e9063306 in pthread_mutex_lock () at /lib64/libpthread.so.0
#3  0x00007f219e0aa31a in PLATFORM::CMutex::Lock() () at /usr/lib64/libcec.so.3.0
#4  0x00007f219e0aa438 in PLATFORM::CLockObject::CLockObject(PLATFORM::CMutex&, bool) ()
    at /usr/lib64/libcec.so.3.0
#5  0x00007f219e0c7761 in CEC::CCECAdapterMessageQueue::Write(CEC::CCECAdapterMessage*) ()
    at /usr/lib64/libcec.so.3.0
[... cut ...]
#15 0x00007f21d649950a in cCECRemote::Connect() (this=this@entry=0xd5e4290) at cecremote.cc:259
#16 0x00007f21d649c08c in cCECRemote::cCECRemote(cCECGlobalOptions const&, cPluginCecremote*) (this=0xd5e4290, options=..., plugin=<optimized out>) at cecremote.cc:181
#17 0x00007f21d649d8e2 in cPluginCecremote::Start() (this=0x24ee610) at cecremoteplugin.cc:131
#18 0x00000000004dff15 in cPluginManager::StartPlugins() (this=this@entry=0x7fff59374620)
    at plugin.c:376
#19 0x00000000004767d0 in main(int, char**) (argc=<optimized out>, argv=<optimized out>) at vdr.c:841
```

opdenkamp, 2015-11-25:
right, threads 1 and 7 aren't happy. will check what's going on there. can you link the code of this `cecremoteplugin`?

UEckhardt, 2015-11-25:
The description and code can be found on [http://www.uli-eckhardt.de/vdr/cec.en.shtml] . The link of the code can be found at the end of the page.

UEckhardt, 2016-01-07:
I have an other core from a hang. Since this backtrace is somewhat long, I try to attach it as file.
[gdb.txt](https://github.com/Pulse-Eight/libcec/files/81477/gdb.txt)

opdenkamp, 2016-01-07:
```
Thread 8 (Thread 0x7f30ecdf6700 (LWP 1149)):

#13 0x00007f30ff0b4c10 in CEC::CCECClient::SendPowerOnDevices(CEC::cec_logical_address) () at /usr/lib64/libcec.so.3.0
#14 0x00007f30ff0c927c in CEC::CLibCEC::PowerOnDevices(CEC::cec_logical_address) () at /usr/lib64/libcec.so.3.0
#15 0x00007f313ee8df3c in cCECRemote::Action() (this=0xc5fba50) at cecremote.cc:582
```

```
Thread 1 (Thread 0x7f3152562740 (LWP 1109)):

#45 0x00007f30ff0df7f2 in CEC::CCECBusDevice::GetVendorId(CEC::cec_logical_address, bool) () at /usr/lib64/libcec.so.3.0
#46 0x00007f30ff0c137d in CEC::CCECProcessor::RegisterClient(std::shared_ptr<CEC::CCECClient>) () at /usr/lib64/libcec.so.3.0
#47 0x00007f30ff0c8cb3 in CEC::CLibCEC::Open(char const*, unsigned int) () at /usr/lib64/libcec.so.3.0
#48 0x00007f313ee8c586 in cCECRemote::Connect() (this=0xc5fba50) at cecremote.cc:291
```

you're calling `CEC::CLibCEC::PowerOnDevices()` before `CEC::CLibCEC::Open()` returned. not very nice that it locks up now, but this won't work anyway.

UEckhardt, 2016-01-08:
Thanks for the info. Indeed I have found a racing condition in my application.

opdenkamp, 2016-01-19:
closing this issue then :)

closed 2016-01-19 by opdenkamp