Pulse-Eight/libcec #156: CEC and HDMI issues

paulyz, 2015-11-21:
"when I power on my RPI2 running on 5.95.5, somehow, my home theater system turns on too. If I manually turn off home theater system and reboot RPI2 - then home theater do not turn on. With 5.0.8 I never had this problem.
My system: TV panasonic, connected to RPI2 over HDMI and controlled over CEC (RPI2 powered from TV's USB). Home theater system connected over HDMI (I can manually turn on home theater system through TV's remote controller)."

"I found one more interesting thing, my TV has 4 HDMI ports. 1 - empty, 2 - home theater system, 3 - RPI2, 4 - empty. If I press on remote AV button to change input source, go to empty HDMI 4 and then back to HDMI 3, always home theater system turns on automatic after couple seconds. This happens with Kodi15. With Kofi 14 everything OK. Looks like RPI2 on Kodi sends wrong signal to HDMI 2."

log: http://xbmclogs.com/psmxnv1cf#

opdenkamp, 2015-11-25:
this changed in libCEC indeed: https://github.com/Pulse-Eight/libcec/commit/588f8f1213a892b01e981b69f4bf1508492194b0

most people will want this to happen, but there's always people who don't like it ;-)
will look at adding a setting for it in Kodi and libCEC

mk01, 2015-12-03:
@opdenkamp 

beside 'don't like' group it breaks all setups with libCEC not attached via AVR. most annoying is AVRs 'restoring' last active inputs (or predefined) by sending routing change (80).- (what is not wrong as AVR wasn't powered On by SystemAudioMode Request) - breaking desired setup completely. 

btw: xbmc/libcec has everything needed, just the setting is ignored by xbmc's CEC code. there is even a PR few months old https://github.com/xbmc/xbmc/pull/7245

321liftoff, 2016-03-05:
Just wanted to confirm this behavior in another setup.  Samsung TV Input 3 connected to Cubox-i4 running Openelec 6.0.3.  Yamaha AVR connected to Samsung TV Input 2.  CEC Peripheral is enabled and all other sub-CEC settings are off or ignore/none, including "Wake devices when deactivating the screensaver"  When the AVR is off, switching to Input 3 causes the AVR to power on.  When the TV is turned off, turning it on when already on Input3 and pressing arrow cursor on remote to deactivate dim-screen turns on the AVR.

Link to log below:
http://sprunge.us/FiJJ

paulyz, 2016-06-29:
@opdenkamp when? :) Because, I still on kodi 14 and can't upgrade

mdodig, 2016-09-01:
Any news with this one?

commit 736dd56b (2016-10-25): API cleanup. bump to 4.0..0. see debian/changelog.in for details. added setting to configure automatic avr wakeup. closes #156
closed 2016-10-25 by opdenkamp