Pulse-Eight/libcec #163: How to see CEC traffic in eventghost

Stiefmeister, 2015-11-29:
I was wondering if there is a way to see all CEC traffic as events in Eventghosts?
Currently only traffic from predefined events is shown in Eventghost it seems.

Thanks a lot!

opdenkamp, 2015-11-30:
there's no setting for this. you'd have to change a line in the add-on to get this now: https://github.com/Pulse-Eight/libcec/blob/master/src/EventGhost/__init__.py#L106

change `cec.CEC_LOG_WARNING` into `cec.CEC_LOG_TRAFFIC`

the add-on is installed in the addons dir of EventGhost

Stiefmeister, 2015-11-30:
Thank you!
But apparently that does not show the traffic as events you can use in Eventghost to trigger activities but just as log/information.
What I would like to do is the following:
Get an event in Eventghost when my HTPC is set as active source from LG Beamer and use this event to trigger start of Kodi Mediacenter.

If it is not possible to show all traffic as event is there a way to add specific raw cec commands that are shown as events in Eventghost?

Thank you!!!

opdenkamp, 2015-12-01:
reopened as feature request. it's currently not implemented

winescb, 2016-12-11:
Just wondering if this was ever implemented and if so what would be the proper way to upgrade from 3.0 as pulse-eight doesn't seem to keep their website updated.

Thanks and appreciate all the great work!