Pulse-Eight/libcec #164: Compile fails on clean Raspbian install

attaxia, 2015-12-01:
Following the instructions from the readme on a Raspberry Pi 2 using a clean install of Raspbian Jessie the following error occurs when compiling libcec:

```
pi@raspberrypi:~/libcec/build $ cmake -DRPI_INCLUDE_DIR=/opt/vc/include -DRPI_LIB_DIR=/opt/vc/lib ..
CMake Error at src/cec-client/CMakeLists.txt:21 (find_package):
  By not providing "Findp8-platform.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "p8-platform", but CMake did not find one.

  Could not find a package configuration file provided by "p8-platform" with
  any of the following names:

    p8-platformConfig.cmake
    p8-platform-config.cmake

  Add the installation prefix of "p8-platform" to CMAKE_PREFIX_PATH or set
  "p8-platform_DIR" to a directory containing one of the above files.  If
  "p8-platform" provides a separate development package or SDK, be sure it
  has been installed.

-- Configuring incomplete, errors occurred!
See also "/home/pi/libcec/build/CMakeFiles/CMakeOutput.log".
```

The platform compiled fine.

opdenkamp, 2015-12-01:
You didn't include the platform lib before you built. But because I'm currently doing some work on that in master, you better take the release branch instead for now. Readme will be updated

closed 2015-12-01 by opdenkamp

Tristan79, 2016-11-18:
The instructions on building it on the pi are unclear for me. I am stuck having the same problem as above... so how do I include the platform lib before building? It a fresh jessy install.

avisiedo, 2017-02-20:
Install by hand "p8-platform".  I have done the following:
`
$ cd /tmp
$ git clone --depth 1 https://github.com/Pulse-Eight/platform/tree/p8-platform-2.1.0.1
$ cd platform
$ mkdir build
$ cd build
$ cmake ..
$ make
$ sudo make install
`

And then continue with the build of kodi.

Regards

opdenkamp, 2017-07-13:
I'll add a basic readme to the platform library with build instructions.

opdenkamp, 2017-07-13:
Basic instructions have been added https://github.com/Pulse-Eight/platform/blob/master/README.md